machinekit/mksocfpga #3: Altera Implemention of SRL16E faulty

the-snowwhite, 2016-03-06:
The Mesa sources use a xilinx specialized ip core several places, so It would be great to have a fully working Altera style replacement.

The one I have been able to come up with needs some small adjustments, most likely by adding a separate Q15. 

the only related to this implementation source of the xilinx core is here:

https://www.pa.msu.edu/hep/d0/ftp/run2b/l1cal/hardware/channel_link_tester/channel_link_tester_firmware/common/srl16e_i.vhd

This is most likely the issue that is limiting jogging to one direction and giving the joint following error in axis gui.

In quartus it gives following connection errors:

-----

 
    +----------------------------------------------------------------------------------------------------------------------------------------------------------------+
    ; Port Connectivity Checks: "HostMot2:HostMot2_inst|stepgen:\makestepgens:generatestepgens:0:usg:stepgenx|SRL16E:\steptable:0:asr16e|Shiftreg16:Shiftreg16_inst" ;
    +---------+--------+----------+----------------------------------------------------------------------------------------------------------------------------------+
    ; Port    ; Type   ; Severity ; Details                                                                                                                          ;
    +---------+--------+----------+----------------------------------------------------------------------------------------------------------------------------------+
    ; q[9..1] ; Output ; Info     ; Connected to dangling logic. Logic that only feeds a dangling port will be removed.                                              ;
    +---------+--------+----------+----------------------------------------------------------------------------------------------------------------------------------+

    +--------------------------------------------------------------------------------------------------------------------+
    ; Port Connectivity Checks: "HostMot2:HostMot2_inst|stepgen:\makestepgens:generatestepgens:9:usg:stepgenx"           ;
    +----------+--------+----------+-------------------------------------------------------------------------------------+
    ; Port     ; Type   ; Severity ; Details                                                                             ;
    +----------+--------+----------+-------------------------------------------------------------------------------------+
    ; hold     ; Input  ; Info     ; Stuck at GND                                                                        ;
    ; stout[1] ; Output ; Info     ; Connected to dangling logic. Logic that only feeds a dangling port will be removed. ;
    +----------+--------+----------+-------------------------------------------------------------------------------------+

cdsteinkuehler, 2016-03-06:
That's something I can help with.  I'll try to get a PR together soon (just about to head out to dinner with the family), but here's an untested snippit of VHDL if you're anxious to get started:

````
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity SRLC16E is
    port (
        d           : in  std_logic;
        ce          : in  std_logic;
        clk         : in  std_logic;
        a0          : in  std_logic;
        a1          : in  std_logic;
        a2          : in  std_logic;
        a3          : in  std_logic;
        q           : out std_logic;
        q15         : out std_logic );
end SRLC16E;

architecture arch of SRLC16E is

    signal reg          : std_logic_vector(15 downto 0);

begin

    process (clk)
    begin
        if rising_edge (clk) then
            if ce='1' then
                reg <= reg(14 downto 0) & d;
            end if;
        end if;
    end process;

    -- Assign outputs
    q15 <= reg(15);
    q   <= reg(to_integer(unsigned(a3 & a2 & a1 & a0)));

end arch;
````

the-snowwhite, 2016-03-06:
Thanks

NOTE:
What really is a PITA is the joint following error triggered on every 
attempt to go in - direction.

I have been using the megawizard and had forgotten to include the clock 
enable signal.
However I think it can be done in a more elegant fashion without the 
megawizard plug-in
so you are very welcome to look over the latest commits to master and 
work branches.

---

What really is a PITA is the joint following error triggered on every 
attempt to go in - direction.

I'm now rebooting to jessie so that I can make some measurements of the 
dir and step pins again.

the step pins did pulse on + jogs earlier, however dir pins stayed low..

cdsteinkuehler, 2016-03-07:
On 3/6/2016 5:49 PM, Michael Brown wrote:
> 
> NOTE:
> What really is a PITA is the joint following error triggered on every 
> attempt to go in - direction.

Yes, that's a major pain.  For testing, you should just craft a HAL
file with the hm2 driver and something simple (like a sine wave
generator) feeding the position or velocity input of the stepgen so
you can see what's going on w/o needing to launch the whole linuxcnc
stack with the GUI and everything.  HAL is very powerful all by itself.

I'm just about caught up at work and am hoping to have some time to
help debug this next week.  Sorry I haven't been able to do more so far.

-- 
Charles Steinkuehler
charles@steinkuehler.net

the-snowwhite, 2016-03-08:
done

closed 2016-03-08 by the-snowwhite