machinekit/mksocfpga #9: Support Sensible Firmware Names & Multiple Configurations

cdsteinkuehler, 2016-04-14:
The generated bit file for all current projects is named "soc_system.rbf", which will rapidly cause problems once more than one person is using this code on a single board.  The Mesanet firmware, for example, not only has different firmware builds for each physical FPGA platform, but also several 'flavors' of the firmware depending on which daughter cards are connected.

For example, the 5i25 (which Michael Brown based the initial DE0-Nano configuration on) has at least 14 different bit files (7i74x2.bit, 7i76_7i74.bit, etc).

I have started work on an expansion card for the DE0-Nano which could potentially support up to four DB25 connectors.  When combined with the various daughter cards available from Mesa, this makes a *LOT* of firmware permutations!  We need to think about how to keep this from causing mass confusion, and how to make it easy to build custom firmware configurations (either upon request, or hopefully easy enough so actual users could do it).

mhaberler, 2016-04-14:
yes, sorely needed (my [hack](https://github.com/machinekit/machinekit/commit/be123dc81b204fa9ddc6abc1fa264cba9feda95a) with the custom deb is only for a start to put things in their proper place and get the SD image off the ground

I see these dimensions:

- taxonomy: naming the source and build artefacts in this repo 
- naming the debian package(s) (one per bitfile? the whole zoo per platform?)
- use branches? (I guess no - but that has implications for the build process; rebuilding everything from scratch will be a number_of_configs * 20min affair; a persistent container which retains artefacts might help but the dependencies MUST be bulletproof for this to work; if we rebuild the whole zoo we'll hit the travis CPU limit sooner or later, so we might have to move this to our server - maybe @kinsamanka has an idea)
- use per-config subdirectories under HW/QuartusProjects/DE0_NANO_SOC_GHRD ?
- a lot can be driven by build environment variables, like to pass a make target through the CI system

Ideally the deb packaging happens in the build vm/container because the information _what_ was actually rebuild is available there

alternatively, if we have a persistent build vm/container we can package outside (shared directory with host) and always wrap up all artefacts into a deb

cdsteinkuehler, 2016-04-14:
On 4/14/2016 2:17 PM, Michael Haberler wrote:
> yes, sorely needed (my hack 
> <https://github.com/machinekit/machinekit/commit/be123dc81b204fa9ddc6abc1fa264cba9feda95a> 
> with the custom deb is only for a start to put things in their proper place and 
> get the SD image off the ground

Understood, but the soc_system name should probably die as soon as
possible.

I don't have any great ideas for how to deal with this yet, but one
observation is the hm2 code is quite stable.  That means I would
expect most of the changes to this repo would need to trigger the
build of *ONE* configuration rather than build/rebuild everything each
commit.

In other words, someone is going to add or tweak something like a
DE0-Nan0_7i77x2 configuration and _that_ will need to get built, but
all 15 other DE0-Nano configs (and the configurations for various
other boards) will *NOT* need to be (re)built.

...but that's going to be a fancy trick for an automated CI environment!

-- 
Charles Steinkuehler
charles@steinkuehler.net

mhaberler, 2016-04-14:
so what's the moving parts in a config? theory:

- platform: `DE0-Nano`, ` DE1..`, `SoCkit`  
- IO: direct GPIO pin connect vs Peter's tentacle boards (7ixx)
- stepgen, pwmgen, encoder, 3pwmgen etc: max instance counts

anything left out?

we're not limited to DOS filenames so it's possible to have configs named like 
`<platform>-<adaptercount>-<io-method>-<instancecounts>` (ideally that name is parseable to deduct capabilities)

example: `de0_nano-0-gpio-6s2e2p`  `de0_nano-1-7i77-6s2e2p`  `de0_nano-2-7i77:7i76-6s2e2p`

the-snowwhite, 2016-04-16:
This next large message is a sketch of how to proceed solutions and links to relevant info I would like to propose.
Partitioning --> hardned ip (soc) core, incremental compilition..... dynamic or static reconfiguration

the-snowwhite, 2016-04-16:
First split the design into 2 (Virtual)partitions:   Soc and fpga

1 (soc) main partition (HPS , CPU + ipcores and logic wires for all physical board parts).

This is made into a hardware hardened ip  that is routed and placed physicially in the chip,
this part of the design only needs to be compiled once for every board, regardless of number of added configs.
This part will Include provisions (have reserved resources) for interfacing to All hostmot2 logic.

2 (fpga)
This part is the hostmot2 playground:

first partition:
A static interface to the Soc partition

Implement a new hostmot2.(vhd / sv) top file that can act as a bridge / interface:

Build hostmot2 cores as module blocks: (4 pwm , 4 encoders)

Number of identical cores in each block is calculated / estimated from each blaoc having simlar size in terma of fpga resources used to generate them:

Every hm2 module block has same interface to top / bridge interface core:

How many hostmot2 cores ?

+

Partitioning and incremential Compilation:

https://engineering.purdue.edu/ISLWiki/pub/Variations/FpgaFlow/bestpractices.pdf

Dynamic / Static Partial Reconfiguration.

https://en.wikipedia.org/wiki/Reconfigurable_computing#High-performance_computing

[(Best Practices for Incremental Compilation Partitions and Floorplan Assignments]
(https://engineering.purdue.edu/ISLWiki/pub/Variations/FpgaFlow/bestpractices.pdf)

https://www.altera.com/en_US/pdfs/literature/wp/wp-01137-stxv-dynamic-partial-reconfig.pdf

[Not that we need this other way yet, but will just mention this should also works for the Cyclone V family we are using]
(https://www.altera.com/content/dam/altera-www/global/en_US/pdfs/literature/ug/ug_partrecon.pdf)

http://www.eejournal.com/archives/articles/20150512-alteraturbo/

[Creating Design Partitions](http://quartushelp.altera.com/14.1/mergedProjects/comp/increment/comp_pro_qid_create_design_partitions.htm)

http://wl.altera.com/education/training/courses/OBLUEINTRO

Prerequisites
We recommend completing the following courses:
Creating a System Design with Qsys [Creating a System Design with Qsys "(OQSYSCREATE)" 37 Minutes Online Course](http://wl.altera.com/education/training/courses/OQSYSCREATE)

Introduction to Qsys [Introduction to Qsys "(OQSYS1000)" 26 Minutes Online Course](http://wl.altera.com/education/training/courses/OQSYS1000)

Introduction to the Qsys System Integration Tool [Introduction to the Qsys System Integration Tool "(IQSYS101)" 8 Hours Instructor-Led or Virtual Class Course (non free)](http://wl.altera.com/education/training/courses/IQSYS101)

The Quartus Prime Software: Foundation "(Online Training)" [The Quartus Prime Software: Foundation (Online Training) (ODSW1110)
216 Minutes Online Course](http://wl.altera.com/education/training/courses/ODSW1110)

Using the Quartus Prime Software: An Introduction [Using the Quartus Prime Software: An Introduction (ODSW1100) 81 Minutes Online Course](http://wl.altera.com/education/training/courses/ODSW1100)

[Cash Cow:]
(http://wl.altera.com/education/training/courses/IPR100)

Blueprint Platform Designer.
[Spectra-Q Engine](https://www.altera.com/products/design-software/fpga-design/quartus-prime/features/spectra-q.html)

the-snowwhite, 2016-04-16:
Note: some edits above

cdsteinkuehler, 2016-05-05:
I have been reviewing the Mesa configuration setup since I need so build a bitfile that supports my just-assembled DE0-Nano_DB25 board for connecting DB25 daughtercards to the DE0.  As a refresher, the Mesa configurations define two packages, one for the physical board and one for the hostmot2 logic that gets build.  The configuration is chosen by uncommenting the desired one from a commented list of choices in the top-level FPGA file.

The top-level file for the DE0-Nano is not VHDL, but system verilog, which makes using packages this way somewhat difficult.  Also, the "uncomment the one you want" approach does not really lend itself to building multiple configurations in an automated way.

The "Mesa way" of a list of packages could be easily implemented with a VHDL wrapper between the system verilog top-level and the hostmot2 instance.  This would probably cause the least amount of disruption to the existing source code, and make the most sense to anyone working with both our code and the actual hostmot2 sources from Mesa.

Alternately, VHDL configurations (or other late binding) techniques could be used to select how to configure the hostmot2 instance, or the package names could be overloaded and the FPGA project file would include the desired variant (eg: there are three pins files: PINS_CFG1.vhd, PINS_CFG2.vhd, & PINS_CFG3.vhd, each of which provide the PINS package, with only one file being included in any particular build).

It mostly depends on if we want to have a bunch of FPGA project files, or if we one one project file we can configure by passing in a configuration flag of some sort.  I'm not sure which makes more sense for the automated build setups.

NOTE: Some options may require features of VHDL2008 (package generics could be very helpful).

the-snowwhite, 2016-05-06:
Yes
Just a side note about how the current Quartus config system works:

As long as you stick to the 5i25 board you can compile for any daughter card including you own custom pin file by replacing exactly 1 file:

the one in line 4 in the .qip file here:

https://github.com/machinekit/mksocfpga/blob/master/HW/hm2/config/hm2_i25_G540x2_34_config.qip#L4 <https://github.com/machinekit/mksocfpga/blob/master/HW/hm2/config/hm2_i25_G540x2_34_config.qip#L4>

Or you can create a new copy of this qip. file rename it after the pin file,,, etc.

Changing the +board file here will however not work due to the reason Charles has stated below.

cdsteinkuehler, 2016-05-06:
On 5/6/2016 3:28 AM, Michael Brown wrote:
> Yes
> Just a side note about how the current Quartus config system works:
>
> As long as you stick to the 5i25 board you can compile for any daughter card
> including you own custom pin file by replacing [...]

Yes, it's easy to change the pinout for a single configuration.

I'm trying to work out how we can manage building multiple
configurations from the same source tree.

Also, I would propose migrating the board-specific files in the
hm2/config directory to the specific project directory (eg:
QuartusProjects/DE0-NANO_SOC_GHRD/).  Otherwise, the config directory
will become *HUGE* (see for example how many pin configurations there
are just for the 5i25 board in the Mesa source).

Finally, I don't yet understand how the information from the pins file
is accessed in the top-level system-verilog file.  In the Mesa source
there is a line specifically pulling in the particular pins package
desired for the build, eg:

    use work.PIN_7I77x2_34.all;

There should be something similar for the top-level ghrd.sv file:

    use work.PIN_G540x2_34.all;

...to clarify the PinDesc in this line:

https://github.com/machinekit/mksocfpga/blob/master/HW/QuartusProjects/DE0_NANO_SOC_GHRD/ghrd.sv#L394

...is really supposed to be PIN_G540x2_34.PinDesc.  Where/how does
this happen (asks the VHDL guy who only vaguely understands verilog).

-- 
Charles Steinkuehler
charles@steinkuehler.net

the-snowwhite, 2016-05-06:
First a quick post to point out that there actually are 3 working board 
projects in the
QuartusProjects folder.

So your proposal would immediately triple that amount with 3 versions of 
identical files,
+ for each new board added.
Im sure keeping the pin file in 1 directory, eventually with some sub 
folders, for pin files only working
on 1 board would be more manageable ?

---

Also due to the mesa probing functions (first being commented out and 
the deleted later) being left out from the mesa pci driver that was used 
as template for the original hm2_soc.c driver.

None of the information from the pin files make their way into the hal.

Only daughter-board config info that gets in is the static info put into 
the hm2_soc.c file.

----
I made some tests the other day,but
As I have been complaining about elsewhere I'm missing something in my 
workflow that can enable me to online notes of such findings and just 
need to link them in right now wheh they are relevant.
----

I will see if I can dig them up, asap

cdsteinkuehler, 2016-05-06:
On 5/6/2016 8:34 AM, Michael Brown wrote:
> First a quick post to point out that there actually are 3 working board
> projects in the
> QuartusProjects folder.
> 
> So your proposal would immediately triple that amount with 3 versions of
> identical files,
> + for each new board added.

Only because you have defaulted to having each board look like a 5i25
clone.  Ultimately, each board should have it's own ID and make use of
the unique I/O pinned out on that specific board (eg: the DE0 has
dramatically different available I/O than the SoCKIT).

> Also due to the mesa probing functions (first being commented out and
> the deleted later) being left out from the mesa pci driver that was used
> as template for the original hm2_soc.c driver.
>
> None of [...]

This is going to be a huge problem.  The driver needs to pull the hm2
configuration data from the IDROM compiled into the FPGA or things
will get *REALLY* confusing!  Am I correct that you're saying this is
not currently happening?

-- 
Charles Steinkuehler
charles@steinkuehler.net

the-snowwhite, 2016-05-06:
This is what you get even with this (edited) pin file:

https://github.com/the-snowwhite/mksocfpga/blob/hostmot3/HW/hm2/config/PIN_G540x2_34.vhd

	machinekit@mksocfpga:~/machinekit/src$ cat  /var/log/linuxcnc.log

	Apr 28 14:46:00 mksocfpga msgd:0: startup pid=2382 flavor=rt-preempt rtlevel=1 usrlevel=1 halsize=524288 shm=Posix gcc=4.9.2 version=unknown
	Apr 28 14:46:00 mksocfpga msgd:0: ØMQ=4.0.5 czmq=3.0.2 protobuf=2.6.1 libwebsockets=<no version symbol>
	Apr 28 14:46:00 mksocfpga msgd:0: configured: sha=50e9958
	Apr 28 14:46:00 mksocfpga msgd:0: built:      Apr 27 2016 14:54:55 sha=50e9958
	Apr 28 14:46:00 mksocfpga msgd:0: register_stuff: actual hostname as announced by avahi='mksocfpga.local'
	Apr 28 14:46:00 mksocfpga msgd:0: zeroconf: registering: 'Log service on mksocfpga.local pid 2382'
	Apr 28 14:46:00 mksocfpga msgd:0: rtapi_app:2387:user accepting commands at ipc:///tmp/0.rtapi.a42c8c6b-4025-4f83-ba28-dad21114744a
	Apr 28 14:46:01 mksocfpga msgd:0: zeroconf: registered 'Log service on mksocfpga.local pid 2382' _machinekit._tcp 0 TXT "uuid=a42c8c6b-4025-4f83-ba28-dad21114744a" "instance=ec60de26-0d4f-11e6-98ac-823af0754b95" "service=log" "dsn=ipc:///tmp/0.log.a42c8c6b-4025-4f83-ba28-dad21114744a"
	Apr 28 14:46:01 mksocfpga msgd:0: hal_lib:2387:rt hm2: loading Mesa HostMot2 driver version 0.15
	Apr 28 14:46:01 mksocfpga msgd:0: hal_lib:2387:rt hm2_soc: loading Mesa AnyIO HostMot2 socfpga driver version 0.8
	Apr 28 14:46:01 mksocfpga msgd:0: hal_lib:2387:rt hm2: no firmware specified in config modparam!  the board had better have firmware configured already, or this won't work
	Apr 28 14:46:01 mksocfpga msgd:0: hal_lib:2387:rt hm2/hm2_5i25.0: 34 I/O Pins used:
	Apr 28 14:46:01 mksocfpga msgd:0: hal_lib:2387:rt hm2/hm2_5i25.0:     IO Pin 000 (P3-01): IOPort
	Apr 28 14:46:01 mksocfpga msgd:0: hal_lib:2387:rt hm2/hm2_5i25.0:     IO Pin 001 (P3-14): PWMGen #0, pin Out0 (PWM or Up) (Output)
	Apr 28 14:46:01 mksocfpga msgd:0: hal_lib:2387:rt hm2/hm2_5i25.0:     IO Pin 002 (P3-02): StepGen #0, pin Step (Output)
	Apr 28 14:46:01 mksocfpga msgd:0: hal_lib:2387:rt hm2/hm2_5i25.0:     IO Pin 003 (P3-15): IOPort
	Apr 28 14:46:01 mksocfpga msgd:0: hal_lib:2387:rt hm2/hm2_5i25.0:     IO Pin 004 (P3-03): StepGen #0, pin Direction (Output)
	Apr 28 14:46:01 mksocfpga msgd:0: hal_lib:2387:rt hm2/hm2_5i25.0:     IO Pin 005 (P3-16): StepGen #4, pin Step (Output)
	Apr 28 14:46:01 mksocfpga msgd:0: hal_lib:2387:rt hm2/hm2_5i25.0:     IO Pin 006 (P3-04): StepGen #1, pin Step (Output)
	Apr 28 14:46:01 mksocfpga msgd:0: hal_lib:2387:rt hm2/hm2_5i25.0:     IO Pin 007 (P3-17): IOPort
	Apr 28 14:46:01 mksocfpga msgd:0: hal_lib:2387:rt hm2/hm2_5i25.0:     IO Pin 008 (P3-05): StepGen #1, pin Direction (Output)
	Apr 28 14:46:01 mksocfpga msgd:0: hal_lib:2387:rt hm2/hm2_5i25.0:     IO Pin 009 (P3-06): StepGen #2, pin Step (Output)
	Apr 28 14:46:01 mksocfpga msgd:0: hal_lib:2387:rt hm2/hm2_5i25.0:     IO Pin 010 (P3-07): StepGen #2, pin Direction (Output)
	Apr 28 14:46:01 mksocfpga msgd:0: hal_lib:2387:rt hm2/hm2_5i25.0:     IO Pin 011 (P3-08): StepGen #3, pin Step (Output)
	Apr 28 14:46:01 mksocfpga msgd:0: hal_lib:2387:rt hm2/hm2_5i25.0:     IO Pin 012 (P3-09): StepGen #3, pin Direction (Output)
	Apr 28 14:46:01 mksocfpga msgd:0: hal_lib:2387:rt hm2/hm2_5i25.0:     IO Pin 013 (P3-10): Encoder #0, pin A (Input)
	Apr 28 14:46:01 mksocfpga msgd:0: hal_lib:2387:rt hm2/hm2_5i25.0:     IO Pin 014 (P3-11): Encoder #0, pin B (Input)
	Apr 28 14:46:01 mksocfpga msgd:0: hal_lib:2387:rt hm2/hm2_5i25.0:     IO Pin 015 (P3-12): Encoder #0, pin Index (Input)
	Apr 28 14:46:01 mksocfpga msgd:0: hal_lib:2387:rt hm2/hm2_5i25.0:     IO Pin 016 (P3-13): IOPort
	Apr 28 14:46:01 mksocfpga msgd:0: hal_lib:2387:rt hm2/hm2_5i25.0:     IO Pin 017 (P2-01): IOPort
	Apr 28 14:46:01 mksocfpga msgd:0: hal_lib:2387:rt hm2/hm2_5i25.0:     IO Pin 018 (P2-14): PWMGen #1, pin Out0 (PWM or Up) (Output)
	Apr 28 14:46:01 mksocfpga msgd:0: hal_lib:2387:rt hm2/hm2_5i25.0:     IO Pin 019 (P2-02): StepGen #5, pin Step (Output)
	Apr 28 14:46:01 mksocfpga msgd:0: hal_lib:2387:rt hm2/hm2_5i25.0:     IO Pin 020 (P2-15): IOPort
	Apr 28 14:46:01 mksocfpga msgd:0: hal_lib:2387:rt hm2/hm2_5i25.0:     IO Pin 021 (P2-03): StepGen #5, pin Direction (Output)
	Apr 28 14:46:01 mksocfpga msgd:0: hal_lib:2387:rt hm2/hm2_5i25.0:     IO Pin 022 (P2-16): StepGen #9, pin Step (Output)
	Apr 28 14:46:01 mksocfpga msgd:0: hal_lib:2387:rt hm2/hm2_5i25.0:     IO Pin 023 (P2-04): StepGen #6, pin Step (Output)
	Apr 28 14:46:01 mksocfpga msgd:0: hal_lib:2387:rt hm2/hm2_5i25.0:     IO Pin 024 (P2-17): IOPort
	Apr 28 14:46:01 mksocfpga msgd:0: hal_lib:2387:rt hm2/hm2_5i25.0:     IO Pin 025 (P2-05): StepGen #6, pin Direction (Output)
	Apr 28 14:46:01 mksocfpga msgd:0: hal_lib:2387:rt hm2/hm2_5i25.0:     IO Pin 026 (P2-06): StepGen #7, pin Step (Output)
	Apr 28 14:46:01 mksocfpga msgd:0: hal_lib:2387:rt hm2/hm2_5i25.0:     IO Pin 027 (P2-07): StepGen #7, pin Direction (Output)
	Apr 28 14:46:01 mksocfpga msgd:0: hal_lib:2387:rt hm2/hm2_5i25.0:     IO Pin 028 (P2-08): StepGen #8, pin Step (Output)
	Apr 28 14:46:01 mksocfpga msgd:0: hal_lib:2387:rt hm2/hm2_5i25.0:     IO Pin 029 (P2-09): StepGen #8, pin Direction (Output)
	Apr 28 14:46:01 mksocfpga msgd:0: hal_lib:2387:rt hm2/hm2_5i25.0:     IO Pin 030 (P2-10): Encoder #1, pin A (Input)
	Apr 28 14:46:01 mksocfpga msgd:0: hal_lib:2387:rt hm2/hm2_5i25.0:     IO Pin 031 (P2-11): Encoder #1, pin B (Input)
[... cut ...]
	May  6 13:58:40 mksocfpga rtapi:0: unload: 'hm2_soc' not loaded
	May  6 13:58:40 mksocfpga rtapi:0: unload: 'motmod' not loaded
	May  6 13:58:40 mksocfpga msgd:0: rtapi_app exit detected - scheduled shutdown
	May  6 13:58:42 mksocfpga msgd:0: msgd shutting down
	May  6 13:58:42 mksocfpga msgd:0: zeroconf: unregistering 'Log service on mksocfpga.local pid 1873'
	May  6 13:58:42 mksocfpga msgd:0: log buffer hwm: 0% (37 msgs, 3964 bytes out of 524288)
	May  6 13:58:42 mksocfpga msgd:0: normal shutdown - global segment detached

the-snowwhite, 2016-05-06:
I am currently working on turning the config system, the other way around, so that pins are assigned from the hal.
And so that hal ip cores / ip core blocks can be swapped in and out via the -dtb blobs.

In the hostmot3 experiment which makes much more sense to me right now.
------

> Only because you have defaulted to having each board look like a 5i25
clone.  Ultimately, each board should have it's own ID and make use of

Yes we did agree upon starting all new projects as a 5i25 clone to make things more simple.

So All current configs can be moved to a sub-folder named 5i25-board, which still will have a 3 to 1 ratio compared to not keeping the config  and project folders separate.  (I have not seen if the added Vivado project also depends on the current 5i25 pin files ?)

Adding board folders in the config folder for the pin files, seems more manageable for the mksocfpga hostmot2 compatibility to work for multiple boards and fpga vendors.

cdsteinkuehler, 2016-05-06:
On 5/6/2016 9:15 AM, Michael Brown wrote:
> So All current configs can be moved to a sub-folder named 5i25-board, which
> still will have a 3 to 1 ratio compared to not keeping the config and project
> folders separate. (I have not seen if the [...]

OK, that's fine...the location of the files is a minor detail.

What I don't currently understand is if there's enough of the original
hostmot2 functionality in place to allow things like changing the
number of stepgen and smart serial instances in the FPGA and have that
get picked up by the driver.  This functionality is mandatory for me
to create a proper configuration for the hardware setup I'm attempting
(a DE0-Nano + my DB25 adapter board + 7i85S + 7i76).

-- 
Charles Steinkuehler
charles@steinkuehler.net

the-snowwhite, 2016-05-06:
https://github.com/the-snowwhite/mksocfpga/blob/master/HW/hm2/idrom_tools.vhd

Processes the pin config, the top .sv file does not need to processes anything (Xilinx projects included AFAIK ..?)

the-snowwhite, 2016-05-06:
https://github.com/machinekit/machinekit/blob/master/src/hal/drivers/mesa-hostmot2/hm2_pci.c#L460

That is what is needed to put back in to get the mesa config probe function re-included again.

I almost lost the project, back whe I went drirectly from the hdl programming and at high speed attempted to morph that pci driver file in 2 or 3 dayes ... :-)

So a lot of oversimplification happened to get initial basic functioning to work and iron out the
stepper only moves in one direction bug, for instance ....

I should be very easy for a c programmer to add that function now.

cdsteinkuehler, 2016-05-06:
A note for anyone following along.

The current setup is hard-coded and cannot be changed from the top
level ghrd.sv file.  The default PIN_G540x2_34 configuration is pulled
in by default in the hostmot2.vhd file:

https://github.com/machinekit/mksocfpga/blob/master/HW/hm2/hostmot2.vhd#L69

Crafting a way to manage the configuration without dramatically
changing the existing hostmot2 setup (with a unique VHDL package for
each 'flavor') is going to be challenging.  Ideas welcome.

If systemverilog can understand the VHDL types well enough, it might
be possible to do something like

    import PINS_PKG::*;

...in the ghrd.sv file, then build with a verilog macro defined at
compile time, eg: "'define PINS_PKG PIN_G540x2_34", then pass the
appropriate values from the package to the hostmot2 instance.

I don't work enough with cross-language builds to know if this will
work or not, or if it's at all portable between the Xilinx and Altera
tools.

-- 
Charles Steinkuehler
charles@steinkuehler.net

cdsteinkuehler, 2016-05-06:
I think the best way to handle the configuration details is using the
Makefile with an auto-generated VHDL file.  I do this in my day-job to
work around some limitations of the build tools, and to do things like
incorporate commit IDs and build-date into the HDL code.

PROPOSAL:
Craft a 'wrapper' for hostmot2.vhdl (wrap_hostomt2.vhdl.in or similar)
that has place-holders for the board and pin package names.  As part
of the make build process, the place-holders in the "stock" wrapper
file would be replaced with build-time values supplied from the
makefile and the build can then proceed normally.  This is basically
automating the manual "comment/uncomment" configuration being done in
the stock top-level hostmot2 files from Mesanet.  The wrapper file
would not strictly be necessary if a way could be found to merge VHDL
generics into the ghrd.sv top level file, but IMHO using a small
wrapper file is cleaner than trying to dynamically change the
top-level design file.

Individual configurations could then be built with something like:

    make rbf BOARD=DE0-Nano PINS=G540x2_34

Thoughts?

-- 
Charles Steinkuehler
charles@steinkuehler.net

the-snowwhite, 2016-05-07:
OK I am thinking
I will be out today for going to the forest today with my family and will first return tonight.

AS the changes also affect the Vivado project added It would be great to hear:
@dkhughes  opinion about the proposed changes.

I am also having a hard time keeping track of all issues related to the mksoc project, so I very much need some help in implementing the Milestone that should be able to server as this missing place holder, or to be brought out of this misbeleif / misconception of hope if that is not the case:

https://github.com/machinekit/machinekit/issues/930

cdsteinkuehler, 2016-05-08:
The changes don't affect the Vivado project...they are limited to the Makefile in the DE0_Nano project (and adding some files in the hm2/config directory).

I have run into another issue, however.  There is no provision for mapping the hostmot2 port pins arbitrarily.  It is assumed that iobits[0] is always the first GPIO on the first port.  This makes sense for the Mesa configurations which have the FPGA and connectors on the same board, but can't really deal with the DE0-Nano, where various daughter-cards can be made to hook to different Mesa remote daughtercards (for instance one board for DB25 interfaced boards, and a different board for 50-pin boards).

So some additional "shim" layer is needed to be able to map the hostmot2 I/O pins to the proper top-level I/O pins.  I am currently taking the view that a "board" matches the Mesa view of a board, and includes both the SoC platform (ie: DE0, DE1, SoCKIT) and the adapter board used to provide Mesa compatible connectors.  So I will be adding a DE0_Nano_SoC_DB25 project that will potentially have various PIN configurations to support a variety of Mesa daughtercards (up to 4 total).  Adding a new project this way also dramatically reduces the changes required to any existing code, so I'm far less likely to break anything or cause merge conflicts once I get things working.  There is a bit of code duplication, but that's mostly already there (eg: there are already several soc_system.qsys designs and associated support files that are all basically identical).

the-snowwhite, 2016-05-08:
>
> The changes don't affect the Vivado project...they are limited to the
Makefile in the DE0_Nano project (and adding some files in the hm2/config
directory).
>
> I have run into another issue, however. There is no provision for mapping
the hostmot2 port pins arbitrarily. It is assumed that iobits[0] is always
the first GPIO on the first port. This makes sense for the Mesa
configurations which have the FPGA and connectors on the same board, but
can't really deal with the DE0-Nano, where various daughter-cards can be
made to hook to different Mesa remote daughtercards (for instance one board
for DB25 interfaced boards, and a different board for 50-pin boards).

I noticed that there does not seen to be a Real pinmux controlable from the
hal. I have projected plans for this in the hostmot3 version.

>
> So some additional "shim" layer is needed to be able to map the hostmot2
I/O pins to the proper top-level I/O pins. I am currently taking the view
that a "board" matches the Mesa view of a board, and includes both the SoC
platform (ie: DE0, DE1, SoCKIT) and the adapter board used to provide Mesa
compatible connectors. So I will be adding a DE0_Nano_SoC_DB25 project that
will potentially have various PIN configurations to support a variety of
Mesa daughtercards (up to 4 total). Adding a new project this way also
dramatically reduces the changes required to any existing code, so I'm far
less likely to break anything or cause merge conflicts once I get things
working. There is a bit of code duplication, but that's mostly already
there (eg: there are already several soc_system.qsys designs and associated
support files that are all basically identical).

Yes agree.

dkhughes, 2016-05-08:
@cdsteinkuehler Your proposed method is how I did it in the Vivado side, but since I'm a noob with Makefiles I just wrote a bash script that does the work. See:

[https://github.com/machinekit/mksocfpga/blob/master/HW/VivadoProjects/make_bitfile.sh](https://github.com/machinekit/mksocfpga/blob/master/HW/VivadoProjects/make_bitfile.sh)

Correct me if I am wrong, but the shim layer for IO pins *is* the PIN file. That file does all of the io mapping for the hm2 that is register accessible. You then define a hardware pin constraint that assigns the proper ball on the package separately. If your daughter card is directly connected to a ball on the FPGA, then there shouldn't be any major change needed to the design. If you have a bus connected daughter card, I think you would have to make a custom com layer for it to work unless it spoke the mesanet serial protocol already built in for their daughter cards.

cdsteinkuehler, 2016-05-08:
On 5/8/2016 3:47 PM, dkhughes wrote:
> @cdsteinkuehler <https://github.com/cdsteinkuehler> Your proposed method is how 
> I did it in the Vivado side, but since I'm a noob with Makefiles I just wrote a 
> bash script that does the work.

Yes, that's pretty much what I am going to be doing.  I'm not sure if
a shell script like you've got or a Makefile tweak to run the sed
"preprocessor" will be easier to integrate into the automated builds.

It's not much difference to me either way, we'll see what the CI gurus
say they want.

> Correct me if I am wrong, but the shim layer for IO pins /is/ the PIN file. That
> file does all of the io mapping for the hm2 that is register accessible. You
> then define a hardware pin constraint [...]

There's no bus, but the existing hostmot2 configuration via the PIN
file is not enough to properly support my DB25 adapter board.  The
problem is essentially that for the existing Mesa configurations, the
iobits() array of signals is pinned out to specific FPGA pins that
have a known connection to the expansion connectors (either DB25 or
50-pin, depending on the board).  For the DE0-Nano (and most any other
off-the-shelf development SoC platform), there is another layer of
translation needed.  The PIN file specifies which iobits() signal has
each particular function, and the top-level reference design has
generic I/O pins hooked to off-board connectors.  There is no
provision in the PIN files to handle which generic I/O pin is hooked
to which particular Mesa DB25/50-pin daughter-board connector.

For a specific example using the DE0-Nano and my DB25 adapter board:

* The DE0-Nano has two 36-bit GPIO busses: GPIO_0 and GPIO_1, each of
which connects to a 40-pin 0.1" header

* The hostmot2 instance exports an iobits() bus, which connects to the
physical FPGA I/O pins

* To assist with PCB layout, I 'scrambled' the GPIO_0/GPIO_1
connections to the Mesa DB25 connectors, plus I added two LED signals
that are not part of the iobits() signal bus

* I could fix this with the FPGA pinout (which is what would be done
for an official Mesanet board), but then the GPIO_0/GPIO_1 pin names
no longer match the development board naming (causing general
confusion for everyone, and added support burden going forward)

...so I'm considering the combination of an adapter card and an
off-the-shelf SoC development platform a "board" and crafting a
Quartus project for each combination (ie: DE0-Nano + DB25 adapter card
is one project...if someone makes a 50-pin adapter, or a DB25 adapter
with a different pinout, that will be a different Quartus project).  I
will hard-code the connections between the iobits() signals from the
hostmot2 instance and the GPIO_0/GPIO_1 buses used in the dev-kit
reference design, at which point I can craft multiple PIN files to
deal with whatever combinations of Mesanet daughter-cards are connected.

-- 
Charles Steinkuehler
charles@steinkuehler.net

mhaberler, 2016-05-08:
@cdsteinkuehler dont use sed too early

consider using cpp and -Dsmybol for composing variant configurations (where `symbol` stands for a config), using cpp directives if/then/else, include, define

if the later stages barf on the cpp linemarkers (`# linenum filename flags`) they still can be removed 

I can do an example if you want

happy to compose makefiles, given a list of dependencies (like files artefacts depend on), the list of artefacts, and say a shell script which translates the former to the latter

cdsteinkuehler, 2016-05-08:
On 5/8/2016 5:09 PM, Michael Haberler wrote:
> @cdsteinkuehler <https://github.com/cdsteinkuehler> dont use sed too early

But I like sed!  ;-)

> consider using cpp and -Dsmybol for composing variant configurations (where 
> |symbol| stands for a config), using cpp directives if/then/else, include, define

It doesn't much matter to me what mechanism is used, what matters is
that it integrates well with both the FPGA tools and with the CI build
process.  The various constraints I am aware of are:

* Only *ONE* FPGA build process can run at a time (on a specific build
host and in a specific project directory).  In other words, if there
are six PIN configurations for a specific FPGA project directory, they
must be built sequentially or in independent build environments (eg:
multiple docker containers, one per build)

* The FPGA project will include and depend on a source file that is
not actually under revision control (or is not usable without some
modification...like using the cpp to pre-process)

* Something (sed, cpp, whatever) needs to create this file,
substituting a few keywords with appropriate text based on the desired
build configuration

So...do the CI gurus have a preferred method for handling this?

Is "make CONFIG=PIN_XYZ" better or worse than "./configure PINS=XYZ &&
make"?  Would something else be preferred?

-- 
Charles Steinkuehler
charles@steinkuehler.net

mhaberler, 2016-05-08:
dont worry about the execution mode (parallel/sequential), that's taken care of easily
then, one base job can be easily cloned and different global variables (like CONFIG=...) injected
there is also specific support in jenkins for [multi-configuration project setups](https://wiki.jenkins-ci.org/display/JENKINS/Matrix+Project+Plugin)

the flow I thought of is:

- you have a toplevel file (in git) which uses if/then/else/include/define etc, and after processing by cpp yields say a verilog or vhdl source file, or whatever these tools need - that is an interim artefact the vendor toolchain then chews on
- the build process calls this step iterating over a number of variants, say by injecting different -Dsymbol values
- each run yields some artefeacts which are packaged under the config name and uploaded to the apt repo

you dont have to cook up a makefile - a shell script with say two variants defined by an argument are enough for a start

dkhughes, 2016-05-09:
@cdsteinkuehler That example is exactly what I had with my custom PCB - I also had to scramble the gpio pinout, and the location of the stepgen outputs. The Vivado project structure analogs what you describe where each SoM can have multiple carriers, in which the PIN file attached to that carrier handles the dirty muxing.

> * I could fix this with the FPGA pinout (which is what would be done
> for an official Mesanet board), but then the GPIO_0/GPIO_1 pin names
> no longer match the development board naming (causing general
> confusion for everyone, and added support burden going forward) 

I'm not following here. Why wouldn't the pin names match? Since the ball mapping is happening outside of hostmot2 (outside of what the HAL sees) and is essentially a permanent mux, where is the discrepancy? If you map IO<0> to the pin labeled GPIO_0 on your board aren't you good to go? Would they not match the 5i25 or another DE0 board?

Looking at the hostmot2 code in the hal, pins.c has a few relic restrictions on port size that I think we can eliminate. Then, with proper llio configuration (not constant values like in the current overlay driver), this should all be completely configurable from the PINDESC in vhdl, and the config passed to the LLIO driver holding port width and number of ports. I think we would just have to place the pins in the proper order in the PINDESC file (that's how I shuffled them on my custom carrier), and they should appear sane in a hal file. Hmmmm.....I need to study the hal code a little more...

the-snowwhite, 2016-05-09:
I would like to propose that instead of creating a physical I/O adapter board.

Add a “virtual” one in the FPGA fabric instead.

Right now the I/O wires are routed directly to the GPIO pins.

Basically 1 hdl “rewire” daughter board adapter interface hdl module could do exactly the same, compensating
for the PCB layout issues.

1 for the DB25 and 1 for the 50 pin.

I have not looked into how to instead combine all this into a bi-directional, tri-stateable, pin muxer. that could load the direct setting,
via some sort of config file.

However inserting a pinmux or 2 before the GPIOx pins after the hostmot2 top core in the top level file could prove valuable.. ?

Michael B.

mhaberler, 2016-05-09:
@cdsteinkuehler  - here's a stab at a jenkins multi-configuration project: https://jenkins.machinekit.io/job/mksocfpga-test2/

see in Configure - Configuration Matrix where the config names are enumerated
these are chewed upon in turn (Execute Shell further down)

mhaberler, 2016-05-09:
re artefact naming and packaging -  what I'll do with the multiple configs is:

- each config gets a `/lib/firmware/socfpga/<configuration>` diretcory
- rbf dtb dts live there as soc_system.rbf soc_system.dts soc_system.dtb

A `<configuration>` is the dir name under HW/QuartusProject, so DE0_NANO_SOC_GHRD orDE0_Nano_SoC_DB25

dkhughes, 2016-05-09:
The overlay driver would require increasing the maximum firmware length in hostmot2 if the path is going to significantly grow by adding more subdirectories:

[https://github.com/machinekit/machinekit/blob/master/src/hal/drivers/mesa-hostmot2/hostmot2.h#L45](https://github.com/machinekit/machinekit/blob/master/src/hal/drivers/mesa-hostmot2/hostmot2.h#L45)

It's used here:

[https://github.com/machinekit/machinekit/blob/master/src/hal/drivers/mesa-hostmot2/hostmot2.c#L1216](https://github.com/machinekit/machinekit/blob/master/src/hal/drivers/mesa-hostmot2/hostmot2.c#L1216)

I think the original 30 character limit is unnecessary. I can increase this to a standard path length (128 or so) and then submit a pull request if wanted.

mhaberler, 2016-05-09:
sure

cdsteinkuehler, 2016-05-09:
On 5/9/2016 7:16 AM, Michael Haberler wrote:
> re artefact naming and packaging - what I'll do with the multiple configs is:
>
>   * each config gets a |/lib/firmware/socfpga/<configuration>| diretcory
>   * rbf dtb dts live there as soc_system.rbf [...]

That handles the "board" level (equivalent to the 5i25 or other FPGA
board from Mesa), but how are multiple PIN configurations for the
various daughter-boards dealt with?  I currently have one PIN file for
the DE0_Nano_SoC_DB25:

https://github.com/machinekit/mksocfpga/blob/master/HW/hm2/config/DE0_Nano_DB25/PIN_7I76_7I85S_GPIO_GPIO.vhd

...but there will be a need for many others.

-- 
Charles Steinkuehler
charles@steinkuehler.net

mhaberler, 2016-05-09:
the way I read the Makefile the basename of the artefacts follows the basename of the .qsys file

so one option is to create qsys files per PIN config 

(just an idea; no clue if this makes any sense)

cdsteinkuehler, 2016-05-09:
On 5/9/2016 7:28 AM, Michael Haberler wrote:
> the way I read the Makefile the basename of the artefacts follows the basename 
> of the .qsys file
> 
> so one option is to create qsys files per PIN config
> 
> (just an idea; no clue if this makes any sense)

That doesn't make much sense.  ;-)

The .qsys file is what defines the ARM side of the hardware
configuration and has nothing to do with the hostmot2 instance.

The bit files are named after the top-level design, which is the
poorly chosen "soc_system" in the example designs that are provided
with the development boards.  I have changed the top-level design name
in my new configuration to the more sensible DE0_Nano_DB25, so if you
run "make rbf" you will wind up with an output_files/DE0_Nano_DB25.rbf
file rather than soc_system.rbf.

To build the bitfile with a different PIN configuration, something in
the build chain needs to modify the hostmot2_cfg.vhd file and select
the proper PIN library to import on this line:

https://github.com/machinekit/mksocfpga/blob/master/HW/QuartusProjects/DE0_Nano_SoC_DB25/hostmot2_cfg.vhd#L71

...then (re)run the "make rbf" process and copy the result somewhere
for safe-keeping, perhaps something like:

  cp DE0_Nano_DB25.rbf DE0_Nano_DB25_7I76_7I85S_GPIO_GPIO.rbf

Alternately, it might make sense to name the bitfiles after their
coded boardname values (in the IDROM), so instead of DE0_Nano_DB25,
bit files for this project would be TERADE0N:

https://github.com/machinekit/mksocfpga/blob/master/HW/hm2/config/DE0_Nano_DB25/DE0_Nano_DB25_card.vhd#L75-L76

-- 
Charles Steinkuehler
charles@steinkuehler.net

dkhughes, 2016-05-09:
That is how the xilinx side is working now. Instead of pulling the info from filenames I just added a cmake esque configuration file that told the make_bitfile script how to configure the project.

If that's a bad thing let me know. I would like to start making the two sides more common and less fragmented.

That doesn't make much sense. ;-)

The .qsys file is what defines the ARM side of the hardware
configuration and has nothing to do with the hostmot2 instance.

The bit files are named after the top-level design, which is the
poorly chosen "soc_system" in the example designs that are provided
with the development boards. I have changed the top-level design name
in my new configuration to the more sensible DE0_Nano_DB25, so if you
run "make rbf" you will wind up with an output_files/DE0_Nano_DB25.rbf
file rather than soc_system.rbf.

To build the bitfile with a different PIN configuration, something in
the build chain needs to modify the hostmot2_cfg.vhd file and select
the proper PIN library to import on this line:

https://github.com/machinekit/mksocfpga/blob/master/HW/QuartusProjects/DE0_Nano_SoC_DB25/hostmot2_cfg.vhd#L71

...then (re)run the "make rbf" process and copy the result somewhere
for safe-keeping, perhaps something like:

cp DE0_Nano_DB25.rbf DE0_Nano_DB25_7I76_7I85S_GPIO_GPIO.rbf

Alternately, it might make sense to name the bitfiles after their
coded boardname values (in the IDROM), so instead of DE0_Nano_DB25,
bit files for this project would be TERADE0N:

https://github.com/machinekit/mksocfpga/blob/master/HW/hm2/config/DE0_Nano_DB25/DE0_Nano_DB25_card.vhd#L75-L76

mhaberler, 2016-05-09:
can the Makefile https://github.com/machinekit/mksocfpga/blob/master/HW/QuartusProjects/DE0_Nano_SoC_DB25/Makefile  be assumed to be modifiable by me or is that some artefact which might be overwritten by an update?

mhaberler, 2016-05-09:
here's an example to create a custom hostmot2_cfg.vhd file driven by build parameters
https://github.com/mhaberler/mksocfpga/commit/bb9548337dcd78be8b2464eb1e32a8192da0efc0

the BOARD and PINCONFIG would be passed down from the jenkins job definition which would hold the combinations to build for

about right direction?

the-snowwhite, 2016-05-09:
Just
Some info to think about about possible gpio pin muxing options available.
#18

dkhughes, 2016-05-09:
Configuring the pindesc  and board name in the top level file breaks compatibility with the xilinx ip packagers.

I was thinking, maybe a better way than makefile configuration (or the .in files that i did on the vivado side) would be to add a vhdl configuration file per project, and make it the only one available on path at compile time. That way, all if this is still in vhdl that is below the ip packagers and completely device generic. We have to define the pindesc and board names anyhow, why not a file named impidromconst.vhdl with the values?

cdsteinkuehler, 2016-05-09:
On 5/9/2016 10:56 AM, Michael Haberler wrote:
> here's an example to create a custom hostmot2_cfg.vhd file driven by build
> parameters
> mhaberler@bb95483
> <https://github.com/mhaberler/mksocfpga/commit/bb9548337dcd78be8b2464eb1e32a8192da0efc0>
>
> the BOARD and PINCONFIG would be passed down from the jenkins job definition
> which would hold the combinations to build for
>
> about right direction?

Yes, that looks like the right direction to me.  I don't know how that 
works for Vivado...I don't know any reason it shouldn't work, but I have 
limited experience with the latest Xilinx tools.

One comment is you are using legal VHDL identifiers as tags, so it's 
possible (however unlikely) that they will appear elsewhere in the code 
and be incorrectly replaced.  I suggest using the @ symbol (eg: 
@BOARD@), or for my day job, I use percent (%BOARD%) to tag these keywords.

--
Charles

mhaberler, 2016-05-09:
ah ok, will fix

dkhughes, 2016-05-10:
Is the Altera soc IP compiled all at once with the project, or do they do out of context builds like vivado for IP? That would be the only road block to eliminating the .in files needed per board without the break in IP packager compatibility that the current solution causes.

Eliminating the need for the .in files in the ip directories I think also gives us the benefit of being able to do parallel project builds....

cdsteinkuehler, 2016-05-10:
On 5/10/2016 10:11 AM, dkhughes wrote:
> Is the Altera soc IP compiled all at once with the project, or do
> they do out of context builds like vivado for IP? That would be the
> only road block to eliminating the .in files needed per board without
> the break in IP packager compatibility that the current solution
> causes.

It is a two-step process, but there are no out of context builds.

Step 1: Generate RTL from the *.qsys file:

https://github.com/machinekit/mksocfpga/blob/master/HW/QuartusProjects/DE0_NANO_SOC_GHRD/Makefile#L162-L164

Step 2: Compile the bitfile from the project source code:

https://github.com/machinekit/mksocfpga/blob/master/HW/QuartusProjects/DE0_NANO_SOC_GHRD/Makefile#L237-L240

-- 
Charles Steinkuehler
charles@steinkuehler.net

cdsteinkuehler, 2016-06-03:
I think this issue is close to being resolved.  The naming convention is to use <board name> + <I/O adapter> + <PIN_configuration>, which will result in a variety of FPGA bit files (one per each unique combination) and device-tree files (one per board/adapter combo).

I believe this issue should either be closed, or perhaps should depend on any sub-issues (ie: build system support for generating all the various FPGA bit files and the subsequent firmware debs, any source code changes needed to generate output files with standardized names, etc).

Thoughts?

the-snowwhite, 2016-06-04:
NOTE:
It is now possible to reduce the amount of bit files needed considerably:

Hmmm I'm now in the catch up process once again after getting the pin mux thing to work in my Hostmot3 branch.

The pinmux functionality adds about 2-3% gates to the "standard" 1 GPIO port project that uses around 36%.

At startup all the pins (0-34) are configured as Straight pins, pins 35 and 36 are currendly hardwired to leds 1 and 2.

The routing of each pin is controlled by newly added registers:
 0x1120, and upwards(to max below: 0x1200 ):
Each 32 bit register controls 4 pins via an 8-bit byte. (default max number of pins are 144).

THe changes can be ported back into the hostmot2 version (I have put all the hm2 core generation into wrappers in the Hostmot3 project, work that is unfinished ATM.)

The ability to "live" pin mux, makes it unnecessary to run separate builds for changing pin assignments, however does break comparability with the original hostmot2.vhd, as minor changes are necessary to add this functionality, back into the mksocfpga2 project.

cdsteinkuehler, 2016-06-04:
On 6/4/2016 5:41 AM, Michael Brown wrote:
> NOTE:
> It is now possible to reduce the amount of bit files needed considerably:
>
> Hmmm I'm now in the catch up process once again after getting the pin mux thing
> to work in my Hostmot3 branch.
>
> The [...]

There is more to the PIN configuration packages than just the pinout,
you also need to add/remove functionality (number of stepgens, number
of pwmgens, existence/type/number of encoder inputs & smart serial
ports, etc).

While a superset of several common configurations could be added, this
approach also pushes proper configuration of the hardware onto the
user when they load and configure the driver.  With the existing
bitfile scheme, they just pick the one that matches their hardware and
that's it.

I'm not saying I'm against what you're doing, but if you do add
non-trivial pin muxing, make sure there is an easy to understand and
fool-proof method of configuring it properly for various hardware
combinations.  If anything, the pin multiplexing on the BeagleBone
(and other ARM SoC's) shows how confusing it is for the typical user
to understand and deal with.

-- 
Charles Steinkuehler
charles@steinkuehler.net

cdsteinkuehler, 2016-08-06:
I'm considering this resolved and closing the issue.  We no longer use the same generic name for all FPGA bit files, and it is possible to support multiple hardware configurations (and the corresponding bitfiles) on the same FPGA platform.  See the DE0_Nano_SoC_DB25 configurations for working examples.

We will likely want to create additional configurations and possible rename some project files for consistency, but these can be their own issue if needed.  The basic problem of using one "magic" name for the FPGA bitfile which prompted me to create this issue has been resolved for some time.

closed 2016-08-06 by cdsteinkuehler