machinekit/mksocfpga #11: cleanup needed: contents of SW/MK directory outdated

mhaberler, 2016-04-16:
[SW/MK](https://github.com/machinekit/mksocfpga/tree/master/SW/MK) contains lots of dead bodies and duplication:

the hm2_soc.c/h ll hostmot driver is [superseded](https://github.com/machinekit/machinekit/blob/master/src/hal/drivers/mesa-hostmot2/hm2_soc.c)  which already goes into the mk packages and works

hm2_socfpga-mk.patch is merged

Same for [kernel-drivers:](https://github.com/machinekit/mksocfpga/tree/master/SW/MK/kernel-drivers):

the only one needed is hm2reg_uio - the status of the other stuff is unclear and I am not sure work-in-progress code should be in a machinekit organisation repo

also, I came up with the dkms wrappings for hm2reg_uio and adcreg_uio - this is the preferred method for any kernel drivers downstream

do the Makefile.tar.gz serve any purpose (?)

same for the [config](https://github.com/machinekit/mksocfpga/tree/master/SW/MK/configs/hm2-soc-stepper) - already in mk/mk

proposal: 
- I transfer to https://github.com/mhaberler/machinekit-dkms and we setup the package build from there (not sure if adcreg should be built if there is no using HALcomp)
- any development work for any kmods happens there
- delete SW/MK subtree in toto save for a note on the required kernel config options

@the-snowwhite - as the adcreg code matures:
- kmod updates go to mk/machinekit-dkms via PR
- HAL comps and other working stuff like configs goes to mk/mk via PR

Also, [documentation](https://github.com/machinekit/mksocfpga/tree/master/docs) should go into [docs proper](https://github.com/machinekit/machinekit-docs) but that needs a bit of organizing first

mhaberler, 2016-04-16:
update: just transferred https://github.com/machinekit/machinekit-dkms

the-snowwhite, 2016-04-16:
Note:
I never got the adc_uio to work in any way:
I seems like a bad idea to use uio to interface to a 2 16-bit reg fifo ?

------

adcreg is the driver that functions:
https://github.com/machinekit/mksocfpga/tree/master/SW/MK/kernel-drivers/adcreg

adcfs contains the c++ (api) and userspace test programs that work with it.
https://github.com/machinekit/mksocfpga/tree/master/SW/MK/kernel-drivers/adcfs

mhaberler, 2016-05-01:
@the-snowwhite - I propose to go ahead with the cleanup as outlined above

if no objectsion I will file a PR.

the-snowwhite, 2016-05-01:
https://github.com/mhaberler/machinekit-dkms/blob/master/drivers/adcreg/adcreg.c
https://github.com/mhaberler/machinekit-dkms/blob/master/drivers/adcreg/Makefile
https://github.com/mhaberler/machinekit-dkms/blob/master/drivers/adcreg/dkms.conf

ok ...:

Whats missing is the adcreg driver demo, and documentation examples of how it can be used (works):
(for connecting it to the hal:  creating a adc hal module ?)

https://github.com/machinekit/mksocfpga/blob/master/SW/MK/kernel-drivers/Adc-commands.txt

https://github.com/machinekit/mksocfpga/blob/master/SW/MK/kernel-drivers/adcfs/adcfs.cpp
https://github.com/machinekit/mksocfpga/blob/master/SW/MK/kernel-drivers/adcfs/main.cpp
https://github.com/machinekit/mksocfpga/blob/master/SW/MK/kernel-drivers/adcfs/fpga.h
https://github.com/machinekit/mksocfpga/blob/master/SW/MK/kernel-drivers/adcfs/Makefile

This should be rather straight forward to port into a normal c - c++ style hal module.

mhaberler, 2016-05-01:
right, but that should be a normal HAL component and posted as PR against mk/mk - the dkms repo is the wrong place for this, as is the mksocfpga repo

the-snowwhite, 2019-05-09:
@cdsteinkuehler 
Since the ADC Driver now finally is fully integrated and working in the hal and also the mksocmemio utility,
I will proceed with a PR only keeping:
https://github.com/machinekit/mksocfpga/tree/master/SW/MK/kernel-drivers/hm2reg_uio-module
and
https://github.com/machinekit/mksocfpga/tree/master/SW/MK/dts-overlays
--- plus
https://github.com/machinekit/mksocfpga/tree/master/SW/MK/packaging
and
https://github.com/machinekit/mksocfpga/tree/master/SW/MK/uio_irq_test
?

cdsteinkuehler, 2019-05-09:
AFAIK, the kernel driver isn't really needed, the generic UIO drivers already in the kernel will allow direct memory access and interrupts with a proper device tree overlay.  The device-tree example is useful in theory, but what's in the repo is undocumented and not particularly helpful.

IMHO, only the mksocmemio and uio_irq_test are likely to be useful.

the-snowwhite, 2019-05-09:
ok
Yes I agree
I'm currently compiling some documentation (from recent threads on the google group) and expect to commit a sketch soon to and upodated fork the docs repo (hope for some feedback on that). 

Also in works is a pure cleanup commit to get rid of the "old fuzz".
https://github.com/the-snowwhite/mksocfpga/tree/cleanup_work

BTW: the mksocmemio utility has mananged to "sneek" in:
https://github.com/machinekit/machinekit-hal/blob/master/src/hal/utils/mksocmemio.c

So I guess its safe to delete that folder also.... as long as I remember to put the documentation somewhere ?

closed 2019-05-10 by the-snowwhite