machinekit/mksocfpga #18: Is it viable to add (linux) pinmuxing to the mksocfpga ?

the-snowwhite, 2016-05-09:
Quick search Link collection of what seems to be out there and available.

-----
    PINCTRL (PIN CONTROL) subsystem
    This document outlines the pin control subsystem in Linux

    This subsystem deals with:

    - Enumerating and naming controllable pins

    - Multiplexing of pins, pads, fingers (etc) see below for details

    - Configuration of pins, pads, fingers (etc), such as software-controlled
      biasing and driving mode specific pins, such as pull-up/down, open drain,
      load capacitance etc.

https://www.kernel.org/doc/Documentation/pinctrl.txt

    Pin Control Sybsystem – Building Pins and
    GPIO from the ground up

http://dflund.se/~triad/papers/pincontrol.pdf

    Software Reconfigurable Pin Multiplexing

http://www.flex-logix.com/pin-muxing/

    High-level block diagram of Snickerdoodle

    Differentiating the board even more than the number of GPIO is the design choice that allows users to pair Snickerdoodle’s Wi-Fi chip with a cellphone app called Connect. Once paired, Connect facilitates uploading custom field-programmable gate array (FPGA) configuration files to the board, controlling such  operations as GPIO assignments, pin multiplexing, and overall system performance.

    Use a cell phone app to load custom configurations to the snickerdoodle's FPGA

http://usadye.ru/full-desc/aHR0cDovL21ha2V6aW5lLmNvbS8yMDE1LzEwLzEyL3NuaWNrZXJkb29kbGUtZGV2LWJvYXJkLWZwZ2EtYXJtLXByb2Nlc3Nvci98fEdQSU8gQ2FyZHx8MHx8TmV3c3x8MA

http://vytm.in/Lg6JRw#http://makezine.com/2015/10/12/snickerdoodle-dev-board-fpga-arm-processor/

the-snowwhite, 2016-05-25:
Quick search Link collection of what seems to be out there and available.

-----
    PINCTRL (PIN CONTROL) subsystem
    This document outlines the pin control subsystem in Linux

    This subsystem deals with:

    - Enumerating and naming controllable pins

    - Multiplexing of pins, pads, fingers (etc) see below for details

    - Configuration of pins, pads, fingers (etc), such as software-controlled
      biasing and driving mode specific pins, such as pull-up/down, open drain,
      load capacitance etc.

https://www.kernel.org/doc/Documentation/pinctrl.txt

    Pin Control Sybsystem – Building Pins and
    GPIO from the ground up

http://dflund.se/~triad/papers/pincontrol.pdf

    Software Reconfigurable Pin Multiplexing

http://www.flex-logix.com/pin-muxing/

    High-level block diagram of Snickerdoodle

    Differentiating the board even more than the number of GPIO is the design choice that allows users to pair Snickerdoodle’s Wi-Fi chip with a cellphone app called Connect. Once paired, Connect facilitates uploading custom field-programmable gate array (FPGA) configuration files to the board, controlling such  operations as GPIO assignments, pin multiplexing, and overall system performance.

    Use a cell phone app to load custom configurations to the snickerdoodle's FPGA

http://usadye.ru/full-desc/aHR0cDovL21ha2V6aW5lLmNvbS8yMDE1LzEwLzEyL3NuaWNrZXJkb29kbGUtZGV2LWJvYXJkLWZwZ2EtYXJtLXByb2Nlc3Nvci98fEdQSU8gQ2FyZHx8MHx8TmV3c3x8MA

http://vytm.in/Lg6JRw#http://makezine.com/2015/10/12/snickerdoodle-dev-board-fpga-arm-processor/

the-snowwhite, 2017-09-07:
GPIO mux functionality added in this merge:
https://github.com/machinekit/mksocfpga/pull/87
https://github.com/machinekit/mksocfpga/commit/dcacaab2ec7f7c16b874d2acdb40c56e39a29605

dkhughes, 2017-09-07:
Just in case you haven't seen it, but this is working on the Zynq side already. Input/output bus is split at the top level, and I exposed the DDR register. I did it to allow implementation of gated logic on the I/O without using a software component in the HAL since as you mentioned, internal connections must be resolved. See:

https://github.com/machinekit/mksocfpga/blob/master/HW/zynq-ip/hm2_ip_wrap/src/hostmot2_ip_wrap.vhd.in#L110

As a side note, why the push to verilog? The rest of the project is already in VHDL.

the-snowwhite, 2017-09-08:
I have implemented the GPIO mux so that each GPIO destination is controlled via hm2 registers:
https://github.com/machinekit/mksocfpga/blob/master/HW/QuartusProjects/Common/gpio_adr_decoder_reg.sv#L26

This way it is possible to control the routing directly from the hal.

How do you control the routing ?

About language preferences:
I understand how to create functional code in Verilog / Systemverilog. (C-like in my eyes)
VHDL Is only readable for me as the philosophy behind how it is supposed to work is incomprehensive 
for a self learner like me.  :-)

dkhughes, 2017-09-08:
> How do you control the routing ?

I used the DDR registers that are already built in here:

[https://github.com/machinekit/mksocfpga/commit/cd601ac6625114ae2c5d8b6e9e91a7ba3e44e0fc](https://github.com/machinekit/mksocfpga/commit/cd601ac6625114ae2c5d8b6e9e91a7ba3e44e0fc)

Then, in a hal file you use the standard `gpio.NNN.is_ouput` syntax and the hostmot2 driver handled the rest automatically. It was the path of least resistance at the time to splitting the buses and still having HAL control.

the-snowwhite, 2017-09-08:
> cd601ac <https://github.com/machinekit/mksocfpga/commit/cd601ac6625114ae2c5d8b6e9e91a7ba3e44e0fc>

But … this is confusing:

I am talking about routing to from any physical GPIO pin on the HW board.

Are you able to route the fixed outputs of the hm2 cores like PWM og Stepper pins with this scheme ?

dkhughes, 2017-09-08:
Oh, I see what you are saying. To route the pin functions around you would have to add a big mux to all of output pins, then have the selectors based on every available module in the firmware pin descriptor. The DDR could still be used the way I showed for input/output determination and allow connecting the pins to this mux. Man, that would be a big mux though - and one that needs to be optional. Designs like mine are already up against the placement limit and while that functionality would be nice I wouldn't want to buy a larger FPGA for it.

dkhughes, 2017-09-08:
I guess for small designs it would work well - 32IO would be like 32 32->1 muxes connected to the GPIO ddr selection component...I'm running 72IO though currently.

cdsteinkuehler, 2017-09-08:
On 9/8/2017 11:15 AM, dkhughes wrote:
> I guess for small designs it would work well - 32IO would be like 32 32->1 muxes connected to the GPIO ddr selection component...I'm running 72IO though currently.

Yeah, I can see where it might be useful, but it's a lot of logic and
almost anything that doesn't need to be changed live at run-time can
be done with ZERO extra logic by just playing with the FPGA project
and PIN_* files.

It would, however, be nice to have the I/O signals at the top-level of
the project *NOT* be tri-state.  I've run into issues with the passing
of tri-state signals from the top level to the hostmot2 instance,
which some of the FPGA tool chains don't really like.

So I like pushing the I/O buffers and DDR signal to the top level,
just make the "any-to-any" mux part optional.

-- 
Charles Steinkuehler
charles@steinkuehler.net

closed 2023-11-11 by the-snowwhite