machinekit/mksocfpga #19: Missing 'board definition file' for the microzed board

claudiolorini, 2016-05-10:
With a standard Vivado installation, in order to rebuild the FPGA bitstream with the make_bitfile.sh script,
you need to add the correct 'board definition file' for the microzed (or other platform like the Z-turn).

I propose to add a directory
.../mksocfpga/HW/VivadoProjects/board_files/

that will contain the 'board definition' structure for each board 
  microzed_7010
  zturn-7020
  ...

the Vivado init script 'init.tcl' have to address the correct diirectory like this: 
set_param board.repoPaths [list ".../mksocfpga/HW/VivadoProjects/board_files/"]

is that reasonable? is there a better place to place the board configurations?

the-snowwhite, 2016-05-10:
I think there is also a board file concept for Quartus projects.
Sø there could instead be a top level board-files folder with Altera and
Xilinx subfolders

dkhughes, 2016-05-10:
@claudiolorini  You have to do what? You shouldn't have to add any file. make_bitfile is called with the path to the conf file in the project directory of the bitfile you want to build. You shouldn't have had to make any adjustments for the board itself.

dkhughes, 2016-05-10:
The bitfile organization is done in a som/carrier method right now which parallels what we were talking about before in the Altera side. I think the conf files need to go, on both sides (e.g. the .in files). See 
[https://github.com/machinekit/mksocfpga/issues/9#issuecomment-217919870](https://github.com/machinekit/mksocfpga/issues/9#issuecomment-217919870)

for my proposal which doesn't break IP packagers.

nilrods, 2016-05-10:
@claudiolorini I think you referencing the vivado board definition that defines the block design parameters for the zynq boards other than microzed, like memory part, nic settings, etc correct?
If so, I would think there has to be some mechanism for this, whether board files or just tcl scripts to change the parameters for different zynq boards.

Just my thoughts.

dkhughes, 2016-05-10:
@nilrods Yes, that would need to be done by copying the project  and then editing the tcl scripts. I have almost completed the zturn as a second project, I've just been waiting on the board to arrive to debug.

If that is the issue, and not a compilation error with the micro zed project, can we close this issue aND open one that properly reflects missing other zynq boards?

nilrods, 2016-05-10:
@dkhughes From what I read on the initial post on this issue that is what it sounded like Claudio was saying, but I could be way off here. We'll let him chime in.

claudiolorini, 2016-05-10:
hello,
@dkhughes when i try to run:
`
$ ./make_bitfile.sh microzed/microzed_jd2cb/config
`
from a 'standard' Vivado installation (not from a docker container) i obtain the following:

****** Vivado v2015.4 (64-bit)
  **** SW Build 1412921 on Wed Nov 18 09:44:32 MST 2015
  **** IP Build 1412160 on Tue Nov 17 13:47:24 MST 2015
    ** Copyright 1986-2015 Xilinx, Inc. All Rights Reserved.

Sourcing tcl script '/home/cl/.Xilinx/Vivado/2015.4/init.tcl'
source /home/cl/FPGA/fpga-projects/mksocfpga/HW/VivadoProjects/microzed/microzed_jd2cb/scripts/microzed_jd2cb_bp.tcl
....
bla bla bla
....
ERROR: [Board 49-71] The board_part definition was not found for em.avnet.com:microzed_7010:part0:1.0. The project's board_part property was not set, but the project's part property was set to xc7vx485tffg1157-1. Valid board_part values can be retrieved with the 'get_board_parts' Tcl command. Check if board.repoPaths parameter is set and the board_part is installed from the tcl app store.
INFO: [Common 17-206] Exiting Vivado at Tue May 10 19:33:44 2016...

i's complaining about the missing board definition file: 
-->''em.avnet.com:microzed_7010:part0:1.0"
to make it run i had to add the appropriate bdf like you can see here:
https://github.com/claudiolorini/mksocfpga/tree/master/HW/VivadoProjects/board_files
and tell Vivado where to look for bdf in the 
.Xilinx/Vivado/2015.4/init.tcl 

with that modification all seems fine...
C.

claudiolorini, 2016-05-10:
...just to try to clarify things: 
i suppose this could happen if you don't do a "full-optional-yes-fload-my-disk" installation of Vivado.
Available boards in my install is shown here:
[image]
only zedbord is listed. The installation procedure admit installing only board of a choice.
Just to have a consistent system i suggest to add bdf.

btw, @dkhughes your bdf for the microzed is ver. 1.1, but snooping around i've found only 1.0...

dkhughes, 2016-05-10:
Oh, okay, I'm with you. This is not in the docker image. The board definitions are automatically installed by the docker build step. I didn't include them in this project at the time in case they were updated (the installer guaranteed you had the latest version for any bug fixes). *All* of the board definitions from avnet only take a few mbs if I remember correctly.

I will remove the reliance on the board file being present in the tcl tonight when I get a chance.

dkhughes, 2016-05-10:
The installer for the board files used by the docker image should be here:

[http://zedboard.org/sites/default/files/documentations/Avnet%20Zed%20Board%20Definitions%202015_3or4_160331_2.zip](http://zedboard.org/sites/default/files/documentations/Avnet%20Zed%20Board%20Definitions%202015_3or4_160331_2.zip)

claudiolorini, 2016-05-10:
Thanks, i'll get the newest bdfs from there.
relaxing the checks in the scripts should avoid the problem, doesn't it?
in that case we can close!
C.

closed 2016-05-10 by claudiolorini