machinekit/mksocfpga #30: Next Version of ConfigRom - Move to remove firmware specifics from driver layer of hostmot2

dkhughes, 2016-05-22:
Here is an issue to track the feasibility, and necessary firmware changes  to eliminate the firmware specific knowledge held by the hostmot2 drivers. Description of the firmware (e.g., friendly header naming, the number of LEDs, etc.) should be provided by the firmware the same way IO descriptors are provided now. There may be an order required for parsing due to HAL restrictions, but that should be straightforward to implement on the driver side.

As an aside, during integration to avoid breaking the original mesanet boards, we can bump the configrom version of these changes and check before applying the new method of parsing.

mhaberler, 2016-08-05:
I guess this is pretty much done with the fwid proto message?

cdsteinkuehler, 2016-08-06:
Agreed.  This issue is resolved by the now-working firmware-ID ProtoBuf message added to the FPGA logic.

closed 2016-08-06 by cdsteinkuehler