machinekit/mksocfpga #33: Official name for DE0 board

cdsteinkuehler, 2016-05-22:
The names used for the DE0 boards are inconsistent, sometimes the shorter "DE0-Nano" is used, and sometimes the full "DE0-Nano-SoC" is used.  I would like to make usage consistent across the various pin, configuration, and project files, but which should be used?

Technically, DE0-Nano-SoC is more correct (there is actually a DE0-Nano board that *JUST* has an FPGA with no ARM cores), but it seems long, particularly when appending an adapter board ID (ie: DE0_Nano_SoC_DB25).

Thoughts?

If no one has a strong opinion, I'll probably make everything the "long" version, to match the project names under HW/QuartusProjects.

the-snowwhite, 2016-05-23:
I have strongly tried to use the verbose version:
DE0 Nano Soc everywhere to not get this board confused with the Arm/cpu-less version, which would be rather disappointing for a new user attempting to run mksocfpga.

Another option would be to adapt the Rockitboards name Atlas Soc, lhowever that has allways been my second choice...

cdsteinkuehler, 2016-05-23:
On 5/23/2016 1:30 PM, Michael Brown wrote:
> I have strongly tried to use the verbose version:
> DE0 Nano Soc everywhere to not get this board confused with the Arm/cpu-less 
> version, which would be rather disappointing for a new user attempting to run 
> mksocfpga.

Then it's agreed.  I'll update the places where I'm using the short
version to include the full name "DE0_Nano_SoC".

-- 
Charles Steinkuehler
charles@steinkuehler.net

closed 2016-06-03 by cdsteinkuehler