machinekit/mksocfpga #38: Request for review

cdsteinkuehler, 2016-06-03:
I have been working on a scheme to build multiple configurations from a single Quartus project, and would appreciate a review of the general direction I'm heading.  The constraints are:

* Each supported board has it's own directory in the HW/QuartusProjects directory
* Each board has one or more PIN_*.vhd configuration files, found in the hostmot2/config/<board> directory
* Only one Quartus build can be running in a given Quartus project directory at any given time
* Jenkins launches a single Docker instance to build any/all of the various FPGA bit files

So, what I envision is:

* Each project directory has it's own build script that if run without options will build all PIN configurations for that board.  Optionally, specific configuration(s) can be built if specified on the command line.  This covers the "only one build at a time in each project directory" constraint.
* There is a top level Makefile (or build script) that will execute all the lower-level build scripts (possibly in parallel)

I have a preliminary example of a build script for a specific board (the DE0_Nano_DB25) available for review:
https://github.com/cdsteinkuehler/mksocfpga/blob/build-scripts/HW/QuartusProjects/DE0_Nano_SoC_DB25/build.sh

Let me know if anyone sees any glaring issues with this general way forward.  There are some potential issues (like a top-level Makefile that executes multiple project builds in parallel, but each project build might use multiple cores while compiling), but nothing too serious.  If no one sees any significant problems with this scheme, I'll see if I can setup a test Jenkins build that pulls from my repo and builds all the Quartus SoC targets.

mhaberler, 2016-06-04:
no, I do  not see any showstoppers. 

What you could do is start from a  copy of the mksocfpga job (go to https://jenkins.machinekit.io/view/machinekit/job/mksocfpga/ -> Copy project); in theory both mksocfpga and the new copy should trigger on a merge in the gh repo

this job would stop at saving the build artefacts (the downstream mksocfpga-packaging job specifically triggers on mksocfpga build complete only) - I'll look into packaging once those artefacts are usable

the-snowwhite, 2016-06-04:
>    the hostmot2/config/ directory

Sorry for coming in somewhat sideways, however I'm now having results to
share about the hostmot2 pinmux functionality I suggested in #18.
It is now functional and ready for finalisation.
If the enhancement is acceptable then one more bitfile per pinout will no
longer be needed. ..:-)

cdsteinkuehler, 2016-06-04:
On 6/4/2016 6:00 AM, Michael Brown wrote:
> Sorry for coming in somewhat sideways, however I'm now having results to
> share about the hostmot2 pinmux functionality I suggested in #18.
> It is now functional and ready for finalisation.
> If the enhancement is acceptable then one more bitfile per pinout will no
> longer be needed. ..:-)

I have yet to closely review the code, but I have no fundamental
problems with the direction you're taking.  That said, as stated
previously I'm working on getting something setup that's as close to
the existing Mesa infrastructure as possible, both to help with
ongoing tracking of upstream and to (hopefully) confuse end users as
little as possible.

Once that's going (hopefully soon), I suspect the changes you've made
will allow a merge of the two DE0_Nano_SoC projects, reducing the
total number of generated *.rbf files without loosing any functionality.

-- 
Charles Steinkuehler
charles@steinkuehler.net

cdsteinkuehler, 2016-06-06:
The initial version of a master build process has been added.  A top-level Makefile looks for build.sh files in any HW/QuartusProject/<project> directories and executes all that are found.  This allows fine-grain control over how projects handle any required configuration.

closed 2016-06-06 by cdsteinkuehler