machinekit/mksocfpga #41: current firmware packages lack DPLL support

mhaberler, 2016-06-06:
if I am not mistaken, none of the binaries currently produced support DPLL and hence IRQ. 

Any specific reason for this?

This means they are useless for external thread timing.

cdsteinkuehler, 2016-06-07:
On 6/6/2016 5:07 PM, Michael Haberler wrote:
> if I am not mistaken, none of the binaries currently produced support DPLL and
> hence IRQ.
>
> Any specific reason for this?

An artifact of when I copied the project files.  I'll get IRQ support 
added soon-ish.

-- 
Charles Steinkuehler
charles@steinkuehler.net

mhaberler, 2016-06-09:
@cdsteinkuehler @dkhughes:

just verified - socfpga 1.149 DE0_Nano_SoC_DB25.7I76_7I76_7I76_7I76.rbf has the dpll functionality back, IRQ's work as before.

thanks!

closed 2016-06-09 by mhaberler