machinekit/mksocfpga #42: Enhance IDROM to include board specific details

cdsteinkuehler, 2016-06-07:
The existing IDROM contents allow the device driver to auto-configure most everything that changes based on the PIN_* VHDL packages, but there is still a substantial amount of hardware-specific detail required in the driver.  See for example:

https://github.com/machinekit/machinekit/blob/master/src/hal/drivers/mesa-hostmot2/hm2_pci.c#L482-L657

I propose adding these board-level details to the FPGA as a new ModuleIDType which would provide a protobuf message in ROM that could be read by the driver.  The protobuf message would be generated from details provided in the FPGA project, making a single place to store and maintain hardware details for a particular FPGA project.

Thoughts on this and/or suggestions for alternate implementations are welcome, as well as any discussion on the details of exactly what should be included.

mhaberler, 2016-06-08:
here is an example firmware.proto message spec, and a Python example showing how to create and encode such a descriptor, outputting a Intel Hex file:

https://github.com/mhaberler/machinekit/commits/fpga-descriptor  (top two commits)

example run: http://paste.debian.net/725146/

I added this to the machinekit repo for now because the protobuf build support is all in place there 

if this is to your liking, I'll transplant to mksocfpga

cdsteinkuehler, 2016-06-08:
On 6/8/2016 12:22 AM, Michael Haberler wrote:
>
> if this is to your liking, I'll transplant to mksocfpga

That looks great!

I would probably add the board name to the message as well.  The 
filename will include the board name, but if you're just looking at the 
FPGA contents you might not know the filename used to configure the 
FPGA, and I can envision some scenarios where the driver isn't 
configuring the FPGA (ie: FPGA programmed by u-Boot or by hardware at 
power-on) and thus doesn't know what bitfile was used.

Otherwise I think you've got everything from the big case statement in 
the driver, and I don't see anything in the board VHDL Package that 
isn't already in the existing IDROM:

https://github.com/machinekit/mksocfpga/blob/master/HW/hm2/config/DE0_Nano_SoC_DB25/DE0_Nano_SoC_DB25_card.vhd

-- 
Charles Steinkuehler
charles@steinkuehler.net

mhaberler, 2016-06-08:
ok, added that

NB there is no overhead of any unused fields, so feel free to propose anything deemed fit
other than that it's important to:
- not use **required**, only **optional** and **repeated**
- not change existing field numbers out there

so I'll should look into having this integrated and usable from mksocfpga then?

mhaberler, 2016-06-08:
wrt to decoding in hm2_soc_ol, all I need to know is:

- the location offset of the descriptor (length field followed by msg blob)
- if no descriptor is present, the length field should be zero to avoid confusion

cdsteinkuehler, 2016-06-08:
On 6/8/2016 6:13 AM, Michael Haberler wrote:
>
> so I'll should look into having this integrated and usable from mksocfpga then?

I would wait for a bit until we've got something through the full chain.

It seems easy enough to hand-build a message blob for now.  Especially 
given the hassle of updating the Docker build container, which I don't 
think even has python installed.

-- 
Charles Steinkuehler
charles@steinkuehler.net

cdsteinkuehler, 2016-06-08:
On 6/8/2016 6:23 AM, Michael Haberler wrote:
> wrt to decoding in hm2_soc_ol, all I need to know is:
>
>    * the location offset of the descriptor (length field followed by msg blob)
>    * if no descriptor is present, the length field should be zero to avoid confusion

You will get the location of the message via the IDROM, I plan on adding 
it as another GTag value for use in the ModuleIDType array (the part of 
the IDROM the driver processes to find out things like how many stepgen 
and encoder channels are present).

If there is no message, there will be no entry for it in the IDROM.

Is the length field already part of the message (ie: contained in the 
generated *.hex file) or do I need to compute and add it?

-- 
Charles Steinkuehler
charles@steinkuehler.net

mhaberler, 2016-06-08:
ah perfect

the hex file already has length/value: https://github.com/mhaberler/machinekit/blob/fpga-descriptor/src/machinetalk/firmware-tag/generate.py#L42-L46
you want to fix the loadaddr I guess, I made one up (in case you use the .hex; the .bin is just the length/blob bits without load address I guess)

mhaberler, 2016-06-08:
btw - since I'm using max_size and max_count in the proto file, there is an upper bound of the underlying message size and it is a #define in machintalk/generated/firmware.npb.h:

`````
/* Maximum encoded size of messages (where known) */
#define pb_Connector_size                        87
#define pb_Firmware_size                         3069  
                                              // ^--- that's the space you'd need 
                                              // to be prepared for
                                              // add the length word (4bytes)!
````

cdsteinkuehler, 2016-06-08:
On 6/8/2016 10:13 AM, Michael Haberler wrote:
> 
> you want to fix the loadaddr I guess, I made one up

The address should start at zero, not 0x400, and the word width should
be 32-bit.  But both of those are minor concerns, I can fixup things
manually if needed for now, and I won't know all the requirements until
I actually get something compiling.

The large size is somewhat concerning but can be dealt with.  It would
probably be better to define settings so we're either under 2-KByte or
make effective use of a 4-KByte ROM.

-- 
Charles Steinkuehler
charles@steinkuehler.net

cdsteinkuehler, 2016-08-06:
As with issue #30 this issue is resolved by the recently added and now working Firmware-ID ProtoBuf message structure added to the FPGA logic.  Any missing or incorrect protobuf message structures in the various FPGA designs should be tracked as new issue(s), if needed.

closed 2016-08-06 by cdsteinkuehler