machinekit/mksocfpga #46: Altera + 7i76: encoder,  gpio: unable to get to work

mhaberler, 2016-07-03:
hi @cdsteinkuehler - here at electrolab.fr helping Goldo to set up an Altera DE0 with an 7i76 card, steppers, and an encoder

I'm using the 7i76x4 firmware (see branch https://github.com/mhaberler/machinekit/commits/goldo)

I can generate step/dir signals just fine and move the motor
we tried to add the spindle encoder and follow the 7i76 manual - pretty sure we hit the right pins (encoder jumpered for ttl input, tb3 pins 7+10 for A/B) - halshow shows no change on input-a/input-b)

same thing for gpios 

either I have a blunder in the config, I completely misunderstand the pin numbering, or there's an issue with the firmware

I will try to reproduce at home (back Jul13 so no hurry)

cdsteinkuehler, 2016-07-03:
Thanks for the report.  There could be any of a number of things
wrong, but most should probably be easily fixable.

I'll review the pinout and configuration to see if I notice any goofs.

-- 
Charles Steinkuehler
charles@steinkuehler.net

mhaberler, 2016-08-05:
done, issue sorted

closed 2016-08-05 by mhaberler