machinekit/mksocfpga #47: Where can I find development setup instructions?

luminize, 2016-07-07:
Hi guys,

I've been looking here and there of instructions/info if there a a latest image (like RCN's generated images) available, or concise instructions on how to use theDe0-nano-soc with the mesa net 7i76 card.
I know mhaberler has made one in the past, and I can use that one, but I'm clueless if that's still up to date.

I'm ok with installing this myself (jessie, apt-repo etc), but I would like to know if I should deviate (and where) from the machinekit.io/docs instructions.

hedgehog, 2017-04-13:
@luminize Hope this helps - buyer beware - I haven't used these in anger but they are what I've discovered so far.  this will also serve as a place holder for me:
http://blog.machinekit.io/2016/11/de0-nano-soc-update-on-sd-card-images.html
and
http://www.machinekit.io/docs/man/man9/hm2_soc_ol/
and
https://gist.github.com/mhaberler/89a813dc70688e35d8848e8e467a1337

the-snowwhite, 2019-05-09:
I would very much like to be able to close this thread sucessfully and have my notes ready now.
I have a plugin called   Asciidoctor.js Live Preview, this however does not seem to render the asciidoc files in my fork of mkdocs correctly. 
@luminize 
@ArcEye 
@cdsteinkuehler 
Now I am currently trying to wrap my head around using these instructions to see if I can get able to preview the docs I submit and maybe edit and add pages locally via the edit this page button:
https://gist.github.com/mhaberler/ede68eae275c38a5364734c065a1b9c0

Im currently pulling the docker image (seems ok), however I do't yet get these lines:

    review and run the serve.sh script in the toplevel directory of your website
    review the values in config_preview.yml which were generated by the first run

And have  also not figured out how to run this on my local clone of 
https://github.com/the-snowwhite/machinekit-docs

luminize, 2019-05-10:
It’s been some months since i ran it locally. I’m not at my laptop right now, so i’ll get back to you on how i run it.

ArcEye, 2019-05-10:
You need a clone of machinekit-docs.

Then install jekyll and gem

From the root dir of your clone run `jekyll serve`
The first time round it will error at every missing gem, which you need to install with
`gem install  xxxx.gem`
When you finally satisfy the deps, it will build and  serve the website with the below banner.
```
Auto-regeneration: enabled for '/usr/src/machinekit-docs'
    Server address: http://127.0.0.1:4000//
  Server running... press ctrl-c to stop.
```

The asciidoctor plugin is useful for a single page, but running through jekyll you get a better overview.

the-snowwhite, 2019-05-11:
@ArcEye thanks I got that to work, and now have a getting started guide ready except for being able to link to the in guide mentioned actual sd-card images.
I have created 2 new updated Stretch images:
1 console (512MB) and 1 desktop (1.1GB) version covering all the supported DEx(x) boards.
@cdsteinkuehler 
@luminize 
Instead of linking to the (obscured) google drive folder I would like to ask/request if its possible to place them in a new folder here:
http://deb.machinekit.io/uploads/de0-nano/   ?

the-snowwhite, 2019-05-11:
This is where I would place the link to the images:
https://github.com/the-snowwhite/machinekit-docs/commit/6d9616cbb4c7489c2ca351b34cb686ae1be07414#diff-495ababac5a198a7ba7c43dd23f5494dR42

ArcEye, 2019-05-11:
@the-snowwhite 
I have access to the repo and all the ancilliary dirs etc.

If you pass me a link to the images, I will download them and put them in a dir called 
http://deb.machinekit.io/uploads/de10-nano/
otherwise they will get confused with the earlier ones.

When that is confirmed, you can amend your links and send a PR for your docs

the-snowwhite, 2019-05-11:
Great
It came to my mind that github now has a release feature so I have put up the files here:
https://github.com/the-snowwhite/soc-image-buildscripts/releases/tag/2.1
I'm a bit ambilalent now as of where it would be best to link to from MKDocs ?
Also it would be great to get some external feedback on testing of the new sd-images 
best before they are uploaded to: http://deb.machinekit.io/uploads/de10-nano/
...

ArcEye, 2019-05-11:
OK, I can't test the images however, only have a DE0.

If you do a post to the forum asking for tester(s) for the images, once done you can advertise
the presence of them under uploads.

I will put them there tomorrow, but leave you to do the advertising and push your PR with the link to them when you are happy with it all.
I will also look at the docs in your branch and think about where any internal links need to go so that they are prominent.
Cheers

the-snowwhite, 2019-05-11:
Great

ArcEye, 2019-05-12:
@the-snowwhite 
Everything is uploaded with a brief README file
http://deb.machinekit.io/uploads/de10-nano/

Regards the docs entries.
I think it needs a separate entry for BBB images and DEx(x) images.
This will greatly assist navigation as currently images are mentioned without any reference as to what they are for.

In docs/getting-started,  change single link to 
```
=== Images

link:../machinekit-images[Machinekit Images for BeagleBoneBlack]

link:../machinekit-de10-images[Machinekit Images for DE0/DE10-NANO-Soc boards]
```

Then create new file `machinekit-de10-images.asciidoc` to contain your additions currently appended to `machinekit-images.asciidoc`
Might be as well to include the last para from `machinekit-images.asciidoc`, re writing to SD card, in your doc also.

I think doing this will remove the need for further links, as it is big and bold on the getting started page.

the-snowwhite, 2019-05-12:
OK , I have now tested both the DE0 and DE10, console and desktop sd-images and adjusted the initial getting started instructions accordingly:
https://github.com/the-snowwhite/machinekit-docs/tree/DEx_work
forum thread is here:
https://groups.google.com/forum/#!topic/machinekit/R2vzsY4OkNI
 everything still looks fine from here :-)

the-snowwhite, 2019-05-12:
Thanks
OK
Got the new Toc pushed

the-snowwhite, 2019-05-12:
I will make the PR now

the-snowwhite, 2019-05-12:
https://github.com/machinekit/machinekit-docs/pull/304

the-snowwhite, 2019-05-12:
Now looking at the top of this thread:

> how to use theDe0-nano-soc with the mesa net 7i76 card

@cdsteinkuehler 
@luminize 
there is still more to do ?

the-snowwhite, 2019-05-12:
Lookin at @mhaberler 's:
https://gist.github.com/mhaberler/89a813dc70688e35d8848e8e467a1337
For my own development work:
As the first full build is very slow on the armhf cores,
I havn't been using rip install for my development work the last 2 years
Instead I have been using:

    docker pull dovetailautomata/mk-cross-builder:armhf_9
    cd machinekit
    export JOBS=32
    scripts/build_docker -t armhf_9 -c deb

This used to build the armhf .deb packages in around 2 minutes or so, then it took about 1 min more to install them manually...
I have no idea if this method has been updated/adapted to the new mk-hal /mk-cnc repos ?

ArcEye, 2019-05-13:
> I have no idea if this method has been updated/adapted to the new mk-hal /mk-cnc repos ?

We have been using @zultron s mk-cross-builder dockers for quite some while now :+1:

the-snowwhite, 2019-05-13:
Argument for making a case for recommending @zultron s mk-cross-builder dockers instead of rip builds:

Buster:
First build: (downloads the docker image)

    /machinekit-hal$ time scripts/build_docker -t armhf_10 -c deb
    real    2m35,537s
    user    0m0,343s
    sys     0m0,221s

Following builds:

    /machinekit-hal$ time scripts/build_docker -t armhf_10 -c deb
    real    1m20,881s
    user    0m0,144s
    sys     0m0,176s

Stretch:
First build: (downloads the docker image)

    /machinekit-hal$ time scripts/build_docker -t armhf_9 -c deb
    real    3m24,621s
    user    0m0,362s
    sys     0m0,205s

Following builds:

    /machinekit-hal$ time scripts/build_docker -t armhf_9 -c deb
    real    1m49,291s
    user    0m0,141s
    sys     0m0,182s

the-snowwhite, 2019-05-13:
Here are the remaining notes I had when I began this documentation process:
https://gist.github.com/the-snowwhite/65ede1ec7a4530035164efcaa27df3a2

Sparked by this thread in the forum:
https://groups.google.com/forum/#!topic/machinekit/eVhvTnuhblE