machinekit/mksocfpga #48: How to make custom pin layout firmware files

luminize, 2016-07-07:
Last year I have set up (on a windows pc) the toolchain for writing a custom .BIT file for a mesa 5i20 card. (I experimented with bspi iirc). Would it be an idea to have a docker container or vagrant setup so it's easy to change functions and pins (let's say "configuring" for the uninitiated) for cards?

dkhughes, 2016-07-07:
I have a Vivado 2015.4 based docker image already available here:

[https://github.com/dkhughes/vivado-docker](https://github.com/dkhughes/vivado-docker)

and I'm pretty sure there is an altera based one as well. If you want to change bitfiles for the older Spartan based cards, you will probably need an ISE based setup, and I haven't tried to dockerize that yet. 

As soon as I finish the zturn stuff I am planning on updating the Vivado image to the latest version which has a supposedly working web downloader, and no requirement for a license file anymore. I can maybe take a look at doing an ISE image as well.

For the SOC's changing a config is basically down to editing the relevant board file. We do really need to document the format of the module descriptor and the pin struct in that file though. Shoot, I've made 3 custom pin files now and I still forget how it's formatted.

luminize, 2016-08-07:
Hi, I've built custom pin layouts thru a jenkins job. see this thread: https://github.com/machinekit/mksocfpga/issues/55 and https://jenkins.machinekit.io/view/userjobs/
Thanks!

closed 2016-08-07 by luminize