machinekit/mksocfpga #52: handling of Firmware ID protobuf message unclear

mhaberler, 2016-07-17:
continuation of discussion in https://github.com/machinekit/mksocfpga/pull/51 (PR is the wrong place); intended to hash out the logic how this descriptor is handled.

where we are:
- we can encode a protobuf message in the FPGA firmware, accessible through a Gtag ID (F8) or at a well-known address (0xF800)
- the first word at 0xF800 is the length field (u32). If zero, no fwid message is present. @cdsteinkuehler _is this a safe assumption to make?_
- we can read and decode this message in hostmot2.c
- if the length field is unreasonable, or the protobuf decoding fails, a message should be logged
- the values in this message are intended to default the hard-coded values in hm2_llio as [set here](https://github.com/machinekit/machinekit/blob/master/src/hal/drivers/mesa-hostmot2/hm2_soc_ol.c#L401-L406):

````
    brd->llio.num_ioport_connectors = 2;
    brd->llio.pins_per_connector = 17;
    brd->llio.ioport_connector_name[0] = "P3";
    brd->llio.ioport_connector_name[1] = "P2";
    brd->llio.fpga_part_number = 0; // "6slx9tqg144";
    brd->llio.num_leds = 2;
`````

requirements:
- any existing drivers must continue to work unchanged.
- there will be firmware blobs which have that fwid message, and some may not (or not yet). Handle both cases gracefully
- especially during development it makes sense to override any fwid message in the FPGA image (or pretend it was there). This can be done by passing a 'fake' encoded message+length up from llio.

my current idea of a flow hence is:
- if the message length field is unreasonable, or the protobuf decoding fails, a message should be logged and llio values left unchanged but operation continues (in case there are random values in the msg length location)
- if a fwid proto message is present (either in the FPGA image, or passed 'up' via the llio driver), override `brd->llio` values `num_ioport_connectors pins_per_connector ioport_connector_name.. fpga_part_number  num_leds`
- applying the fwid proto message can be skipped by the "nofwid" keyword in the config string, in which case the values passed up from the llio driver remain untouched; this keyword sets the new field `hostmot2_t.config.skip_fwid`
- else leave values in `brd->llio` alone
- a llio driver supporting a 'fake' fwid message will take that from a file, as directed by the "descriptor=<fwidmsg.bin>" flag - this file contains the override message
- llio "passes up" this fwid message in the new struct hm2_lowlevel_io_struct fields `fwid_msg` and `fwid_len`
- hm2_register() inspects the `fwid_msg` and `fwid_len` fields and - if non-zero - takes this message instead of the one from the FPGA image.

mhaberler, 2016-07-17:
to robustify, I think we should change the protobuf message layout in memory as follows:

0xf800 - 'FWID'  - this is a cookie
0xf804 - length field
0xf808....max - actual message

I'll post a changed mif file and rebuild

mhaberler, 2016-07-17:
building the cookie version now: https://github.com/machinekit/mksocfpga/commit/06ad3ae27510b9bfa425a5024331bbc923ddddc2

cdsteinkuehler, 2016-07-17:
On 7/17/2016 5:35 AM, Michael Haberler wrote:
> continuation of discussion in #51
> <https://github.com/machinekit/mksocfpga/pull/51> (PR is the wrong place);
> intended to hash out the logic how this descriptor is handled.
>
> where we are:
>
>   * we [...]

Probably, but it's not really a safe way to code.

I do not understand the chicken-and-egg problem.  The only thing
really required to read the message is physical access to the hm2
register space (ie: the card got mmapped and SW can do reads), and the
mmap() is already happening in hm2_soc.c.

The process should be something like:

* mmap() card into address space (details via device tree)

* Verify hostmot2 "cookie"

* Scan IDROM for new FWID tag

* Read and process FWID message

* Populate board-> as appropriate

* Call hm2_register()

I realize this might duplicate a bit of code between the low-level
hm2_soc.c and hostmot2.c, but it's not much code and could be factored
out into a common file if you're that worried about it.

...actually, it looks like it's pretty much already there:

https://github.com/machinekit/machinekit/blob/master/src/hal/drivers/mesa-hostmot2/hm2_soc.c#L325-L346

Now all you need is a "get_module_descriptor()" function, that can be
easily written based on hm2_read_module_descriptors():

https://github.com/machinekit/machinekit/blob/master/src/hal/drivers/mesa-hostmot2/hostmot2.c#L664-L741

...then you've got a reliable way to find whether or not there is a
protobuf message available, and it's starting address.  You can
process the message and fill out the llio settings between mmap'ing
the FPGA and calling hm2_register:

https://github.com/machinekit/machinekit/blob/master/src/hal/drivers/mesa-hostmot2/hm2_soc.c#L394-L399

-- 
Charles Steinkuehler
charles@steinkuehler.net

cdsteinkuehler, 2016-07-17:
On 7/17/2016 6:45 AM, Michael Haberler wrote:
> building the cookie version now: 06ad3ae 
> <https://github.com/machinekit/mksocfpga/commit/06ad3ae27510b9bfa425a5024331bbc923ddddc2>

WARNING:
When I used the MIF file you created previously, Quartus complained
that it was broken.  I had to add one to the depth field to make
Quartus happy.  In your new file, depth is listed as 43, but there are
actually 44 values (0x00 to 0x2b, inclusive).

I'm not sure if that will cause the firmware build to fail, but it might.

-- 
Charles Steinkuehler
charles@steinkuehler.net

mhaberler, 2016-07-17:
yes, I noticed this bug in mif.py - fixed in https://github.com/machinekit/mksocfpga/commit/50e25205df297eab2a22bbb2cca088a7f86c501f

dkhughes, 2016-07-17:
I'm late to the party, but what if we just changed the register function a little in hostomot2.c I think the only required thing in the fpga firmware struct is the number and names of the gpio headers. Can we simply enforce an order, i.e. read the firmware stuff first? Whatever check hostmot2 is doing of the firmware struct can then happen afterward, and if we didn't find a firmware struct in the rom, we fall back on enforcing a prepopulated one.

Having the LLIO aware of the contents of the firmware breaks the encapsulation of the read write interface from the functionality section of the driver.

mhaberler, 2016-07-17:
>I do not understand the chicken-and-egg problem. ...
The process should be something like:

>* mmap() card into address space (details via device tree)

>* Verify hostmot2 "cookie"

>* Scan IDROM for new FWID tag

>* Read and process FWID message

>* Populate board-> as appropriate

>* Call hm2_register()

evidently ;) the current flow is like so:

- you can mmap only after you know the FPGA is loaded
- mapping is NOT done directly by the llio driver, but as a _consequence_ of hm2_register() so at the time you are proposing to do things the FPGA memory space is simply not available
- hm2_register()  parses the config string and only THEN decides to call back into llio.program_fpga to load the dtbo which eventually loads the fw - if a firmware= string was given in config=
- only after that succeeds the first llio.read() or lllio.write() operation [does the actual mmap](https://github.com/machinekit/machinekit/blob/master/src/hal/drivers/mesa-hostmot2/hm2_soc_ol.c#L192-L206).
- moving the dtbo load down to llio.rtapi_app_init() was something we already had, and it was not a good idea because it would have created a special case over all other hm2 drivers wrt loading syntax - you cant use config="..." to pass the dtbo

This means the fwid read must be interleaved into hm2_register() at a point where the FPGA is loaded and it is safe to do a llio.read() but no references to the llio fields have been made yet - this is [here](https://github.com/machinekit/machinekit/blob/master/src/hal/drivers/mesa-hostmot2/hostmot2.c#L1300)

dont worry, it will work ;)

dkhughes, 2016-07-17:
@mhaberler Yes, I think you're definitely on the right track. This is the check that fails loading the fpga with an "incorrect" firmware. We can move it down after the module descriptors are read in a few lines later.

[https://github.com/machinekit/machinekit/blob/master/src/hal/drivers/mesa-hostmot2/hostmot2.c#L1366](https://github.com/machinekit/machinekit/blob/master/src/hal/drivers/mesa-hostmot2/hostmot2.c#L1366)

cdsteinkuehler, 2016-07-17:
On 7/17/2016 8:43 AM, Michael Haberler wrote:
> 
>   * hm2_register() parses the config string and only THEN decides to call back
>     into llio.program_fpga to load the dtbo which eventually loads the fw - if a
>     firmware= string was given in config=

Yeah, sorry.  That's me reading C code.  I read this line:

https://github.com/machinekit/machinekit/blob/master/src/hal/drivers/mesa-hostmot2/hm2_soc.c#L394

...as "call hm2_soc_program_fpga" rather than "copy the magic function
address".  :-/

I get = and == mixed up a lot too.  ;-)

> This means the fwid read must be interleaved into hm2_register() at a point
> where the FPGA is loaded and it is safe to do a llio.read() but no references to
> the llio fields have been made yet - [...]

Yeah, lots of ways around this, but I'd probably add a "process fwid"
function to llio, then look for and optionally call it around line
1383 in hostmot2.c

-- 
Charles Steinkuehler
charles@steinkuehler.net

mhaberler, 2016-07-17:
@dkhughes - re ordering - I guess that's where I'll wind up

actually there's another such ordering dependency already there, [see this ](https://github.com/machinekit/machinekit/blob/master/src/hal/drivers/mesa-hostmot2/hostmot2.c#L847-L876) - HM2_GTAG_IOPORT is parsed before all others in case other modules depend on its values

mhaberler, 2016-07-17:
patchup time, this approaches working state. I'll post a rough first cut later, need to run.

the custom message override works, as does setting of the llio data fields such that hm2_register() doesnt trip 

on the firmware.proto file - I guess I'll reshuffle a bit since the assumption seems to be all connectors have the same number of pins, so the Connector.pins member doesnt make sense and should move to Firmware.pins

mhaberler, 2016-07-17:
curious - 5i25 has two leds on board so num_leds == 2

the de0 nano has no such leds, rather a gazillion others - so rather num_leds == 0?

now there are two leds on @cdsteinkuehler's de0-nano adapter board - are these supposed to be tied to CR01 and CR02 ?

cdsteinkuehler, 2016-07-17:
On 7/17/2016 11:35 AM, Michael Haberler wrote:
> 
> on the firmware.proto file - I guess I'll reshuffle a bit since the assumption 
> seems to be all connectors have the same number of pins, so the Connector.pins 
> member doesnt make sense and should move to Firmware.pins

Please keep the pins-per-connector generic.  I have been considering
making a board that has both 50-pin and 25-pin connectors, and there's
also the possibility to have a custom board with a variety of
connector types/sizes (for instance one for step/dir pins, one for
encoder inputs, and one for GPIO).

There's already enough assumptions and hard-coded stuff in the driver,
let's not make any more!

-- 
Charles Steinkuehler
charles@steinkuehler.net

cdsteinkuehler, 2016-07-17:
On 7/17/2016 11:42 AM, Michael Haberler wrote:
> curious - 5i25 has two leds on board so num_leds == 2
> 
> the de0 nano has no such leds, rather a gazillion others - so rather num_leds == 0?
> 
> now there are two leds on @cdsteinkuehler <https://github.com/cdsteinkuehler>'s 
> de0-nano adapter board - are these supposed to be tied to CR01 and CR02 ?

"LED" in this instance means LED outputs from the hm2 instance.

The DE0_Nano_SoC_DB25 should have 4 LEDs, two on each GPIO header.

-- 
Charles Steinkuehler
charles@steinkuehler.net

mhaberler, 2016-07-17:
@cdsteinkuehler ok will leave firmware.proto as is, and take pins_per_connector from Connector[0].pins.

I just noticed that hm2_idrom_t just has a single field [port_width](https://github.com/machinekit/machinekit/blob/master/src/hal/drivers/mesa-hostmot2/hostmot2.h#L203) which [seems to apply to all io ports](https://github.com/machinekit/machinekit/blob/master/src/hal/drivers/mesa-hostmot2/hostmot2.c#L610), so I did not see how a board with different numbers of pins per connector can be described by the current hm2_idrom_t struct

cdsteinkuehler, 2016-07-17:
On 7/17/2016 2:59 PM, Michael Haberler wrote:
> @cdsteinkuehler <https://github.com/cdsteinkuehler> ok will leave firmware.proto
> as is, and take pins_per_connector from Connector[0].pins.
>
> I just noticed that hm2_idrom_t just has a single field [...]

Currently, it can't.  But see Peter's comments about the upcoming
IDROM V4, and IMHO we should not design ourselves into an artificial
constraint with the protobuf message format.

...because in the future we may have boards with different connectors,
and there's very little reason not to allow this in the message.  I
don't think a few extra integer values will significantly eat into the
2KB maximum message size.  :)

-- 
Charles Steinkuehler
charles@steinkuehler.net

mhaberler, 2016-07-17:
today's yield:  https://github.com/machinekit/machinekit/pull/1008

mhaberler, 2016-07-19:
I changed mk/mksocfpga and https://jenkins.machinekit.io/view/machinekit/job/mksocfpga/configure to generated a per-config MIF file.

The idea is to have a HW/firmware-tag/%CONFIG%.py file which returns the protobuf message as needed, example [7I76_7I76_7I76_7I76.py](https://github.com/machinekit/mksocfpga/blob/773b1e546fc20c3c86f85f27d4d399d34c537687/HW/firmware-tag/7I76_7I76_7I76_7I76.py
) 

[build.sh calls upon genfwid.py with the config name as argument](https://github.com/machinekit/mksocfpga/commit/8cba30adcea94601eaaf32b151fa23ff429a6413) to generate the MIF file

dkhughes, 2016-07-19:
Ok, I'm working on getting this into the zynq world. Am I right that the fwid stuff is basically a rom? Is it synthesized or specific to Altera? I guess that the mif file is then used as the initialization file that is compiled into the bitstream? I want to make the zynq side match the altera side but some of the choices so far have not been xilinx tools friendly, and I'm hoping I can keep inline without too much hoop jumping.

mhaberler, 2016-07-19:
yes, just a ROM.  The contents are not specific to Altera as it just communicates values to fill in in the llio struct; the MIF file format might be.

In case the Xilinx software does not understand MIF files, you'll have to modify for the Xilinx format as needed here: https://github.com/machinekit/mksocfpga/blob/master/HW/firmware-tag/genfwid.py#L42-L45

btw happy to take instructions how to build Xilinx firmware debs - the Docker container is in place on jenkins.machinekit.io!

mhaberler, 2016-07-20:
btw.. documenting a protobuf message works nicely:  [proto source](https://github.com/machinekit/machinetalk-protobuf/blob/master/src/machinetalk/protobuf/firmware.proto)   yields [documentation](http://www.machinekit.io/docs/machinetalk/protobuf/#firmware-proto)

great job, @ArcEye !

cdsteinkuehler, 2016-07-20:
On 7/19/2016 3:31 PM, dkhughes wrote:
> Ok, I'm working on getting this into the zynq world. Am I right that the fwid
> stuff is basically a rom? Is it synthesized or specific to Altera? I guess that
> the mif file is then used as the [...]

Sorry about using the MIF file and the Altera ram/rom megafunction.

It's fine to infer this ROM, you'll just need the python code tweaked to 
generate something you can pull into a vhdl file.  I used the mif file 
because there were already python solutions for creating hex and mif 
files and I'm used to using the Altera RAM functions (that apparently 
don't exist in Xilinx-land...apparently you're supposed to infer 
everything?!?).

I agree it would be good for the Xilinx and Altera to match as much as 
possible.  If an inferred solution gets implemented for Xilinx, I can 
change the Altera side over to match.

-- 
Charles Steinkuehler
charles@steinkuehler.net

mhaberler, 2016-07-24:
[here is a ROM ](http://git.linuxcnc.org/gitweb?p=hostmot2-firmware.git;a=blob;f=src/sine16.vhd;h=6335e7062577e9404530946ba89eab7a6fa97507;hb=9106c721b44b76fd5b99bf98062a75dfdb28cf20)by Peter, ok to rehash that?

cdsteinkuehler, 2016-07-26:
On 7/24/2016 10:34 AM, Michael Haberler wrote:
> here is a ROM 
> <http://git.linuxcnc.org/gitweb?p=hostmot2-firmware.git;a=blob;f=src/sine16.vhd;h=6335e7062577e9404530946ba89eab7a6fa97507;hb=9106c721b44b76fd5b99bf98062a75dfdb28cf20>by 
> Peter, ok to rehash that?

That doesn't have quite the proper interface for the register I/O and
to be instantiated in Altera block ram (at least the address needs to
be registered).

If you want to infer a ROM for the Firmware ID message, the main thing
is generating the constant array.  I can add the bits of VHDL you need
to make it a ROM, or the constant could go in a stand-alone package
(like the PIN_*.vhd files) and get imported by the ROM code.

-- 
Charles Steinkuehler
charles@steinkuehler.net

dkhughes, 2016-07-27:
I've been reading about this a little bit. Looks like we should be able to synthesize a rom that will work on both Altera and Xilinx devices. If I understand it correctly, we can also directly load a mif file using straight VHDL. I will experiment and report back as I finish catching the Xilinx side up to the new build system.

cdsteinkuehler, 2016-11-01:
Implemented by @dkhughes in #62

closed 2016-11-01 by cdsteinkuehler