machinekit/mksocfpga #53: firmware snag sheet: DE0_Nano_SoC_DB25.7I76_7I76_7I76_7I76

mhaberler, 2016-07-18:
I think we should document experience on a per-firmware basis, so starting here. Documenting the current status of exploring this fw version.

configuration:

- one Charles' adapter board 'DE0-Nano-DB25 V1.0' on gpio0
- one 7i76 connected to P2 of same
- mk: my firmware-proto-dev branch https://github.com/mhaberler/machinekit/commit/49acfec7bfc9f88d721d9a3c7c95528b1eef331b
- [hal config
](https://github.com/mhaberler/machinekit/blob/49acfec7bfc9f88d721d9a3c7c95528b1eef331b/configs/hm2-soc-stepper/irqtest.hal)
- [ linuxcnc.log](http://static.mah.priv.at/public/irqtest.log)
- following http://www.mesanet.com/pdf/parallel/7i76man.pdf
- 12V 'field power', 5V supply via DB25 connector (4.75V at TB2-1 TB2-6)
- the config uses hm2 IRQ but this should not matter
- W4, W5, W6 jumpered towards DB25 (single-ended operation)

configuration:
- 5 stepgens
- 1 encoders

what I am not sure about: the hm2 config line, in particular the `sserial_port=` syntax, is this correct?

` loadrt hm2_soc_ol config="firmware=socfpga/dtbo/DE0_Nano_SoC_DB25.7I76_7I76_7I76_7I76.dtbo num_stepgens=5 num_encoders=2 sserial_port_0=00xxxxxx"   timer1=211812352 `

what works:
- the stepgens create step+/- and dir+/- signals and their  on TB2 and TB3(2-5) outputs as documented
- encoders: fed by 500Hz A/B signal from Rigol 1022U, 5V, 90deg shift, to pins TB3 - ENCA+ ENCB+
- gives velocity 2000; shifting the phase from -90 to +90 deg gives velocity -2000 - looks reasonable
- increase frequency to 100kHz - works fine (velocity 40000/-40000)
- increase frequency to 650kHz - works fine
- stops working around 700kHz.

Yee-haw!

what does not work yet:

- gpios - could be my problem understanding the 7i76 manual vs HAL pin naming
manual says:
````
TB6 CONNECTOR PINOUT
TB6 PIN I/O TB6 PIN I/O
1 INPUT0
`````
I assume TB6-1 corresponds to hm2_de0n.0.gpio.000.in ? help me across the street ;)
hm2_de0n.0.gpio.000.in shows noise.

- leds on DPIO adapter board (this fwid has num_leds = 2 so I'll fix it to have 4, but I guess the ones on GPIO0 would be CR01/CR02 regardless) - should blink as connected to clock signal, no reaction

- this image seems lack any pwmgen functions? (log: `Jul 18 11:51:35 mksocfpga msgd:0: hal_lib:1649:rt hm2/hm2_de0n.0: PWMGen: 0`)

cdsteinkuehler, 2016-07-18:
On 7/18/2016 8:05 AM, Michael Haberler wrote:
> what does not work yet:
>
>   * gpios - could be my problem understanding the 7i76 manual vs HAL pin naming
>     manual says:
>
> |TB6 CONNECTOR PINOUT TB6 PIN I/O TB6 PIN I/O 1 INPUT0 |
>
> I assume TB6-1 [...]

The I/O signals on TB5 and TB6 of the 7i76 are connected to the
sserial port.  If that's not working, you won't get anything in HAL.

The hm2_de0n.0.gpio.000.in is a physical pin on the FPGA (not on the
sserial link).  This pin looks like it's a direction output:

https://github.com/machinekit/mksocfpga/blob/master/HW/hm2/config/DE0_Nano_SoC_DB25/PIN_7I76_7I76_7I76_7I76.vhd#L113

...so I'm not sure why you're seeing noise.  I would expect it to
mirror the state of the direction pin, but I haven't crawled through
the GPIO logic in the hostmot2 vhdl source.

Regardless, if you get the sserial port working, you should be able to
talk to the 7i76 I/O pins.

-- 
Charles Steinkuehler
charles@steinkuehler.net

luminize, 2016-07-18:
> On 18 Jul 2016, at 15:42, cdsteinkuehler <notifications@github.com> wrote:
> 
> Regardless, if you get the sserial port working, you should be able to
> talk to the 7i76 I/O pins.

@mhaberler I remember trying for 4 hours, pulling my hears, grinding my teeth, and other general discomfort to find why the GPIO’s weren’t working. If i’m correct you should make sure you have field power on TB1 before starting MK. If you lose field power, then you’ll also lose the GPIO pins via sserial.

mhaberler, 2016-07-19:
looking at the wrong pins.. it's been a long time.
from reading log and code, the smartserial code is definitely working, it recognizes the 7i76's (now 2 connected)

I guess the correspondence is:

- TB6 1..16 (INPUT0..15)  == hm2_de0n.0.7i76.0.0.input-00..15
- TB5 1..16 (INPUT16..31) == hm2_de0n.0.7i76.0.0.input-16..31
- TB6 17..24 (OUTPUT0..OUTPUT7) == hm2_de0n.0.7i76.0.0.output-00..07
- TB5 17..24 (OUTPUT8..OUTPUT15) == hm2_de0n.0.7i76.0.0.output-08..15

mhaberler, 2016-07-19:
@luminize FYI: the sserial.. param in loadrt hm2_soc_ol is NOT needed - I have:

`loadrt hm2_soc_ol config="firmware=socfpga/dtbo/DE0_Nano_SoC_DB25.7I76_7I76_7I76_7I76.dtbo num_stepgens=5 num_encoders=2 " `

mhaberler, 2016-07-19:
re maximum values - just scoped the stepgen at 3.6Mhz (100ns steplen/stepspace).

that should do for an end to softstepping ;)

luminize, 2016-07-20:
one can do some serious machine control for 20 steppers, 4 encoders and 192 IO... Great!

luminize, 2016-07-20:
I just this afternoon replicated @mhaberler his setup from image.
Self-soldered @cdsteinkuehler adapter board with 1 7i76 board. Great work guys!

mhaberler, 2016-07-20:
I have updated the FirmwareID message in 7I76_7I76_7I76_7I76 config to num_leds=4

cdsteinkuehler, 2016-07-20:
On 7/20/2016 3:18 PM, Michael Haberler wrote:
> I have updated the FirmwareID message in 7I76_7I76_7I76_7I76 config to num_leds=4

It may be non-obvious, but the num_leds for the 7I76_7I85S_GPIO_GPIO 
config should also be four.  The assumption is a DB25 adapter card is 
used for both GPIO0 and GPIO1 on the DE0-Nano board, so there are a 
total of four DB25 connectors and four LEDs.  Even if two of the DB25 
connectors are just assigned to have GPIO pins and are not talking to a 
specific Mesa daughter card.

-- 
Charles Steinkuehler
charles@steinkuehler.net

mhaberler, 2016-07-20:
I sort of guessed your intent - [here](https://github.com/machinekit/mksocfpga/blob/master/HW/firmware-tag/7I76_7I85S_GPIO_GPIO.py#L12)

mhaberler, 2016-07-20:
hm2 ignoramus here. help me understand this:

if a sserial daughter board is connected, I get pins  like:
````
    75        bit   OUT         FALSE  hm2_de0n.0.7i76.0.0.input-00  			0
..
    75        bit   OUT         FALSE  hm2_de0n.0.7i76.0.0.input-31  			0
...
    75        bit   IN          FALSE  hm2_de0n.0.7i76.0.0.output-00 			0 <== clock
...
````
and I know on which TBx-y these pins end up on the 7i76 and they work fine.

now, the case - no daughter board connected -  so no smartserial and all, so the above pins naturally do not show up

I still see pins like (those are present in the above case as well):
````
    74        bit   OUT         FALSE  hm2_de0n.0.gpio.000.in        			0
...
    74        bit   OUT          TRUE  hm2_de0n.0.gpio.067.in        			0
````

that sounds like 4 connectors at 17pins each.  log says:
````
Jul 20 23:40:01 mksocfpga msgd:0: hal_lib:3187:rt hm2/hm2_de0n.0: 68 I/O Pins used:
Jul 20 23:40:01 mksocfpga msgd:0: hal_lib:3187:rt hm2/hm2_de0n.0:     IO Pin 000 (GPIO0.P2-01): IOPort
...
Jul 20 23:40:01 mksocfpga msgd:0: hal_lib:3187:rt hm2/hm2_de0n.0:     IO Pin 066 (GPIO1.P3-12): IOPort
Jul 20 23:40:01 mksocfpga msgd:0: hal_lib:3187:rt hm2/hm2_de0n.0:     IO Pin 067 (GPIO1.P3-13): IOPort
````

does this suggest hm2_de0n.0.gpio.000.in reflects  IO Pin 000 (GPIO0.P2-01) ? Can I just use hm2_de0n.0.gpio.0xx.in as general gpios in this case?

cdsteinkuehler, 2016-07-21:
On 7/20/2016 6:56 PM, Michael Haberler wrote:
>
> does this suggest hm2_de0n.0.gpio.000.in reflects IO Pin 000 (GPIO0.P2-01) ? Can
> I just use hm2_de0n.0.gpio.0xx.in as general gpios in this case?

Yes.

Each physical pin on the FPGA typically has two functions.  All pins 
have a GPIO function, and most pins have a secondary function as well 
(step, dir, pwm, encoder, etc).

This can be seen in the PIN_*.vhd files, where each pin has a GPIO entry 
(IOPortTag) and most have a secondary function as well:

https://github.com/machinekit/mksocfpga/blob/master/HW/hm2/config/DE0_Nano_SoC_DB25/PIN_7I76_7I85S_GPIO_GPIO.vhd#L113-L129

Note the GPIO only connectors have no secondary function:

https://github.com/machinekit/mksocfpga/blob/master/HW/hm2/config/DE0_Nano_SoC_DB25/PIN_7I76_7I85S_GPIO_GPIO.vhd#L113-L129 

-- 
Charles Steinkuehler
charles@steinkuehler.net

mhaberler, 2016-07-21:
comprende, thanks! will explore pins tomorrow.

mhaberler, 2016-07-21:
do I understand correctly:

- no sserial daughter card - SSerialTag-tagged pins fall back to GPIO
- on 4x7i76 - without daughter card: the only thing missing is the _additional_ GPIO pins "hidden" behind the sserial
- AFAICT the stepgen and encoder pins would still be available at the DB25 pins because they are not behind a sserial?

(more ignoramus questions..):
- what else could possibly behind sserial ports? (I guess the RS485 functions.. where do I look that up?)
- what is the point of more than 1 sserial port per DB25 (like [here](https://github.com/machinekit/mksocfpga/blob/master/HW/hm2/config/DE0_Nano_SoC_DB25/PIN_7I76_7I76_7I76_7I76.vhd#L184-L188))?

- does [this line suggest LED1 is disabled ](https://github.com/machinekit/mksocfpga/blob/master/HW/hm2/config/DE0_Nano_SoC_DB25/PIN_7I76_7I76_7I76_7I76.vhd#L150)if the pin is used for a Qcounter?

I guess I can follow the 'LEDs dont blink' issue by probing GPIO0/1 pins 1 and 19 respectively?

unseenlaser, 2016-07-21:
Michael ,
their is only one sserial per 7i76 or 7i77 ,open to connections on TB2 (
7I76 for example )  any other devices devices would be daisychained , and
notified via it's own address
for instance the 7i73 pendant on mode 1 is addressed in the setup as
sserial_port_0=100xxx

i believe that the 5i25 uses 2 sserial buss  1 for multiplexing the GPIO
and the other for multiplexing the encoders and analog spindle
i'm not 100% as this is by memory , i'd need to check  the vhdl files .

Sarah

-- 

The information contained in this message is confidential and is intended
for the addressee only. If you have received this message in error or there
are any problems please notify the originator immediately. The unauthorised
use, disclosure, copying or alteration of this message is strictly
forbidden. This mail and any attachments have been scanned for viruses
prior to leaving the RcTechnix network. RcTechnix will not be liable for
direct, special, indirect or consequential damages arising from alteration
of the contents of this message by a third party or as a result of any
virus being passed on.

RcTechnix reserves the right to monitor and record e-mail messages being
sent to and from this address for the purposes of investigating or
detecting any unauthorised use of its system and ensuring effective
operation.

(c) RcTechnix

unseenlaser, 2016-07-21:
Michael

further info , on a standard 5i25 bit file built for a 7i76 gives you ( per
db25 )
        IOPortTag & x"00" & SSerialTag & SSerialTX0Pin,     -- I/O 10
PIN 7
        IOPortTag & x"00" & SSerialTag & SSerialRX0Pin,     -- I/O 11
PIN 8
        IOPortTag & x"00" & SSerialTag & SSerialTX1Pin,     -- I/O 12
PIN 9
        IOPortTag & x"00" & SSerialTag & SSerialRX1Pin,     -- I/O 13
PIN 10

my understanding now is that theirs 2 SSerial channels per connector in
this case

-- 

The information contained in this message is confidential and is intended
for the addressee only. If you have received this message in error or there
are any problems please notify the originator immediately. The unauthorised
use, disclosure, copying or alteration of this message is strictly
forbidden. This mail and any attachments have been scanned for viruses
prior to leaving the RcTechnix network. RcTechnix will not be liable for
direct, special, indirect or consequential damages arising from alteration
of the contents of this message by a third party or as a result of any
virus being passed on.

RcTechnix reserves the right to monitor and record e-mail messages being
sent to and from this address for the purposes of investigating or
detecting any unauthorised use of its system and ensuring effective
operation.

(c) RcTechnix

unseenlaser, 2016-07-21:
as far as the led1 , i believe that it's a typo , as the led function is
not defined
and still needs to be attached to a pin , i cant see the quadrature pins in
any way to be part of the led

-- 

The information contained in this message is confidential and is intended
for the addressee only. If you have received this message in error or there
are any problems please notify the originator immediately. The unauthorised
use, disclosure, copying or alteration of this message is strictly
forbidden. This mail and any attachments have been scanned for viruses
prior to leaving the RcTechnix network. RcTechnix will not be liable for
direct, special, indirect or consequential damages arising from alteration
of the contents of this message by a third party or as a result of any
virus being passed on.

RcTechnix reserves the right to monitor and record e-mail messages being
sent to and from this address for the purposes of investigating or
detecting any unauthorised use of its system and ensuring effective
operation.

(c) RcTechnix

cdsteinkuehler, 2016-07-21:
On 7/21/2016 6:05 AM, Michael Haberler wrote:
> do I understand correctly:
>
>   * no sserial daughter card - SSerialTag-tagged pins fall back to GPIO

Mostly.  I'd say the sserial pins don't get instantiated and the GPIO 
function is always there, but I'd have to crawl through the driver code 
to see exactly what happens.

>   * on 4x7i76 - without daughter card: the only thing missing is the
>     /additional/ GPIO pins "hidden" behind the sserial

Yes.

>   * AFAICT the stepgen and encoder pins would still be available at the DB25
>     pins because they are not behind a sserial?

Yes.

> (more ignoramus questions..):
>
>   * what else could possibly behind sserial ports? (I guess the RS485
>     functions.. where do I look that up?)

There are lots of things that can be hooked up via the serial interface:

http://store.mesanet.com/index.php?route=product/category&path=69_70

>   * what is the point of more than 1 sserial port per DB25 (like here
>     <https://github.com/machinekit/mksocfpga/blob/master/HW/hm2/config/DE0_Nano_SoC_DB25/PIN_7I76_7I76_7I76_7I76.vhd#L184-L188>)?

I believe on the 7i76, one sserial link is used for the on-board I/O 
connections, and the other is provided as an RS485 differential link for 
hooking to remote serial devices (see above).

>   * does this line suggest LED1 is disabled
>     <https://github.com/machinekit/mksocfpga/blob/master/HW/hm2/config/DE0_Nano_SoC_DB25/PIN_7I76_7I76_7I76_7I76.vhd#L150>if
>     the pin is used for a Qcounter?

No, that's an off-by-one goof on my part when doing the cut-and-paste.

Note that the I/O 17 comment is describing the header line, not the 
first pin.  The comments for I/O 17 through I/O 33 and LED1 need to drop 
down one line.

> I guess I can follow the 'LEDs dont blink' issue by probing GPIO0/1 pins 1 and
> 19 respectively?

Yes.

-- 
Charles Steinkuehler
charles@steinkuehler.net

mhaberler, 2016-07-24:
the LED issue is solved - numbering:

CR01/CR02 are GPIO1 pin 19 and 1
CR02/CR03 are GPIO0 pin 19 and 1

while a bit counterintuitive, they work

cdsteinkuehler, 2016-07-26:
On 7/24/2016 3:28 AM, Michael Haberler wrote:
> the LED issue is solved - numbering:
> 
> CR01/CR02 are GPIO1 pin 19 and 1
> CR02/CR03 are GPIO0 pin 19 and 1
> 
> while a bit counterintuitive, they work

Hmm...that should probably be fixed.  I would have expected:

CR1 = GPIO0 pin 1
CR2 = GPIO0 pin 19
CR3 = GPIO1 pin 1
CR4 = GPIO1 pin 19

...based on how they're connected:

https://github.com/machinekit/mksocfpga/blob/master/HW/QuartusProjects/DE0_Nano_SoC_DB25/DE0_Nano_SoC_DB25.vhd#L454-L457

Perhaps the hm2 driver numbers them oddly?

-- 
Charles Steinkuehler
charles@steinkuehler.net