machinekit/mksocfpga #54: firmware snag sheet: DE0_Nano_SoC_DB25.7I76_7I85S_GPIO_GPIO.rbf

mhaberler, 2016-07-19:
loading the DE0_Nano_SoC_DB25.7I76_7I85S_GPIO_GPIO.rbf file fails, [logfile here](http://static.mah.priv.at/public/DE0_Nano_SoC_DB25.7I76_7I85S_GPIO_GPIO.txt)

reproduce-by (no boards connected):

`````
loadrt hostmot2 debug_idrom=1 debug_module_descriptors=1 debug_modules=1 debug_pin_descriptors=1
loadrt hm2_soc_ol config="firmware=socfpga/dtbo/DE0_Nano_SoC_DB25.7I76_7I85S_GPIO_GPIO.dtbo"
````

the key error message seems to be:

`````
Jul 19 16:44:58 mksocfpga msgd:0: hal_lib:3483:rt hm2/hm2_de0n.0: found duplicate Module Descriptor for Muxed Encoder (inconsistent firmware), not loading driver
Jul 19 16:44:58 mksocfpga msgd:0: hal_lib:3483:rt hm2/hm2_de0n.0: failed to parse Module Descriptor 4 of 10 gtag Muxed Encoder
Jul 19 16:44:58 mksocfpga msgd:0: hal_lib:3483:rt hm2_de0n.0: hm2_soc_ol_board fails HM2 registration
````

[full logfile here](http://static.mah.priv.at/public/DE0_Nano_SoC_DB25.7I76_7I85S_GPIO_GPIO.txt)

mhaberler, 2016-07-19:
@cdsteinkuehler - thinking through how we prepare the fwid protobuf message per-config, I guess one variant of generate.py per config? if OK I can prepare that

if I get this right, the DE0_Nano_SoC_DB25.7I76_7I85S_GPIO_GPIO would have num_leds=2 and DE0_Nano_SoC_DB25.7I76_7I76_7I76_7I76 would have num_leds=4?

cdsteinkuehler, 2016-07-20:
Hmm...it looks like hostmot2 may not like having regular and muxed encoders enabled in the same firmware.  I'll review when I get back from the mountains.

mhaberler, 2016-07-22:
I know these issues are expanding into a bit of a Q&A session, but maybe this helps somebody downstream as well

so..

- set up my own build job at https://jenkins.machinekit.io/view/userjobs/job/mksocfpga-mah/
- started from mk/mksocfpga master to https://github.com/mhaberler/mksocfpga/commits/mah

I tried a few things:
- see if I can fix this Encoder/Muxed encoder loading issue myself, it seems that worked
- add 2 PWMgens via bare GPIO
- add a 3pwmgen via bare GPIO

loading the resulting RBF: 

[linuxcnc.log](https://github.com/mhaberler/mksocfpga/blob/mah/log/linuxcnc.log)  
[pins-and-params](https://github.com/mhaberler/mksocfpga/blob/mah/log/pins-params.log)

- [replacing the unmuxed Counter by a muxed one](https://github.com/mhaberler/mksocfpga/commit/b5564e646975dc2304539ee6117966ca26a4909a) removed the loading issue, and the assignments and pins seem to come out right ([inst0](https://github.com/mhaberler/mksocfpga/blob/mah/log/linuxcnc.log#L1137-L1140))
- adding the PWMgens seems to have [worked](https://github.com/mhaberler/mksocfpga/blob/mah/log/linuxcnc.log#L1174-L1179), [params](https://github.com/mhaberler/mksocfpga/blob/mah/log/pins-params.log#L225-L228) and [pins](https://github.com/mhaberler/mksocfpga/blob/mah/log/pins-params.log#L561-L564) but it looks the out pins are missing (Out0 Out1 Not-Enable :-/)
- adding the 3pwmgen [seems to have worked](https://github.com/mhaberler/mksocfpga/blob/mah/log/linuxcnc.log#L1157-L1164), but the inverted (low-side) pins are missing in HAL, only [high-side present](https://github.com/mhaberler/mksocfpga/blob/mah/log/pins-params.log#L259-L263)

do you see any obvious blunders?

luminize, 2016-07-23:
> - adding the PWMgens seems to have worked, params and pins but it looks the out pins are missing (Out0 Out1 Not-Enable :-/)
- adding the 3pwmgen seems to have worked, but the inverted (low-side) pins are missing in HAL, only high-side present

I think I just found some answers.
w.r.t. pwmgen.nn.out0, i think it's named pwmgen.nn.value ([see part of source](https://github.com/machinekit/machinekit/blob/master/src/hal/drivers/mesa-hostmot2/pwmgen.c#L460-L474))

as for the inverted 3pwmgen pins, they are initiated at the hardware level, but not in HAL, also because they are never made ([see part of source](https://github.com/machinekit/machinekit/blob/master/src/hal/drivers/mesa-hostmot2/tp_pwmgen.c#L317-L382))

dkhughes, 2016-07-23:
I would imagine the inverted pins function like the GPIO counterparts, i.e., you simply set 

`board.0.gpio.blah.invert_output` true

Here is the relevant excerpt from the man page:

```
(bit r/w) invert_output
This parameter only has an effect if the "is_output" parameter is true. If this parameter is true, the output value of the GPIO will be the inverse of the value on the "out" HAL pin. Only full GPIO pins and IO pins used as outputs by active module instances have this parameter.
```
and it will be handled automagically by the firmware. No need to specifically use the normal or inverted output. Inputs are a different story, both polarities of those are exposed simultaneously.

mhaberler, 2016-07-24:
it seems to me the HAL pins are not made on purpose - really they serve no value; it's just important the hardware pins are mapped and they probably are, testing right now

mhaberler, 2016-07-24:
interestingly, [this commit](https://github.com/mhaberler/mksocfpga/commit/9b765e1d8b023d9894ab0c80cb21438ebec7db91) made the normal PWM work as expected

it seems we need to always choose the ClockLowTag (50MHz), not the ClockHighTag (200Mhz)

mhaberler, 2016-07-24:
perfect with ClockLowTag, PWM's working as expected

I think I'm getting the hang of it ;)

mhaberler, 2016-07-24:
I think I'll prepare a PR for the 7I76_7I85S_GPIO_GPIO, with a couple more encoders and PWM's thrown in at the GPIO1.P2/3 - sounds reasonable?

mhaberler, 2016-07-26:
the muxed/unmuxed encoder restriction is [mentioned here](https://forum.linuxcnc.org/forum/39-pncconf/26183-mesa-5i25-7i85s#30283)

mhaberler, 2016-07-26:
I changed the [FPGA config like so and it seems to load](https://github.com/mhaberler/mksocfpga/commit/5efcc6a025ddc489891c5ff93bb0d73047dc2b72)

on the HAL side, I guess I am missing some `sserial=<yadayada>` incantation ?

`loadrt hm2_soc_ol config="firmware=socfpga/dtbo/DE0_Nano_SoC_DB25.7I76_7I85S_GPIO_GPIO.dtbo num_pwmgens=0  num_stepgens=8 num_encoders=4"`

mhaberler, 2016-07-27:
Charles explained to me what muxed encoders are about:

> btw do muxed encoders + unmuxed ones in one image make sense? 

Yes

> what's the difference anyway?

The unmuxed encoder has each encoder signal (A, B, Index) on it's own
dedicated FPGA I/O pin which then shows up on the DB25 or 50-Pin
connector to the daughter card.

The muxed encoder has a single pin for A, B, and Index (three pins
total) and an encoded select signal that gets sent to the
daughter-card.  The FPGA logic cycles through the various encoder
select values at a "brisk" rate (in the low MHz range) and the
daughter-card sends the signals for *THAT* encoder over the shared
A/B/Z pins where the VHDL hardware updates the appropriate muxed
encoder instance based on the encoder selected.  It's a scheme to
trade off encoder update speed for reduced I/O count (ie: you can
support 8 muxed encoder inputs with 6 pins (A/B/Z + 3 select pins),
but it would take 24 pins to have 8 unmuxed encoders).

mhaberler, 2016-08-04:
if I read http://www.mesanet.com/pdf/parallel/7i85sman.pdf page 4 (CONTROLLER CONNECTOR) right, there is one pair of encoder pins (MENCA0,MENCA1..) and a single select line ENCMUX, giving a maximum of 4 muxed encoders for now

but the 7i76 has an unmuxed encoder so [this does not make sense](https://github.com/mhaberler/mksocfpga/blob/5efcc6a025ddc489891c5ff93bb0d73047dc2b72/HW/hm2/config/DE0_Nano_SoC_DB25/PIN_7I76_7I85S_GPIO_GPIO.vhd#L127-L129), and given that muxed and unmuxed encoders currently do not coexist, the 7i76_7i85s* configs right now cannot be made to work [until this fix is in](https://github.com/machinekit/mksocfpga/issues/54#issuecomment-235216528)

cdsteinkuehler, 2016-08-04:
IMHO, since the hardware happily supports both types of encoders and there are valid reasons for wanting to have both muxed and unmuxed enoders supported at the same time, the driver should be updated to allow this.  I can tweak the driver early next week (I'm currently traveling) if no one beats me to it.

mhaberler, 2016-08-04:
I'm back from travel and have a few days, I'll give it a stab

mhaberler, 2016-08-04:
thinking about [the fix:](https://github.com/machinekit/mksocfpga/issues/54#issuecomment-235216528)

an alternative to the above scheme would be to treat unmuxed and muxed encoders separately, and add a num_mencoders=<count> argument

the advantage of this approach is that it gives control over which encoders get instantiated; for example, in the 7i76_7i85s_ff config, if one were to desire only muxed encoders on the 7i85s but not the spindle encoder on the 7i76, there would be no way to do so because the unmuxed encoder would always be instantiated

but if we had "... num_encoders=0 num_mencoders=4 it would be possible to instantiate only the muxed encoders on the 7i85s

not sure if this is worth the detour - @cdsteinkuehler, what do you think?

dkhughes, 2016-08-04:
> but if we had "... num_encoders=0 num_mencoders=4 it would be possible to instantiate only the muxed encoders on the 7i85s

I'm not sure about that specific design, but I think of those items as separate and equally valid components to instantiate.

mhaberler, 2016-08-04:
that would suggest complete decoupling at the data structure level and is probably the cleanest way to do this:

- separate hm2.config.num_encoders and hm2.config.num_mencoders values
- hm2_encoder_t encoder and  hm2_encoder_t mencoder;
- in encoder.c, change the internal API to refer to  hm2_encoder_t * instead of 'int instance' to get rid of the hardcoded reference to hm2->encoder 
- disentangle the instantiation 

does it make sense to use different HAL names (like 'mencoder')? it would make it a bit easier to keep track of instance numbers and shows what is what

mhaberler, 2016-08-04:
I have a first cut: https://github.com/mhaberler/machinekit/commits/dual-encoders (top 8 commits)

one 7i76, one 7i85s connected

running Charles' DE0_Nano_SoC_DB25.7I76_7I85S_GPIO_GPIO.rbf which used to fail now loads, encoder pins come up fine - nothing wrong I can discern in the log:

`````
loadrt hostmot2 debug_idrom=1 debug_module_descriptors=1 debug_modules=1 debug_pin_descriptors=1
loadrt hm2_soc_ol config="firmware=socfpga/dtbo/DE0_Nano_SoC_DB25.7I76_7I85S_GPIO_GPIO.dtbo"
``````
log: http://static.mah.priv.at/public/muxed-unmuxed.log
pins and params: http://static.mah.priv.at/public/pinparam.txt

this is untested functionally, will check tomorrow

this is a mechanical double-instantiation of encoders into mencoders
the thing I do not know - do unmuxed and muxed encoders share _anything_ ?  

btw I would be very grateful for an explanation what the TRAM mechanism does - completely in the dark..

cdsteinkuehler, 2016-08-04:
On 8/4/2016 6:23 PM, Michael Haberler wrote:
> 
> do unmuxed and muxed encoders share /anything/ ?

At the hardware level?  No.

See the totally separate "generate" blocks in hostmot2.vhd labeled
makeqcounters and makemuxedqcounters.

...and the pins are assigned uniquely for the muxed/unmuxed qcounters
as well.

> btw I would be very grateful for an explanation what the TRAM mechanism does - 
> completely in the dark..

TRAM = Translation RAM

IIRC my discussions with Peter, it doesn't do anything unless you're
on an EPP parallel-port connected board.

The idea is that the driver can create a list of values to read from
and write to, and can then perform all the reads and writes in a
contiguous block regardless of the desired source/destination address.
 That allows a single burst transfer (supported by EPP and other buses
like PCI/PCIe) to target random addresses.

In other words:

  Read 1 to 4

...might turn into:

  Read Encoder position 1
  Read Encoder position 3
  Read StepGen position 1
  Read GPIO inputs

...and...

  Write 1 to 2

...might turn into:

  Write StepGen 1
  Write GPIO out

In general, the PCI/PCIe overhead of doing individual random reads and
writes is pretty minor so the driver (and maybe the hardware?) doesn't
bother with implementing TRAM.  But for EPP (and maybe the Ethernet
connected boards?) it really does makes a difference.

-- 
Charles Steinkuehler
charles@steinkuehler.net

mhaberler, 2016-08-05:
ah ok, some scatter-gather thing.

thanks - then I'm hopeful this should work!

cdsteinkuehler, 2016-11-01:
Implemented by @mhaberler vi machinekit/machinekit#1019

closed 2016-11-01 by cdsteinkuehler