machinekit/mksocfpga #55: offer: custom firmware builds

mhaberler, 2016-07-21:
in case someone wants to experiment with custom FPGA firmware builds - that is super easy to do on jenkins.machinekit.io

happy to host valiant experimenters ;)

outline:
- fork mk/mksocfpga
- drop me a line so I can raise permissions on jenkins.machinekit.io
- add your github credential under 'credentials'
- generate a new project -> copy from 'mksocfpga'
- change the repo URI 
- edit project configuration as needed
- if you add your github credential - needed so jenkins manages the github hooks - you push to your repo, build gets triggered; not needed for manually triggered build
- the .RBF file can either be directly downloaded, or we can add some web upload destination, say under deb.mah.priv.at/uploads/joe-redsocks

the-snowwhite, 2016-07-21:
Wow

This concept is impressively mind blowing, if I am getting it right... ?

If this makes it possible to be able to do fpga programming / bugfixing etc,from lets say my cell phone on a mountain top,,, it opens up a wide range of options like:

 Doing local or remote firmware updates from anywhere you can hook up to the internet.
Also great for field testing purposes also, when something unexpected that needs a quick fix happens and your workstation is miles away.....

I need to try this out asap...

mhaberler, 2016-07-21:
here it is: https://jenkins.machinekit.io/view/userjobs/job/mksocfpga-the-snowwhite/

mhaberler, 2016-07-21:
you can follow my cargo-culting attempts at https://github.com/mhaberler/mksocfpga/commits/mah and https://jenkins.machinekit.io/view/userjobs/job/mksocfpga-mah

build works, it's just that the nano is 200km away :-/

mhaberler, 2016-07-21:
actually we might make this a travis job so everybody can do this, even without my jenkins

we might have to brush up @cdsteinkuehler 's QuartusBuildVM Docker repo to build on dockerhub so everybody can pull, not sure this is automated

cdsteinkuehler, 2016-08-06:
Updating the QuartusBuildVM docker images is *NOT* automated.  You can easily generate a new image using the Dockerfile, but it will be *HUGE* and unsuitable for sharing.  Directions for shrinking the image to a manageable size are in the dockerfile:

https://github.com/cdsteinkuehler/QuartusBuildVMs/blob/master/Jessie-Quartus-15.1.2/Dockerfile#L53-L82

Questions:
Do I need to generate a new docker image (very painful for me, uploading that much data reliably kills my local internet connection), or should we have official Machinekit (vs. cdsteinkuehler) images on Dockerhub?

If we want to stick with my cdsteinkuehler images, maybe I can use MAH's server to create and upload?  I think that would work much better than my local ISP for pushing GBs of data upstream!

mhaberler, 2016-08-06:
I think we should move to a machinekit dockerhub user
definitely use my server, wont get stuck

if you build them as-is on dockerhub - does it work? or does it exceed some limit?

cdsteinkuehler, 2016-08-06:
On 8/6/2016 4:38 PM, Michael Haberler wrote:
> 
> if you build them as-is on dockerhub - does it work? or does it exceed some limit?

Unknown...I wouldn't be able to upload one even if I wanted to.

I have enough trouble getting the ~5G images uploaded[1], I'd never be
able to complete an ~30G upload.

[1] I have to monitor my uplink status and when it dies I need to
power-cycle my cable-modem quickly enough that the TCP connection
doesn't time-out.  My uplink dies typically 3-5 times when uploading a
5G image, which takes many hours.  :(

-- 
Charles Steinkuehler
charles@steinkuehler.net