machinekit/mksocfpga #61: Xilinx Zybo support?

cdsteinkuehler, 2016-08-11:
I just got a Xilinx Zybo board (for a non-Machinekit project), and wondered if anyone else is familiar with this board or is working with it.  Ideally, I'd like to be able to run a Machinekit Debian uSD image on this board for development, but I'm not familiar enough with the Xilinx Zynq world to know if that's possible (or even makes sense).

Also, does anyone know if there are "Pmod" adapters for motion control like interfaces (stepper motors, servos, encoders, etc)?  I couldn't find anything really suitable for machine control but I've heard the Pmod connectors (really just a 12-pin 0.1" socket) are fairly universal and can hook to lots of different things.

claudiolorini, 2016-08-11:
I've never tried on the Zybo but it's not very different from the Zedboard or the Z-Turn where i've installed successfully MK and Debian, i'm sure it can be done (and make a lot of sense).
There are a lot of PMODs from Digilent for sensors, communication, ecc.. they have some H-bridges and RC-servo but no encoders or anything for 'serious power'.
The beauty of PMODs is for prototyping as they are extremely easy to build, i've done a lot of little modules, Isolated CAN, isolated IOs, sigmadelta modulators, isolated current sensors ecc.. in order to test circuital parts for motor controllers...

luminize, 2016-08-11:
Never knew about these “mods", very interesting. FWIW, I recently saw this page which looks interesting. http://erlerobotics.com/blog/learning/hardware-robot-operating-system-h-ros/

dbskccc, 2016-08-11:
@claudiolorini  I just got a Z-Turn Board now ,would you post some more info for how to run machinekit on it.

claudiolorini, 2016-08-12:
...just a fiew pointers, ask if i'm too generic.

You need a uSSD with 1 fat32 partition, 1 ext4 (and one swap)

 - cross compile a suitable kernel.
for example this is what we usually use for the Zynq platforms (zed,myir,trenz)
https://github.com/andreamerello/linux-zynq-stable/tree/linux-4.6-zynq
if you have the Vivado suite installed:

export CROSS_COMPILE=arm-xilinx-eabi-
source /opt/Xilinx2015.4/SDK/2015.4/.settings64-Software_Development_Kit.sh

for the first build do also:
make ARCH=arm  zynq_edl_defconfig

if you have to do some modification to the kernel config (otherwise skip)
make xconfig ARCH=arm  

make ARCH=arm UIMAGE_LOADADDR=0x8000 uImage -j8
copy the  arch/arm/boot/uImage to your FAT partition on the uSSD

 - prepare a suitable device-tree
make ARCH=arm zynq-zturn-myir.dtb
copy arch/arm/boot/dts/zynq-zturn-myir.dtb to you uSSD as devicetree.dtb

 - cook a boot.bin with Vivado for your FPGA stuff, this is a rather complex stuff 
if you don't know Vivado, you need 
the u-boot (uboot.elf) we use this: https://github.com/francescodiotalevi/u-boot-xlnx 
FSBL.elf and the FPGA bit file. 
in a previous post @dkhughes said it can be done without a FPGA bitfile but i've never tried...

When (if!) you see something booting you can install Debian using a netinstall image (armhf)
from here:
http://ftp.it.debian.org/debian/dists/jessie/main/installer-armhf/current/images/netboot/initrd.gz
  mkdir root-netinst
  cd root-netinst
  cpio -idv < ../initrd
and cp -a 'all the stuff' to the EXT4 partition on the SSD

it should reboot with the netinstall procedure, choose one install of your choice,
i suggest to proceed with a non graphical and then add X and manager of you choice later
with apt-get. 

The process takes some time and a internet connection, but once done you can save the image 
for future projects...

once you have a Debian up tu install MK just follow the instructions from here:
http://www.machinekit.io/docs/developing/machinekit-developing/ 

it should work rather smoothly... unless it doesn't ;-)

C.

the-snowwhite, 2016-08-12:
Please excuse me stepping in sideways however I am th one who manifested the mksocfpga project and I have been developing and maintaining a independant set of scripts sole purposed for producing something that works as simply as possible.

On the Machinekit side the scripts were separated into a repo named buildscripts (the-snowwhite@github.com), the newest version is in master.

Further development of the scripts has continued over supper in my HolosynthED revival repo.
There the current version is now in the work branch in the scripts folder.

u-boot from den. (cross compiled)
Kernel from git … (cross compiled)
Make sd-image or rootfs image 
Generate roots with qemu-debootstrap.

+ main caller script, with usersetup etc.

Be aware that when the kernelscript runs generate make u-boot u-image (the new make deb. packages build option), systems will go into degraded mode,
after an ioctl error(q-emu related) then do a sudo systenctl reset-failed, (unless you ar happy to experience weird system failures on your x86_64 machine)

Run tested on debian jessie / stretch (qemu 2.5+) and also kubuntu 16.4 lts.

——

Bottom line is you will be able to find all the commands you need to build the stuff you need (from scratch).

By putting in the right git urls, etc.

Please let me know hos it fares.

Best Wishes
Michael :-)

the-snowwhite, 2016-08-12:
Ups darn I really hate those wierd spell corrections:
i meant over this summer Not supper....

dkhughes, 2016-08-12:
@dbskccc Please see Issue #20 for z-turn info. If you wait just a couple of days I will have the zynq sd image packaged for both the microzed and zturn boards. Just debugging the last boot issues on zturn today.

We had to cherry pick kernel patches to get the fpga overlay stuff that we use in the mk driver layer into the xilinx kernel, and then port over u-boot to the zturn and make it boot from the root filesystem instead of the first partition so that kernels and firmware are maintainable via debian packages.

@claudiolorini Has some really good advice, but make sure you are compiling an RT patched, *and* fpga overlay patched kernel or mk will fail. No need for a bitfile in the boot partition either, unless you have something you need running immediately on power up. We delay load the firmware in mk when the hm2_soc_ol module gets loaded.

@cdsteinkuehler  Okay, on topic - I glanced at zybo and it looks pretty straightforward to get this stuff going. I don't have one, but I will build a basic project for it and make sure the u-boot pieces are in place. I'm pretty sure zybo is already in my u-boot tree, so it's really just checking to make sure the dts source is compatible.

dkhughes, 2016-08-12:
Oh, it would be helpful to have a reference design that functions for a zybo, to pull ddr settings, etc. out of - if you have on can we dump it somewhere that I can access?

dbskccc, 2016-08-12:
@claudiolorini thanks for your guide
@dkhughes  thanks for your hardwork, I'm looking forward to try your SD image

dkhughes, 2016-08-12:
Yep, board files are even better. The only downside is that the board files have to be included in the install directory of Vivado, you can't keep them out of tree (in 2015.4 at least, it might be fixed in the newer version). 

I will add those board files to the docker image so they are available, which means this should be really easy.

cdsteinkuehler, 2016-09-04:
I have just finished a major project at my RealJob, and starting next
week I expect to be able to work on the Zybo board as part of my
official duties.

What's the expected status of being able to boot one of the Machinekit
Xilinx uSD images on a Zybo?  Should it "just work" or will I have to
replace some key bits (U-Boot/Kernel/dts)?

-- 
Charles Steinkuehler
charles@steinkuehler.net

mhaberler, 2016-09-04:
i think dts and uboot should do
can you point me to an existing uboot branch? I'll setup the build
if that works we might roll and SD image eventually

dkhughes, 2016-09-07:
Looks like we're in luck! The relevant bits are already in the u-boot tree here for the SPL:

[https://github.com/JDSquared/u-boot-xlnx/tree/master/board/xilinx/zynq/zybo_hw_platform](https://github.com/JDSquared/u-boot-xlnx/tree/master/board/xilinx/zynq/zybo_hw_platform)

Kernel should be good to go already unless there are some devices that require drivers to boot that are not common with the other boards. I will look into the dts for zybo this evening and make sure it is compatible with the include we use that has fpga regions.

I will add the board files to the vivado-docker image to help ease project creation and I'll boiler plate something out. I don't have a Zybo to test or get pin info from, I'll need your help with that.

cdsteinkuehler, 2016-09-07:
On 9/7/2016 3:59 PM, dkhughes wrote:
> Looks like we're in luck! The relevant bits are already in the u-boot tree here
> for the SPL:
>
> https://github.com/JDSquared/u-boot-xlnx/tree/master/board/xilinx/zynq/zybo_hw_platform
>
> Kernel should [...]

Great!  So does this mean I should be good to go, or that I need a new
U-Boot and/or dts?

> I will add the board files to the vivado-docker image to help ease project 
> creation and I'll boiler plate something out. I don't have a Zybo to test or get 
> pin info from, I'll need your help with that.

I can test here.  I have a RealJob need to get the Zybo booting with
Debian, so I can spend some time on testing.

-- 
Charles Steinkuehler
charles@steinkuehler.net

dkhughes, 2016-09-08:
You will need a u-boot and SPL compiled specifically for the zybo. All of that early bootloading stuff is compiled per SoM at the moment due to the u-boot configs. The tree I linked should already be good to go, and since it's the one we're using to generate the u-boot packages already I think we just need to get a jenkins job to kick out the u-boot zybo package.

I checked out the kernel tree and the zybo dts is using the zynq7000.dtsi like the other boards which means it should be configured correctly. In fact, the zynq kernel package is already including that dtb in the distribution, so I don't see any work there. Just be aware that we are on a Vanilla kernel, and a few application specific drivers that haven't hit mainline yet would need to be manually added.

So, looking at this, @mhaberler's latest sd test image should boot with the replacement of the u-boot zturn package with a u-boot zybo package.

dkhughes, 2016-09-08:
I updated the jenkins job, it is now producing a zybo u-boot package as well. Just need to edit the omap image builder to use it instead of the zturn or microzed.

the-snowwhite, 2023-11-11:
resolved

closed 2023-11-11 by the-snowwhite