machinekit/mksocfpga #77: modifying zturn_ztio unclear

mhaberler, 2016-08-29:
studying the [zturn_ztio](https://github.com/machinekit/mksocfpga/tree/master/HW/VivadoProjects/zturn/zturn_ztio) config:

Assuming I do not use the Camera, LCD, and ADC connectors I would have the following connectors available at the IO cape:

- J5, J6, J7 with 8 pins each (page 6 1.1.4)
- J3 with 36pins (page 3 1.1.1)

so a total of 60 pins

- why only 34 pins?
- what are the RATES pins? scoping the timers?
- if I were to add the other 26 pins - can I do this just by editing the config files, or should I run the Vivado GUI for that? (if so, how?)
- would I have to follow the IO cape schematic -> CN1/CN2 connector pins -> BGA balls for each of them, or is there some existing map that can be reused?

semi-related:
- should we adapt hostmot2 with "uneven" connector widths?
- would there be any value in reflecting the actual connector/pin number in the HAL name?

dkhughes, 2016-08-29:
> why only 34 pins?

Two reasons:
1) I'm lazy, 60 pins is a lot to call out and as a first example I wasn't sure about what features we need to demo. So, I pulled the most common peripherals and pinned them out as a 34 pin header which gave me a decent amount of each module and duplicates.
2) The zturn board fpga is slightly more resource limited than say the microzed board, mostly due to pin choices which make the fitter pull buffers across sections. This is only a problem when you add *a lot* of features to hm2 though

>  what are the RATES pins? scoping the timers?

Yes, for scoping the dpll timer pins directly

> if I were to add the other 26 pins - can I do this just by editing the config files, or should I run the Vivado GUI for that? (if so, how?)

> would I have to follow the IO cape schematic -> CN1/CN2 connector pins -> BGA balls for each of them, or is there some existing map that can be reused?

Yes, I pulled the pins for the bga from the schematic they provided. In the hostmot2 PIN file comments the io carrier header pin is called out. I want to make a table of the choices, I just haven't had time.

> should we adapt hostmot2 with "uneven" connector widths?
would there be any value in reflecting the actual connector/pin number in the HAL name?

I see no reason why you couldn't have an odd number of header pins, but the PIN files are all based on 144 pins in the vhdl descriptor. That's the only hard limit I've seen.

As for the driver, the hm2_soc_ol patch I submitted doesn't regulate the number of pins at the hal layer provided your firmware protobuf message has the correct amount. See [https://github.com/machinekit/machinekit/commit/5bda85e416512530be2c38d0e1cfa0102ca2bc7b#diff-808b7a7bb9570ae3fe0ecd40f35dfcb9R427](https://github.com/machinekit/machinekit/commit/5bda85e416512530be2c38d0e1cfa0102ca2bc7b#diff-808b7a7bb9570ae3fe0ecd40f35dfcb9R427)

dkhughes, 2016-08-29:
> if I were to add the other 26 pins - can I do this just by editing the config files, or should I run the Vivado GUI for that? (if so, how?)

Oops, missed this one. You need to do a few things if you want to expand the number of pins exposed. If you just want to reconfigure within the 34 pin limit, you just update the PIN file and recompile without needing to touch the tools directly. 

Until someone is comfortable with the generated config files, it's probably easiest to generate the necessary bits using the gui tools. Run the make_bitfile script on the project you want to work with, then open Vivado. Browse to the project_name_created directory and open the created project file. On the left side, click open block design and you will see the high level components it takes to build a zynq firmware.

To increase the ztio project to 60 pins:

1) Increase the width of the IOBits io buffer at the top level. In the GUI, click on the IObits and then change the left setting to 59.
2) Double click on the IO muxing component (in between the port and the hostmot2 instance in that project) and increase the width there accordingly.
3) Double click on the hostmot2_ip_wrap instance and increase the width of the IO bus accordingly.
4) Update your pin file to call out the 60 pins of functionality
5) Update the constraint file to include the new IO bits and their respective BGA pins

Save, and the project should compile. All of the changes above, with the exception of the PINS and physical BGA constraints file, are temporary. To make them permanent, you would export the block diagram and diff it against the tcl file in the scripts folder that generates the block diagram. The generated scripts are specific to your machine, whereas the example scripts use relative directories. It's pretty straightforward to see the changes you need to accept versus the ones to leave out. I use a graphical diff for this (Meld). 

After updating the script files, you can run make_bitfile and it will compile and generate the needed bitfiles for linux programming.

I will document all of this, and I want to make a screencast of what I described above to make a custom project, and to extend an existing project without wrecking an already existing config. I just need work to slow down a bit before I can make those, but it's on the way I promise.

mhaberler, 2016-08-30:
I kindof got this baked to the extent [a bitfile is generated](https://jenkins.machinekit.io/view/userjobs/job/mksocfpga-vivado-mah/6/console) from [this config](https://github.com/mhaberler/mksocfpga/tree/pins64/HW/VivadoProjects/zturn/zturn_ztio64) 

I think I got entangled in misinterpretation of how IDROM PortWidth, IOPorts, IOWidth, the [llio values num_ioport_connectors + pins_per_connector](https://github.com/mhaberler/machinekit/blob/master/src/hal/drivers/mesa-hostmot2/hostmot2-lowlevel.h#L108-L109), and the fwid struct interact

it went like this:
- I deleted the RATES pins (-5 pins)
- I added the J7 connector and completed the J3 connector
- which gives us (J5,J6,J7)@8 + J3@36 = 60pins, plus the 4 switches on U20 totals [64pins represented in this pin file](https://github.com/mhaberler/mksocfpga/blob/pins64/HW/VivadoProjects/zturn/zturn_ztio64/const/PIN_ZTIO_64.vhd) plus the 3 LEDs
- I made U20 a 'connector' and came up with this [fwid message](https://github.com/mhaberler/mksocfpga/blob/pins64/HW/firmware-tag/FWID_ZTIO_64.py#L15-L33)
- I added [J7 and completed J3 in the constraints](https://github.com/mhaberler/mksocfpga/blob/pins64/HW/VivadoProjects/zturn/zturn_ztio64/const/zturn_ztio_pinmap.xdc) and that seems to add up

that bitfile doesnt load though:

`````
...
ug 30 13:06:18 mksocfpga msgd:0: hal_lib:11718:rt hm2/hm2_ztio.0: connector 0 name = 'U20'
Aug 30 13:06:18 mksocfpga msgd:0: hal_lib:11718:rt hm2/hm2_ztio.0: connector 0 pins = 4
Aug 30 13:06:18 mksocfpga msgd:0: hal_lib:11718:rt hm2/hm2_ztio.0: connector 1 name = 'J6'
Aug 30 13:06:18 mksocfpga msgd:0: hal_lib:11718:rt hm2/hm2_ztio.0: connector 1 pins = 8
Aug 30 13:06:18 mksocfpga msgd:0: hal_lib:11718:rt hm2/hm2_ztio.0: connector 2 name = 'J5'
Aug 30 13:06:18 mksocfpga msgd:0: hal_lib:11718:rt hm2/hm2_ztio.0: connector 2 pins = 8
Aug 30 13:06:18 mksocfpga msgd:0: hal_lib:11718:rt hm2/hm2_ztio.0: connector 3 name = 'J3'
Aug 30 13:06:18 mksocfpga msgd:0: hal_lib:11718:rt hm2/hm2_ztio.0: connector 3 pins = 36
Aug 30 13:06:18 mksocfpga msgd:0: hal_lib:11718:rt hm2/hm2_ztio.0: connector 4 name = 'J7'
Aug 30 13:06:18 mksocfpga msgd:0: hal_lib:11718:rt hm2/hm2_ztio.0: connector 4 pins = 8
Aug 30 13:06:18 mksocfpga msgd:0: hal_lib:11718:rt hm2/hm2_ztio.0: invalid IDROM PortWidth 16, this board has 4 pins per connector, aborting load (<-- suppress error exit here)
Aug 30 13:06:18 mksocfpga msgd:0: hal_lib:11718:rt hm2/hm2_ztio.0: IDRom:
Aug 30 13:06:18 mksocfpga msgd:0: hal_lib:11718:rt hm2/hm2_ztio.0:     IDRom Type: 0x00000003
Aug 30 13:06:18 mksocfpga msgd:0: hal_lib:11718:rt hm2/hm2_ztio.0:     Offset to Modules: 0x00000040
Aug 30 13:06:18 mksocfpga msgd:0: hal_lib:11718:rt hm2/hm2_ztio.0:     Offset to Pin Description: 0x000001C0
Aug 30 13:06:18 mksocfpga msgd:0: hal_lib:11718:rt hm2/hm2_ztio.0:     Board Name: MYIRZTIO
Aug 30 13:06:18 mksocfpga msgd:0: hal_lib:11718:rt hm2/hm2_ztio.0:     FPGA Size: 8
Aug 30 13:06:18 mksocfpga msgd:0: hal_lib:11718:rt hm2/hm2_ztio.0:     FPGA Pins: 144
Aug 30 13:06:18 mksocfpga msgd:0: hal_lib:11718:rt hm2/hm2_ztio.0:     Port Width: 16
Aug 30 13:06:18 mksocfpga msgd:0: hal_lib:11718:rt hm2/hm2_ztio.0:     IO Ports: 4
Aug 30 13:06:18 mksocfpga msgd:0: hal_lib:11718:rt hm2/hm2_ztio.0:     IO Width: 64
Aug 30 13:06:18 mksocfpga msgd:0: hal_lib:11718:rt hm2/hm2_ztio.0:     Clock Low: 100000000 Hz (100000 KHz, 100 MHz)
Aug 30 13:06:18 mksocfpga msgd:0: hal_lib:11718:rt hm2/hm2_ztio.0:     Clock High: 200000000 Hz (200000 KHz, 200 MHz)
Aug 30 13:06:18 mksocfpga msgd:0: hal_lib:11718:rt hm2/hm2_ztio.0:     Instance Stride 0: 0x00000004
Aug 30 13:06:18 mksocfpga msgd:0: hal_lib:11718:rt hm2/hm2_ztio.0:     Instance Stride 1: 0x00000040
Aug 30 13:06:18 mksocfpga msgd:0: hal_lib:11718:rt hm2/hm2_ztio.0:     Register Stride 0: 0x00000100
Aug 30 13:06:18 mksocfpga msgd:0: hal_lib:11718:rt hm2/hm2_ztio.0:     Register Stride 1: 0x00000100
Aug 30 13:06:18 mksocfpga msgd:0: hal_lib:11718:rt hm2/hm2_ztio.0: IDROM IOPorts is 4 but llio num_ioport_connectors is 5, driver and firmware are inconsistent, aborting driver load  (<-- suppress error exit here)
Aug 30 13:06:18 mksocfpga msgd:0: hal_lib:11718:rt hm2/hm2_ztio.0: 64 HM2 Pin Descriptors:
Aug 30 13:06:18 mksocfpga msgd:0: hal_lib:11718:rt hm2/hm2_ztio.0:     pin 0:
Aug 30 13:06:18 mksocfpga msgd:0: hal_lib:11718:rt hm2/hm2_ztio.0:         Primary Tag: 0x03 (IOPort)
Aug 30 13:06:18 mksocfpga msgd:0: hal_lib:11718:rt hm2/hm2_ztio.0:     pin 1:
Aug 30 13:06:18 mksocfpga msgd:0: hal_lib:11718:rt hm2/hm2_ztio.0:         Primary Tag: 0x03 (IOPort)
Aug 30 13:06:18 mksocfpga msgd:0: hal_lib:11718:rt hm2/hm2_ztio.0:     pin 2:
Aug 30 13:06:18 mksocfpga msgd:0: hal_lib:11718:rt hm2/hm2_ztio.0:         Primary Tag: 0x03 (IOPort)
Aug 30 13:06:18 mksocfpga msgd:0: hal_lib:11718:rt hm2/hm2_ztio.0:     pin 3:
Aug 30 13:06:18 mksocfpga msgd:0: hal_lib:11718:rt hm2/hm2_ztio.0:         Primary Tag: 0x03 (IOPort)
Aug 30 13:06:18 mksocfpga msgd:0: hal_lib:11718:rt hm2/hm2_ztio.0:     pin 4:
Aug 30 13:06:18 mksocfpga msgd:0: hal_lib:11718:rt hm2/hm2_ztio.0:         Primary Tag: 0x03 (IOPort)
Aug 30 13:06:18 mksocfpga msgd:0: hal_lib:11718:rt hm2/hm2_ztio.0:         Secondary Tag: 0x05 (StepGen)
Aug 30 13:06:18 mksocfpga msgd:0: hal_lib:11718:rt hm2/hm2_ztio.0:         Secondary Unit: 0x00
Aug 30 13:06:18 mksocfpga msgd:0: hal_lib:11718:rt hm2/hm2_ztio.0:         Secondary Pin: 0x82 (Direction, Output)
[... cut ...]
- the [CONFIG.IOPorts, CONFIG.IOWidth, CONFIG.Portwidth and CONFIG.WIDTH](https://github.com/mhaberler/mksocfpga/blob/pins64/HW/VivadoProjects/zturn/zturn_ztio64/scripts/zturn_ztio_bd.tcl#L113-L125) values

- fwid.c just takes the [size of the first connector for pins_per_connector](https://github.com/mhaberler/machinekit/blob/master/src/hal/drivers/mesa-hostmot2/fwid.c#L137-L141) which is silly but I did not know better

I'm happy to teach fwid.c or hostmot2 in general to be more sensible with different-width connectors, but frankly I'm lost with the parameters :-/

dkhughes, 2016-08-30:
These parameters refer to the hostmot2 wrapper:

`CONFIG.IOPorts

The number of IO ports exposed by HM2. hostmot2 assumes all of the ports are even width.

`CONFIG.IOWidth

The total number of IO pins exposed (usually IOPorts * PortWidth)

`CONFIG.PortWidth

The width of an individual port

And this parameter is a member of the input output bus tristate component:

`CONFIG.WIDTH

The total width of the IOBits bus, should equal IOWidth

dkhughes, 2016-08-30:
```
Aug 30 13:06:18 mksocfpga msgd:0: hal_lib:11718:rt hm2/hm2_ztio.0: IDROM IOPorts is 4 but MD IOPort NumInstances is 2, inconsistent firmware, aborting driver load
Aug 30 13:06:18 mksocfpga msgd:0: hal_lib:11718:rt hm2/hm2_ztio.0: failed to parse Module Descriptor 2
```

This says your fwid message has a mismatch with the IDROM.

dkhughes, 2016-08-30:
Oh, the mismatch is in the PIN file here [https://github.com/mhaberler/mksocfpga/blob/pins64/HW/VivadoProjects/zturn/zturn_ztio64/const/PIN_ZTIO_64.vhd#L80](https://github.com/mhaberler/mksocfpga/blob/pins64/HW/VivadoProjects/zturn/zturn_ztio64/const/PIN_ZTIO_64.vhd#L80):

It should read:

```
 (IOPortTag,	x"00",	ClockLowTag,    x"04",	PortAddr&PadT,                  IOPortNumRegs,          x"00",	IOPortMPBitMask),
```

mhaberler, 2016-08-30:
thanks, applied fix!

(hopefully last) questions:

- assuming 64pins and 3 Leds - CONFIG.IOWidth == CONFIG.WIDTH = 64 (or 67?)
- assume 64pins - what is the impact of CONFIG.IOPorts=16 + CONFIG.IOPorts=4 vs say CONFIG.IOPorts=32 + CONFIG.IOPorts=2 (if any)?  from reading the [regmap](http://freeby.mesanet.com/regmap) it looks 1 I/O port register covers 24pins max?

`````
0x1000	I/O port  0..23
0x1004  I/O port 24..47
0x1008  I/O port 48..71
0x100C  I/O port 72..95
0x1010  I/O port 96..127
0x1014  I/O port 128..143
`````
- assuming CONFIG.IOPorts=16 + CONFIG.IOPorts=4 - does this relate to the [connectors in the fwid message](CONFIG.IOPorts=16 + CONFIG.IOPorts=4 ) _at all_?

dkhughes, 2016-08-30:
LEDs are independent of the IO pins. So, to change the LED count, you adjust the CONFIG.NumLEDs(?) parameter, and then make that match in the fwid. For your example CONFIG.WIDTH = 64.

> assume 64pins - what is the impact of CONFIG.IOPorts=16 + CONFIG.IOPorts=4 vs say CONFIG.IOPorts=32 + CONFIG.IOPorts=2 (if any)? from reading the regmap it looks 1 I/O port register covers 24pins max?

I thought the number of IO pins per port was limited to the bus width of the device, but the register map may disagree. For me, I would prefer 2 ports at 32 width because it saves reads/writes from the processor to fpga fabric (which is slow). But, if you're not controlling fast IO, it might not be an issue. 

> assuming CONFIG.IOPorts=16 + CONFIG.IOPorts=4 - does this relate to the connectors in the fwid message at all?

Yes it should. The number of ports in the fwid message should match the number of ports in the PIN file descriptor which is read in at run time, or the driver will fail because it needs to know how many IO ports it should expect for the hal config being loaded. Allocation of the proper address space (e.g., do I need 4 addresses or only 2 to talk to my outputs, etc.), I'm pretty sure is based off of the count of the IOPorts in the module descriptors. That means we could probably eliminate the IO mismatch guard and provided the correct firmware is loaded, it should work just fine. I like the config double check though, it's not much work and only happens at startup.

dkhughes, 2016-08-30:
Oh, btw, when you edit the config file directly, you need to account for the new pin connections. That's why I mentioned it is easier to use the GUI to generate a script you can then diff against the git commited command line script.