machinekit/mksocfpga #96: socfpga-rbf packaging fails with: FATAL: Failed to pull Docker image mhaberler/ubker

the-snowwhite, 2018-05-27:
linked to this PR:
https://github.com/machinekit/mksocfpga/pull/94

the-snowwhite, 2018-05-27:
Console output from Jenkins:
https://jenkins.machinekit.io/job/mksocfpga-packaging-quartus/31/console

Where to find a backup of this docker file ?

Who is running the server,,, is that the next thing that gets lost ?
+ the repo ?

the-snowwhite, 2018-05-27:
Turns out I have a fork and this is the original source:
https://github.com/dkhughes/ubker-docker
Now what todo:
@dkhughes 
@ArcEye 
help ?

ArcEye, 2018-05-28:
This should not have happened, I changed the builder to reference https://hub.docker.com/r/arceye/ubker/

mhaberler/ubker never existed anywhere but the server and when it disappeared I had to make a new one.

I will look later.

ArcEye, 2018-05-28:
I don't know how the build configuration got reverted back to attempt to pull `mhaberler/ubker`.

Lots of screwy things have happened in the past in Jenkins and it looks to be continuing.

I have changed it back to pull `arceye/ubker:jenkins-1030-1032-2` which was already present and works.

The Jenkins job finishes with an 'unstable' result code, due apparently to permission problems with the jenkins user, but it creates the package and uploads it, so I can't be bothered to try to fix that.

Package at http://deb.machinekit.io/debian/pool/main/s/socfpga-rbf/socfpga-rbf_3.194_all.deb

the-snowwhite, 2018-05-29:
Sorry but the new bitfiles are still not online:
looks like it is just packaging the former build:

    machinekit@mksocfpga-nano-soc:~$ apt search socfpga-rbf
    Sorting... Done
    Full Text Search... Done
    socfpga-rbf/stable,now 3.193 all [installed]
      hostmot2 Altera FPGA binaries for CycloneV platforms, https://jenkins.machinekit.io/job/mksocfpga- 
   packaging-quartus/30/

    machinekit@mksocfpga-nano-soc:~$ ls /lib/firmware/socfpga
    DE0_Nano_SoC_Cramps.3x24_cap.rbf      DE0_Nano_SoC_DB25.4s6p1q_GPIO_GPIO_GPIO.rbf  
    DE0_Nano_SoC_DB25.7I77_7I76_GPIO_GPIO.rbf  DE10_Nano_FB_Cramps.3x24_cap.rbf      dtbo
    DE0_Nano_SoC_Cramps.3x24_cap_spi.rbf  DE0_Nano_SoC_DB25.7I76_7I76_7I76_7I76.rbf    
    DE0_Nano_SoC_DB25.A_Volkov.rbf             DE10_Nano_FB_Cramps.3x24_cap_spi.rbf
    DE0_Nano_SoC_Cramps.3x24.rbf          DE0_Nano_SoC_DB25.7I76_7I85S_GPIO_GPIO.rbf   
    DE0_Nano_SoC_DB25.G540_G540_GPIO_GPIO.rbf  DE10_Nano_FB_Cramps.3x24.rbf

I am missing:

    DE10_Nano_SoC_FB_DB25.4s6p1q_GPIO_GPIO_GPIO.rbf  
    DE10_Nano_SoC_FB_DB25.7I76_7I85S_GPIO_GPIO.rbf  DE10_Nano_SoC_FB_DB25.A_Volkov.rbf
    DE10_Nano_SoC_FB_DB25.7I76_7I76_7I76_7I76.rbf    
    DE10_Nano_SoC_FB_DB25.7I77_7I76_GPIO_GPIO.rbf   
    DE10_Nano_SoC_FB_DB25.G540_G540_GPIO_GPIO.rbf

the-snowwhite, 2018-05-29:
I just triggered a build, and result was it wiped out the newest quartus build job #31 and reverted back to #30 
I will now retrigger the quartus build job and see if that gets the new files out.

ArcEye, 2018-05-29:
The files you are looking for WERE produced.

I suspect that because they have new names and folders and you haven't altered the packaging build to use those files, they are simply ignored.

the-snowwhite, 2018-05-29:
I was unable to see your edit in the config change log, and taking a copy of the project lead to the old mharber/uber setting.
I found this build setting I deleted and then did apply .. save.
And that edit showed up in the config list.
[image]

the-snowwhite, 2018-05-29:
Ups:

   I suspect that because they have new names and folders and you haven't altered the packaging build to use those files, they are simply ignored.

I hadn't thought of that.

the-snowwhite, 2018-05-29:
Can you point to where thats done ?

ArcEye, 2018-05-29:
I had nothing to do with this Jenkins job.

It uses `fpm` to package and the files to be packaged are contained in `${STAGING}`
You will see other files get copied from /SW/HW into a dir below this and then a Makefile is run.

When the Quartus build is finished, will be a bit clearer as to what is created and what is passed to the
packaging job.
Can't check properly until both have finished, so don't know if the new files are amongst the `artifacts`
which pass from builder to packager, or if they get missed there.

ArcEye, 2018-05-29:
So long as the new files are captured by the path below, they should be in the archive, so it will just be a case of amending the packaging

[image]

ArcEye, 2018-05-29:
AFAICT, all the files were created and zipped, all were transferred and all were included in the package.

The package that was just created has not been added to the repo, because it is identical to the one already in it.

A `dpkg -c` on both packages shows identical contents
```
drwxr-xr-x 0/0               0 2018-05-29 15:05 ./
drwxr-xr-x 0/0               0 2018-05-29 15:05 ./usr/
drwxr-xr-x 0/0               0 2018-05-29 15:05 ./usr/share/
drwxr-xr-x 0/0               0 2018-05-29 15:05 ./usr/share/doc/
drwxr-xr-x 0/0               0 2018-05-29 15:05 ./usr/share/doc/socfpga-rbf/
-rw-r--r-- 0/0             144 2018-05-29 15:05 ./usr/share/doc/socfpga-rbf/changelog.gz
drwxr-xr-x 0/0               0 2018-05-29 15:05 ./lib/
drwxr-xr-x 0/0               0 2018-05-29 15:05 ./lib/firmware/
drwxr-xr-x 0/0               0 2018-05-29 15:05 ./lib/firmware/socfpga/
drwxr-xr-x 0/0               0 2018-05-29 15:05 ./lib/firmware/socfpga/dtbo/
-rw-r--r-- 0/0            1007 2018-05-29 15:05 ./lib/firmware/socfpga/dtbo/template.dts
-rw-r--r-- 0/0             940 2018-05-29 15:05 ./lib/firmware/socfpga/dtbo/Makefile
-rw-r--r-- 0/0            1050 2018-05-29 15:05 ./lib/firmware/socfpga/dtbo/DE10_Nano_SoC_FB_DB25.G540_G540_GPIO_GPIO.dts
-rw-r--r-- 0/0            1005 2018-05-29 15:05 ./lib/firmware/socfpga/dtbo/DE10_Nano_SoC_FB_DB25.G540_G540_GPIO_GPIO.dtbo
-rw-r--r-- 0/0            1039 2018-05-29 15:05 ./lib/firmware/socfpga/dtbo/DE10_Nano_SoC_FB_DB25.A_Volkov.dts
-rw-r--r-- 0/0             993 2018-05-29 15:05 ./lib/firmware/socfpga/dtbo/DE10_Nano_SoC_FB_DB25.A_Volkov.dtbo
-rw-r--r-- 0/0            1050 2018-05-29 15:05 ./lib/firmware/socfpga/dtbo/DE10_Nano_SoC_FB_DB25.7I77_7I76_GPIO_GPIO.dts
-rw-r--r-- 0/0            1005 2018-05-29 15:05 ./lib/firmware/socfpga/dtbo/DE10_Nano_SoC_FB_DB25.7I77_7I76_GPIO_GPIO.dtbo
-rw-r--r-- 0/0            1051 2018-05-29 15:05 ./lib/firmware/socfpga/dtbo/DE10_Nano_SoC_FB_DB25.7I76_7I85S_GPIO_GPIO.dts
-rw-r--r-- 0/0            1005 2018-05-29 15:05 ./lib/firmware/socfpga/dtbo/DE10_Nano_SoC_FB_DB25.7I76_7I85S_GPIO_GPIO.dtbo
-rw-r--r-- 0/0            1050 2018-05-29 15:05 ./lib/firmware/socfpga/dtbo/DE10_Nano_SoC_FB_DB25.7I76_7I76_7I76_7I76.dts
-rw-r--r-- 0/0            1005 2018-05-29 15:05 ./lib/firmware/socfpga/dtbo/DE10_Nano_SoC_FB_DB25.7I76_7I76_7I76_7I76.dtbo
-rw-r--r-- 0/0            1052 2018-05-29 15:05 ./lib/firmware/socfpga/dtbo/DE10_Nano_SoC_FB_DB25.4s6p1q_GPIO_GPIO_GPIO.dts
-rw-r--r-- 0/0            1005 2018-05-29 15:05 ./lib/firmware/socfpga/dtbo/DE10_Nano_SoC_FB_DB25.4s6p1q_GPIO_GPIO_GPIO.dtbo
-rw-r--r-- 0/0            1041 2018-05-29 15:05 ./lib/firmware/socfpga/dtbo/DE10_Nano_FB_Cramps.3x24_cap_spi.dts
-rw-r--r-- 0/0             997 2018-05-29 15:05 ./lib/firmware/socfpga/dtbo/DE10_Nano_FB_Cramps.3x24_cap_spi.dtbo
-rw-r--r-- 0/0            1037 2018-05-29 15:05 ./lib/firmware/socfpga/dtbo/DE10_Nano_FB_Cramps.3x24_cap.dts
-rw-r--r-- 0/0             993 2018-05-29 15:05 ./lib/firmware/socfpga/dtbo/DE10_Nano_FB_Cramps.3x24_cap.dtbo
-rw-r--r-- 0/0            1033 2018-05-29 15:05 ./lib/firmware/socfpga/dtbo/DE10_Nano_FB_Cramps.3x24.dts
-rw-r--r-- 0/0             989 2018-05-29 15:05 ./lib/firmware/socfpga/dtbo/DE10_Nano_FB_Cramps.3x24.dtbo
-rw-r--r-- 0/0            1046 2018-05-29 15:05 ./lib/firmware/socfpga/dtbo/DE0_Nano_SoC_DB25.G540_G540_GPIO_GPIO.dts
-rw-r--r-- 0/0            1001 2018-05-29 15:05 ./lib/firmware/socfpga/dtbo/DE0_Nano_SoC_DB25.G540_G540_GPIO_GPIO.dtbo
-rw-r--r-- 0/0            1035 2018-05-29 15:05 ./lib/firmware/socfpga/dtbo/DE0_Nano_SoC_DB25.A_Volkov.dts
-rw-r--r-- 0/0             989 2018-05-29 15:05 ./lib/firmware/socfpga/dtbo/DE0_Nano_SoC_DB25.A_Volkov.dtbo
-rw-r--r-- 0/0            1046 2018-05-29 15:05 ./lib/firmware/socfpga/dtbo/DE0_Nano_SoC_DB25.7I77_7I76_GPIO_GPIO.dts
-rw-r--r-- 0/0            1001 2018-05-29 15:05 ./lib/firmware/socfpga/dtbo/DE0_Nano_SoC_DB25.7I77_7I76_GPIO_GPIO.dtbo
-rw-r--r-- 0/0            1047 2018-05-29 15:05 ./lib/firmware/socfpga/dtbo/DE0_Nano_SoC_DB25.7I76_7I85S_GPIO_GPIO.dts
-rw-r--r-- 0/0            1001 2018-05-29 15:05 ./lib/firmware/socfpga/dtbo/DE0_Nano_SoC_DB25.7I76_7I85S_GPIO_GPIO.dtbo
-rw-r--r-- 0/0            1046 2018-05-29 15:05 ./lib/firmware/socfpga/dtbo/DE0_Nano_SoC_DB25.7I76_7I76_7I76_7I76.dts
-rw-r--r-- 0/0            1001 2018-05-29 15:05 ./lib/firmware/socfpga/dtbo/DE0_Nano_SoC_DB25.7I76_7I76_7I76_7I76.dtbo
-rw-r--r-- 0/0            1048 2018-05-29 15:05 ./lib/firmware/socfpga/dtbo/DE0_Nano_SoC_DB25.4s6p1q_GPIO_GPIO_GPIO.dts
-rw-r--r-- 0/0            1001 2018-05-29 15:05 ./lib/firmware/socfpga/dtbo/DE0_Nano_SoC_DB25.4s6p1q_GPIO_GPIO_GPIO.dtbo
-rw-r--r-- 0/0            1041 2018-05-29 15:05 ./lib/firmware/socfpga/dtbo/DE0_Nano_SoC_Cramps.3x24_cap_spi.dts
-rw-r--r-- 0/0             997 2018-05-29 15:05 ./lib/firmware/socfpga/dtbo/DE0_Nano_SoC_Cramps.3x24_cap_spi.dtbo
-rw-r--r-- 0/0            1037 2018-05-29 15:05 ./lib/firmware/socfpga/dtbo/DE0_Nano_SoC_Cramps.3x24_cap.dts
-rw-r--r-- 0/0             993 2018-05-29 15:05 ./lib/firmware/socfpga/dtbo/DE0_Nano_SoC_Cramps.3x24_cap.dtbo
-rw-r--r-- 0/0            1033 2018-05-29 15:05 ./lib/firmware/socfpga/dtbo/DE0_Nano_SoC_Cramps.3x24.dts
-rw-r--r-- 0/0             989 2018-05-29 15:05 ./lib/firmware/socfpga/dtbo/DE0_Nano_SoC_Cramps.3x24.dtbo
-rw-r--r-- 0/0         2491360 2018-05-29 15:05 ./lib/firmware/socfpga/DE10_Nano_SoC_FB_DB25.G540_G540_GPIO_GPIO.rbf
[... cut ...]
-rw-r--r-- 0/0         1691816 2018-05-29 13:29 ./lib/firmware/socfpga/DE0_Nano_SoC_Cramps.3x24_cap_spi.rbf
-rw-r--r-- 0/0         1693592 2018-05-29 13:50 ./lib/firmware/socfpga/DE0_Nano_SoC_Cramps.3x24_cap.rbf
-rw-r--r-- 0/0         1669956 2018-05-29 14:14 ./lib/firmware/socfpga/DE0_Nano_SoC_Cramps.3x24.rbf
drwxr-xr-x 0/0               0 2018-05-29 15:05 ./etc/
drwxr-xr-x 0/0               0 2018-05-29 15:05 ./etc/modprobe.d/
-rw-r--r-- 0/0              50 2018-05-29 15:05 ./etc/modprobe.d/uiohm2.conf
```

the-snowwhite, 2018-05-29:
I changed the dist name to stretch instead of jessie

after sudo apt update i did:

    machinekit@mksocfpga-nano-soc:~$ apt-cache madison socfpga-rbf
    socfpga-rbf |      3.193 | http://deb.machinekit.io/debian stretch/main armhf Packages

then sudo apt install --reinstall socfpga-rbf

So yes the new package does not get added to the repo.

I have noticed an odd (sounds like last time we had that fight)error message at bottom of konsole output.

    
    + /scripts/manual_upload.sh --import /var/lib/jenkins/workspace/mksocfpga-packaging-quartus/debs --distribution stretch --user mkjail --pass ******** 86.59.12.252
    Could not create directory '/home/jenkins/.ssh'.

    Failed to add the host to the list of known hosts (/home/jenkins/.ssh/known_hosts).

    sftp> bye
    Could not create directory '/home/jenkins/.ssh'.

    Failed to add the host to the list of known hosts (/home/jenkins/.ssh/known_hosts).

the-snowwhite, 2018-05-29:
Here are the notes from last time:
https://github.com/machinekit/mksocfpga/pull/92#issuecomment-373943916

ArcEye, 2018-05-29:
I don't quite understand what you are saying.

The package produced by the build yesterday and today is 3.194
http://deb.machinekit.io/debian/pool/main/s/socfpga-rbf/socfpga-rbf_3.194_all.deb
It does have all the new files.  

I think you are saying that Stretch does not have this file in its package list.
That appears to be correct.

I have just added it manually and the Packages file for Stretch now has it, so an `apt update` should sort it out.
```
Package: socfpga-rbf
Version: 3.194
License: GNU General Public License (GPL), version 2.0 or later
Vendor: jenkins@machinekit.io
Architecture: all
Maintainer: jenkins@machinekit.io
Installed-Size: 36594
Homepage: https://github.com/machinekit/mksocfpga.git
Priority: extra
Section: default
Filename: pool/main/s/socfpga-rbf/socfpga-rbf_3.194_all.deb
Size: 12561434
SHA256: e5dd9a0dfe6105fb28dc4f11b8e5c75b76a82ac559cda7b092983aa99eb118d2
SHA1: 4963d30aac299c5837cdc5ec1a5f4c74488c5e69
MD5sum: 08fa57d7aa29e1eb8ab54d9d4c9cea3a
Description: hostmot2 Altera FPGA binaries for CycloneV platforms, https://jenkins.machinekit.io/job/mksocfpga-packaging-quartus/33/
```

@the-snowwhite 
> I have noticed an odd (sounds like last time we had that fight)error message at bottom of konsole output.
```
+ /scripts/manual_upload.sh --import /var/lib/jenkins/workspace/mksocfpga-packaging-quartus/debs --distribution stretch --user mkjail --pass ******** 86.59.12.252
Could not create directory '/home/jenkins/.ssh'.
Failed to add the host to the list of known hosts (/home/jenkins/.ssh/known_hosts).
sftp> bye
Could not create directory '/home/jenkins/.ssh'.
Failed to add the host to the list of known hosts (/home/jenkins/.ssh/known_hosts).
```

The error message arises from the fact that `jenkins` is a user without a home directory and some 
programs (ssh) do not expect to be run by such a user.

I don't think it matters at all, it is a warning that's all.

the-snowwhite, 2018-05-29:
I cannot find this file:

    /scripts/manual_upload.sh
in the workspace.
Is it supposed to be in the mksocfpga repo ?

ArcEye, 2018-05-30:
Looks like there is no specific provision for packages containing `_all.deb` in the name string.

The tests are based upon finding one of the supported distros names in the string, so all other packages are detected and added correctly.

Added such a test additionally to existing ones, see if that cures the problem on next build run.

Simpler than trying to mess with the convoluted logic of the multiple scripts involved, from which it is far from clear why the package was added to Jessie and not Stretch, despite your changing the `distro` tag to `stretch` in the upload script.

the-snowwhite, 2018-06-03:
Ok
I'm close to pushing another PR.

ArcEye, 2018-06-04:
When you do, mark it for my 'review'.
Then I can merge when able to watch it through and take remedial action if required.

the-snowwhite, 2018-06-05:
OK next PR in pipeline

ArcEye, 2018-06-08:
@the-snowwhite 
Subject to your testing, the changes I made seem to have ensured the package now appears to be automatically included in the Stretch Packages list
```
Package: socfpga-rbf
Version: 3.195
License: GNU General Public License (GPL), version 2.0 or later
Vendor: jenkins@machinekit.io
Architecture: all
Maintainer: jenkins@machinekit.io
Installed-Size: 36191
Homepage: https://github.com/machinekit/mksocfpga.git
Priority: extra
Section: default
Filename: pool/main/s/socfpga-rbf/socfpga-rbf_3.195_all.deb
Size: 12151092
SHA256: fc2b5babc89c3865033b177109e11af6b7900e42a3c555112448885eb327d3fb
SHA1: 313b449fe88d964eec4210104d69217a089575d5
MD5sum: 930d23f153f99a464d18a653291453d1
Description: hostmot2 Altera FPGA binaries for CycloneV platforms, https://jenkins.machinekit.io/job/mksocfpga-packaging-quartus/38/
```
Hopefully matter now resolved in full. :clinking_glasses:

closed 2019-05-09 by the-snowwhite