machinekit/mksocfpga #100: License Free partial reconfiguration

the-snowwhite, 2019-04-18:
@cdsteinkuehler 
@mhaberler 
@ArcEye 
@machinekoder 

Just a small heads up on something on my TODO list comming true:

Finally :-) 
IntelFPGA in their upcomming 19.1 Lite release (not yet downloadable), feature partial fpga reconfiguration support for cyclone V devices (license free). 

Other features are the HLS compiler (tool to accelerete c code in the fpga fabric).

https://www.intel.com/content/www/us/en/software/programmable/quartus-prime/download.html

[Intel_Quartus_19.1_features_Devices.pdf](https://github.com/machinekit/mksocfpga/files/3095155/Intel_Quartus_19.1_features_Devices.pdf)
[Intel_Quartus_19.1_features_Design_flow.pdf](https://github.com/machinekit/mksocfpga/files/3095156/Intel_Quartus_19.1_features_Design_flow.pdf)

https://groups.google.com/forum/#!topic/machinekit/BPRQpoyvFm8%5B1-25%5D

https://groups.google.com/forum/#!searchin/machinekit/partial$20reconfiguration|sort:date/machinekit/C_EKyemnE5Q/Ub6ImxWGAwAJ

https://groups.google.com/forum/#!searchin/machinekit/partial$20reconfiguration|sort:date/machinekit/C8xyZ286YOU/8o-pz1f2BAAJ

CycloneV_partial-reconfig-demo.tar.bz2: (proof of partial reconfiguration working in quartus 17.1 see git log)
https://drive.google.com/open?id=0BwyLvgyVIdi8M2gwalBkUG43YkU
https://drive.google.com/file/d/0BwyLvgyVIdi8M2gwalBkUG43YkU/view?usp=sharing

the-snowwhite, 2019-09-07:
So it was Xilinx that got the ball rolling this summer:
[Site](https://www.xilinx.com/products/design-tools/vivado/implementation/partial-reconfiguration.html)
[Docs and quick vids](https://www.xilinx.com/support/documentation-navigation/design-hubs/dh0017-vivado-partial-reconfiguration-hub.html)

Since the Ultra96 is an Ultrascale+ it comes with all the bell and wisles ... :-)

Yairk-ai, 2025-03-05:
Hey,

I have a Cyclone V SoC Development Kit, and I’m trying to implement a small Partial Reconfiguration (PR) project in Quartus Prime Standard Edition, but I’m a bit stuck. Maybe PR is not supported in this version? Could you help me, please?

From what I understand, you mentioned that if I download the Lite version, it will have full PR support for my board. Is that correct?

I really appreciate your help!

Thanks a lot!