machinekit/mksocfpga #102: Package builder reports success but fails to copy the new package into the repo

the-snowwhite, 2019-05-09:
@ArcEye 
Even though the socfpga-rbf packaga builder reports success
The new package does not show up in:
http://deb.machinekit.io/debian/pool/main/s/socfpga-rbf/

The package builder sucessfully creates the package:
Created package {:path=>"debs/socfpga-rbf_3.197_all.deb"}

Then maybe due to this error in the log:

    ++ pwd
    + /scripts/manual_upload.sh --import /var/lib/jenkins/workspace/mksocfpga-packaging-quartus/debs --distribution stretch --user mkjail --pass ******** 86.59.12.252
    Could not create directory '/home/jenkins/.ssh'.

    Failed to add the host to the list of known hosts (/home/jenkins/.ssh/known_hosts).

    sftp> bye
    Could not create directory '/home/jenkins/.ssh'.

    Failed to add the host to the list of known hosts (/home/jenkins/.ssh/known_hosts).

The new package does'nt get out

the-snowwhite, 2019-05-09:
It seems like its the post build script:
manual_upload.sh
That fails.
However this script seems not to be included in the mksocfpga repo.
But I can find it here:
machinekit-hal/.jenkins/manual_upload.sh

ArcEye, 2019-05-09:
I have just looked at adding the packages directly and the package database appears corrupted.

Leave it with me for a bit and I will rebuild it and add the new packages.

Then we can look at the upload script, which would only have put the packages under stretch anyway.

the-snowwhite, 2019-05-09:
Ok

the-snowwhite, 2019-05-09:
OK the package has showed up in the repo and

    sudo apt update
    sudo apt upgrade

worked as they should in stretch

the-snowwhite, 2019-05-09:
En furthur I don't see any reasons anyone should be in need of the socfpga-rbf package on jessie.
Have I overlooked something ?

ArcEye, 2019-05-09:
No, I just rebuilt with everything enabled for Stretch and Jessie.
The package is only in there once, so does not cost anything

the-snowwhite, 2019-05-09:
OK
Thanks for your once again great effort :-)

closed 2019-05-09 by the-snowwhite