machinekit/mksocfpga #105: new DeXX image -> hm2_soc_ol.so: undefined symbol: hm2_unregister

luminize, 2019-05-24:
@the-snowwhite I'm testing your image, I've got a De0 nano, downloaded the console image, followed instructions on setting up, installed the socfpga-rbf and machinekit-hal-rt-preempt packages.

do the following:
```
realtime start
halcmd newinst hm2_soc_ol hm2-socfpga0 config="firmware=/lib/firmware/socfpga/dtbo/DE0_Nano_SoC_DB25.7I76_7I76_7I76_7I76.dtbo"
```
This errors, I get the following error message `do_load_cmd: dlopen: /usr/lib/linuxcnc/rt-preempt/hm2_soc_ol.so: undefined symbol: hm2_unregister`

here's a part of /var/log/linuxcnc.log

```
May 24 18:02:55 mksocfpga-nano-soc rtapi:0: 4:rtapi_app:1988:user pid=1988 flavor=rt-preempt gcc=6.3.0 20170516 git=v0.2~-----~02beec8
May 24 18:02:55 mksocfpga-nano-soc rtapi:0: do_load_cmd: dlopen: /usr/lib/linuxcnc/rt-preempt/hm2_soc_ol.so: undefined symbol: hm2_unregister
May 24 18:02:55 mksocfpga-nano-soc rtapi:0: rpath=/usr/lib/linuxcnc/rt-preempt
May 24 18:02:55 mksocfpga-nano-soc rtapi:0: 1:rtapi_app:1988:user do_load_cmd: dlopen: /usr/lib/linuxcnc/rt-preempt/hm2_soc_ol.so: undefined symbol: hm2_unregister
May 24 18:02:55 mksocfpga-nano-soc rtapi:0: 1:rtapi_app:1988:user rpath=/usr/lib/linuxcnc/rt-preempt
May 24 18:02:55 mksocfpga-nano-soc msgd:0: ulapi:2336:user _ulapi_init(): ulapi rt-preempt v0.2~-----~02beec8 loaded
May 24 18:02:55 mksocfpga-nano-soc msgd:0: ulapi:2336:user halg_xinitfv:271 HAL: singleton component 'hal_lib2336' id=94 initialized
May 24 18:02:55 mksocfpga-nano-soc msgd:0: hal_lib:2336:user --halcmd newinst hm2_soc_ol hm2-socfpga0 config=firmware=/lib/firmware/socfpga/dtbo/DE0_Nano_SoC_DB25.7I76_7I76_7I76_7I76.dtbo
May 24 18:02:55 mksocfpga-nano-soc msgd:0: hal_lib:2336:user halg_exit:293 HAL: removing component 96 'halcmd2336'
May 24 18:02:55 mksocfpga-nano-soc msgd:0: hal_lib:2336:user ulapi_hal_lib_cleanup:235 HAL: lib_module_id=94
May 24 18:02:55 mksocfpga-nano-soc msgd:0: hal_lib:2336:user halg_exit:293 HAL: removing component 94 'hal_lib2336'
May 24 18:02:55 mksocfpga-nano-soc msgd:0: hal_lib:2336:user halg_exit:315 HAL: hal_errorcount()=0
May 24 18:02:55 mksocfpga-nano-soc msgd:0: hal_lib:2336:user halg_exit:316 HAL: _halerrno=0
```

luminize, 2019-05-24:
@the-snowwhite ok, got it sorted. It was the "loadrt hostmot2" I was missing, and the 3 dashes in newinst to get the instance parameters to load correctly.

Guess i was a bit rusty. I can load with the following hal file:
```
loadrt hostmot2
newinst hm2_soc_ol hm2-socfpga0 --- "config=firmware=socfpga/dtbo/DE0_Nano_SoC_DB25.7I76_7I76_7I76_7I76.dtbo"
```
and load as so:
```
machinekit@mksocfpga-nano-soc:~$ halrun -I soc.hal 
msgd:0 stopped
rtapi:0 stopped
rtapi_msgd command:  /usr/libexec/linuxcnc/rtapi_msgd --instance=0 --rtmsglevel=5 --usrmsglevel=5 --halsize=524288
rtapi_app command:  /usr/libexec/linuxcnc/rtapi_app_rt-preempt --instance=0
soc.hal:1: Realtime module 'hostmot2' loaded
soc.hal:2: Realtime module 'hm2_soc_ol' loaded
[  479.373463] fpga_manager fpga0: writing socfpga/DE0_Nano_SoC_DB25.7I76_7I76_7I76_7I76.rbf to Altera SOCFPGA FPGA Manager
```

closed 2019-05-24 by luminize

ArcEye, 2019-05-24:
There are a lot of undefined symbols in that module, it gets the hm2 ones from hostmot2
```
root@INTEL-i7:/downloads# nm -D hm2_soc_ol.so
         U __aeabi_unwind_cpp_pr0
         U __aeabi_unwind_cpp_pr1
         U close
         w __cxa_finalize
         U __errno_location
         U free
         w __gmon_start__
         U halg_exit
         U halg_free_single_str
         U halg_inst_create
         U halg_ready
         U halg_strdup
         U hal_xinit
         U hm2_register
         U hm2_unregister
         w _ITM_deregisterTMCloneTable
         w _ITM_registerTMCloneTable
         U malloc
         U memcpy
         U memset
         U mkdir
         U mmap
         U munmap
         U open
         U opendir
         U read
         U rmdir
00001c6d T rtapi_app_exit
00001c85 T rtapi_app_main
         U rtapi_fs_read
000150dc D rtapi_info_author
000150e0 D rtapi_info_description
000150d8 D rtapi_info_license
000150e4 D rtapi_info_supported_device
00015110 D rtapi_instinfo_address_already_programmed
000150ec D rtapi_instinfo_address_debug
000150f8 D rtapi_instinfo_address_no_init_llio
00015104 D rtapi_instinfo_address_num
00015114 D rtapi_instinfo_description_already_programmed
000150f0 D rtapi_instinfo_description_debug
000150fc D rtapi_instinfo_description_no_init_llio
00015108 D rtapi_instinfo_description_num
0001510c D rtapi_instinfo_type_already_programmed
000150e8 D rtapi_instinfo_type_debug
000150f4 D rtapi_instinfo_type_no_init_llio
00015100 D rtapi_instinfo_type_num
         U rtapi_print
         U rtapi_print_hex_dump
         U rtapi_print_msg
         U rtapi_snprintf
         U strdup
         U strerror
         U strlen
         U strncmp
         U tolower
         U usleep
         U write
         U __xstat
```

ArcEye, 2019-05-24:
Ooops, didn't refresh my screen, you have already sorted it.

luminize, 2019-05-24:
Heh, yeah. I was a bit rusty... but at least it’s preserved for posterity :)

the-snowwhite, 2019-05-31:
I wonder how you got it to work with 3 dashes I only use 2 ... :-)