machinekit/mksocfpga #107: SmartSerial broken in DE10_Nano_FB_Cramps project

blazini36, 2019-08-22:
I've pointed this out in the Machinekit Group but figured I'd post here as a reminder. The Mesa SmartSerial function appears to be broken in this project. I'm unable to test other projects so I'm not sure how far up the ladder it goes.

Tests were done with GPIO1 pins 00+01, as SS rx and tx, using a couple of different RS422 circuits trying to communicate with a Mesa 8i20. I was not able to get this to work. I reached out to PCW on the LinuxCNC forum and basically came to the conclusion that since the SmartSerial version printed on Machinekit startup shows as "version 0" it is broken at the firmware level. Machinekit does not print to the terminal the same way LinuxCNC does but the instantiation of HM2 modules will sometimes dump to a logfile in /var/logs/linuxcnc.log as seen below.

The MK discussion:
[https://groups.google.com/forum/#!topic/machinekit/eVhvTnuhblE%5B51-75%5D](url)

From Linuxcnc forum:
[https://forum.linuxcnc.org/27-driver-boards/36923-sserial-hardware-implementation?start=10](url)

/var/log/linuxcnc.log

`Jul 18 22:07:43 mksocfpga-nano-soc msgd:0: startup pid=2730 flavor=rt-preempt rtlevel=1 usrlevel=1 halsize=524288 shm=Posix cc=gcc 6.3.0 20170516  version=v0.2~-----~58fe987
Jul 18 22:07:43 mksocfpga-nano-soc msgd:0: ØMQ=4.2.1 czmq=4.0.2 protobuf=3.0.0 atomics=gcc intrinsics    libwebsockets=2.0.3
Jul 18 22:07:43 mksocfpga-nano-soc msgd:0: configured: sha=58fe987
Jul 18 22:07:43 mksocfpga-nano-soc msgd:0: built:      Mar 23 2019 18:41:31 sha=58fe987
Jul 18 22:07:43 mksocfpga-nano-soc msgd:0: register_stuff: actual hostname as announced by avahi='mksocfpga-nano-soc.local'
Jul 18 22:07:43 mksocfpga-nano-soc msgd:0: zeroconf: registering: 'Log service on mksocfpga-nano-soc.local pid 2730'
Jul 18 22:07:44 mksocfpga-nano-soc msgd:0: hal_lib:2745:user iocontrol: machine: 'ST_FPGA_SOC_DC1_test2'  version 'unknown'
Jul 18 22:07:44 mksocfpga-nano-soc msgd:0: zeroconf: registered 'Log service on mksocfpga-nano-soc.local pid 2730' _machinekit._tcp 0 TXT "uuid=abc21811-1fb4-4ba9-98f3-6ca4138d9f21" "instance=772a79a2-a9a8-11e9-a5ca-ba072231546c" "service=log" "dsn=ipc:///tmp/0.log.abc21811-1fb4-4ba9-98f3-6ca4138d9f21"
Jul 18 22:07:44 mksocfpga-nano-soc msgd:0: hal_lib:2736:rt hm2: loading Mesa HostMot2 driver version 0.15
Jul 18 22:07:44 mksocfpga-nano-soc msgd:0: hal_lib:2736:rt hm2_soc_ol: loading Mesa AnyIO HostMot2 socfpga overlay driver version 0.9
Jul 18 22:07:44 mksocfpga-nano-soc msgd:0: hal_lib:2736:rt hm2/hm2_5i25.0: Smart Serial Firmware Version 0
Jul 18 22:07:44 mksocfpga-nano-soc msgd:0: hal_lib:2736:rt hm2/hm2_5i25.0: 72 I/O Pins used:
Jul 18 22:07:44 mksocfpga-nano-soc msgd:0: hal_lib:2736:rt hm2/hm2_5i25.0:     IO Pin 000 (GPIO0.P0-01): IOPort
Jul 18 22:07:44 mksocfpga-nano-soc msgd:0: hal_lib:2736:rt hm2/hm2_5i25.0:     IO Pin 001 (GPIO0.P0-02): IOPort
Jul 18 22:07:44 mksocfpga-nano-soc msgd:0: hal_lib:2736:rt hm2/hm2_5i25.0:     IO Pin 002 (GPIO0.P0-03): IOPort
Jul 18 22:07:44 mksocfpga-nano-soc msgd:0: hal_lib:2736:rt hm2/hm2_5i25.0:     IO Pin 003 (GPIO0.P0-04): IOPort
Jul 18 22:07:44 mksocfpga-nano-soc msgd:0: hal_lib:2736:rt hm2/hm2_5i25.0:     IO Pin 004 (GPIO0.P0-05): IOPort
Jul 18 22:07:44 mksocfpga-nano-soc msgd:0: hal_lib:2736:rt hm2/hm2_5i25.0:     IO Pin 005 (GPIO0.P0-06): IOPort
Jul 18 22:07:44 mksocfpga-nano-soc msgd:0: hal_lib:2736:rt hm2/hm2_5i25.0:     IO Pin 006 (GPIO0.P0-07): IOPort`

the-snowwhite, 2019-08-22:
Sorry that this was/is not more clearly globally stated, however:
The DExx ..._Cramps projects are minimalistic as only the hm2 cores in current configs are included, so
SSerial and @blazini36 'c config added here:
https://github.com/the-snowwhite/mksocfpga/tree/sserial-work

the-snowwhite, 2019-08-24:
There is definately something (not) going on.
I noticed that the tx pin was floating (also in the ..._DB 25 7i77_7i77_7i77_7i77 bitfile)
All the SSerial pins actually show up and act/behave as ordinary GPIO pins. here a copy of the commands I used and also the linuxcnc debug output:

    machinekit@mksocfpga-nano-soc:~$ sudo sh -c 'echo  ""> /var/log/linuxcnc.log'
    machinekit@mksocfpga-nano-soc:~$ halrun
    msgd:0 stopped
    rtapi:0 stopped
    rtapi_msgd command:  /usr/libexec/linuxcnc/rtapi_msgd --instance=0 --rtmsglevel=1 --usrmsglevel=1 --halsize=524288
    rtapi_app command:  /usr/libexec/linuxcnc/rtapi_app_rt-preempt --instance=0
    halcmd: loadrt trivkins
    halcmd: loadrt tp
    halcmd: loadrt motmod servo_period_nsec=1000000 num_joints=3 num_aio=51 num_dio=21 tp=tp kins=trivkins
    halcmd: loadrt hostmot2 debug_idrom=1 debug_modules=1
    halcmd: newinst hm2_soc_ol hm2-socfpga0 already_programmed=1 -- config="num_stepgens=3 enable_adc=1  sserial_port_0=00xx" debug=1
    halcmd: addf hm2_5i25.0.read         servo-thread
    halcmd: addf hm2_5i25.0.write        servo-thread
    halcmd: start
    halcmd: setp hm2_5i25.0.gpio.036.is_output true
    halcmd: setp hm2_5i25.0.gpio.036.out 1
    halcmd: setp hm2_5i25.0.gpio.036.out 0
    halcmd: exit
    machinekit@mksocfpga-nano-soc:~$ cat /var/log/linuxcnc.log

    Aug 24 17:00:39 mksocfpga-nano-soc msgd:0: startup pid=2851 flavor=rt-preempt rtlevel=1 usrlevel=1 halsize=524288 shm=Posix cc=gcc 6.3.0 20170516  version=v0.2~-----~3008e4a
    Aug 24 17:00:39 mksocfpga-nano-soc msgd:0: ØMQ=4.2.1 czmq=4.0.2 protobuf=3.0.0 atomics=gcc intrinsics    libwebsockets=2.0.3
    Aug 24 17:00:39 mksocfpga-nano-soc msgd:0: configured: sha=3008e4a
    Aug 24 17:00:39 mksocfpga-nano-soc msgd:0: built:      Jun 15 2019 06:25:18 sha=3008e4a
    Aug 24 17:00:39 mksocfpga-nano-soc msgd:0: register_stuff: actual hostname as announced by avahi='mksocfpga-nano-soc.local'
    Aug 24 17:00:39 mksocfpga-nano-soc msgd:0: zeroconf: registering: 'Log service on mksocfpga-nano-soc.local pid 2851'
    Aug 24 17:00:40 mksocfpga-nano-soc msgd:0: zeroconf: registered 'Log service on mksocfpga-nano-soc.local pid 2851' _machinekit._tcp 49152 TXT "uuid=76731441-5f29-4b88-85a4-6fba2ec67cd7" "instance=ef659a34-c67f-11e9-b9d8-823af0754b92" "service=log" "dsn=tcp://mksocfpga-nano-soc.local:49152"
    Aug 24 17:01:43 mksocfpga-nano-soc msgd:0: hal_lib:2856:rt hm2: loading Mesa HostMot2 driver version 0.15
    Aug 24 17:03:02 mksocfpga-nano-soc msgd:0: hal_lib:2856:rt hm2_soc_ol: loading Mesa AnyIO HostMot2 socfpga overlay driver version 0.9
    Aug 24 17:03:02 mksocfpga-nano-soc msgd:0: hal_lib:2856:rt hm2/hm2_5i25.0: IDRom:
    Aug 24 17:03:02 mksocfpga-nano-soc msgd:0: hal_lib:2856:rt hm2/hm2_5i25.0:     IDRom Type: 0x00000003
    Aug 24 17:03:02 mksocfpga-nano-soc msgd:0: hal_lib:2856:rt hm2/hm2_5i25.0:     Offset to Modules: 0x00000040
    Aug 24 17:03:02 mksocfpga-nano-soc msgd:0: hal_lib:2856:rt hm2/hm2_5i25.0:     Offset to Pin Description: 0x000001C0
    Aug 24 17:03:02 mksocfpga-nano-soc msgd:0: hal_lib:2856:rt hm2/hm2_5i25.0:     Board Name: MESA5I25
    Aug 24 17:03:02 mksocfpga-nano-soc msgd:0: hal_lib:2856:rt hm2/hm2_5i25.0:     FPGA Size: 9
    Aug 24 17:03:02 mksocfpga-nano-soc msgd:0: hal_lib:2856:rt hm2/hm2_5i25.0:     FPGA Pins: 144
    Aug 24 17:03:02 mksocfpga-nano-soc msgd:0: hal_lib:2856:rt hm2/hm2_5i25.0:     Port Width: 24
    Aug 24 17:03:02 mksocfpga-nano-soc msgd:0: hal_lib:2856:rt hm2/hm2_5i25.0:     IO Ports: 3
    Aug 24 17:03:02 mksocfpga-nano-soc msgd:0: hal_lib:2856:rt hm2/hm2_5i25.0:     IO Width: 72
    Aug 24 17:03:02 mksocfpga-nano-soc msgd:0: hal_lib:2856:rt hm2/hm2_5i25.0:     Clock Low: 50000000 Hz (50000 KHz, 50 MHz)
    Aug 24 17:03:02 mksocfpga-nano-soc msgd:0: hal_lib:2856:rt hm2/hm2_5i25.0:     Clock High: 200000000 Hz (200000 KHz, 200 MHz)
    Aug 24 17:03:02 mksocfpga-nano-soc msgd:0: hal_lib:2856:rt hm2/hm2_5i25.0:     Instance Stride 0: 0x00000004
    Aug 24 17:03:02 mksocfpga-nano-soc msgd:0: hal_lib:2856:rt hm2/hm2_5i25.0:     Instance Stride 1: 0x00000040
    Aug 24 17:03:02 mksocfpga-nano-soc msgd:0: hal_lib:2856:rt hm2/hm2_5i25.0:     Register Stride 0: 0x00000100
    Aug 24 17:03:02 mksocfpga-nano-soc msgd:0: hal_lib:2856:rt hm2/hm2_5i25.0:     Register Stride 1: 0x00000100
    Aug 24 17:03:03 mksocfpga-nano-soc msgd:0: hal_lib:2856:rt hm2/hm2_5i25.0: Smart Serial Firmware Version 43
    Aug 24 17:03:03 mksocfpga-nano-soc msgd:0: hal_lib:2856:rt hm2/hm2_5i25.0: 72 I/O Pins used:
    Aug 24 17:03:03 mksocfpga-nano-soc msgd:0: hal_lib:2856:rt hm2/hm2_5i25.0:     IO Pin 000 (GPIO0.P0-01): IOPort
    Aug 24 17:03:03 mksocfpga-nano-soc msgd:0: hal_lib:2856:rt hm2/hm2_5i25.0:     IO Pin 001 (GPIO0.P0-02): IOPort
    Aug 24 17:03:03 mksocfpga-nano-soc msgd:0: hal_lib:2856:rt hm2/hm2_5i25.0:     IO Pin 002 (GPIO0.P0-03): IOPort
    Aug 24 17:03:03 mksocfpga-nano-soc msgd:0: hal_lib:2856:rt hm2/hm2_5i25.0:     IO Pin 003 (GPIO0.P0-04): IOPort
    Aug 24 17:03:03 mksocfpga-nano-soc msgd:0: hal_lib:2856:rt hm2/hm2_5i25.0:     IO Pin 004 (GPIO0.P0-05): IOPort
    Aug 24 17:03:03 mksocfpga-nano-soc msgd:0: hal_lib:2856:rt hm2/hm2_5i25.0:     IO Pin 005 (GPIO0.P0-06): IOPort
    Aug 24 17:03:03 mksocfpga-nano-soc msgd:0: hal_lib:2856:rt hm2/hm2_5i25.0:     IO Pin 006 (GPIO0.P0-07): IOPort
    Aug 24 17:03:03 mksocfpga-nano-soc msgd:0: hal_lib:2856:rt hm2/hm2_5i25.0:     IO Pin 007 (GPIO0.P0-08): IOPort
[... cut ...]
    Aug 24 17:04:15 mksocfpga-nano-soc msgd:0: hal_lib:2856:rt hm2: unloading
    Aug 24 17:04:15 mksocfpga-nano-soc msgd:0: rtapi_app exit detected - scheduled shutdown
    Aug 24 17:04:17 mksocfpga-nano-soc msgd:0: msgd shutting down
    Aug 24 17:04:17 mksocfpga-nano-soc msgd:0: zeroconf: unregistering 'Log service on mksocfpga-nano-soc.local pid 2851'
    Aug 24 17:04:17 mksocfpga-nano-soc msgd:0: log buffer hwm: 3% (206 msgs, 17882 bytes out of 524288)
    Aug 24 17:04:17 mksocfpga-nano-soc msgd:0: normal shutdown - global segment detached

the-snowwhite, 2019-08-24:
Behaivior differs on the /lib/firmware/socfpga/DE10_Nano_SoC_FB_DB25.7I76_7I76_7I76_7I76.rbf
config, as to I cannot set the:

         IOPortTag & x"00" & SSerialTag & SSerialTX0Pin,             -- I/O 10   GPIO_0 06   PIN 7
pin as gpio:
Commands here:

    sudo fw_setenv bitimage /lib/firmware/socfpga/DE10_Nano_SoC_FB_DB25.7I76_7I76_7I76_7I76.rbf
    sudo reboot

    <re login>

    sudo sh -c 'echo  ""> /var/log/linuxcnc.log'
    halrun
    loadrt trivkins
    loadrt tp
    loadrt motmod servo_period_nsec=1000000 num_joints=3 num_aio=51 num_dio=21 tp=tp kins=trivkins
    loadrt hostmot2 debug_idrom=1 debug_modules=1
    newinst hm2_soc_ol hm2-socfpga0 already_programmed=1 -- config="num_stepgens=3 sserial_port_0=00xx" debug=1
    addf hm2_de0n.0.read         servo-thread
    addf hm2_de0n.0.write        servo-thread
    start
    setp hm2_de0n.0.gpio.006.is_output true
    setp hm2_de0n.0.gpio.006.out 1
    setp hm2_de0n.0.gpio.006.out 0
    exit
    cat /var/log/linuxcnc.log

the-snowwhite, 2019-08-24:
log:
    machinekit@mksocfpga-nano-soc:~$ cat /var/log/linuxcnc.log

    Aug 24 17:57:07 mksocfpga-nano-soc msgd:0: startup pid=2651 flavor=rt-preempt rtlevel=1 usrlevel=1 halsize=524288 shm=Posix cc=gcc 6.3.0 20170516  version=v0.2~-----~3008e4a
    Aug 24 17:57:07 mksocfpga-nano-soc msgd:0: ØMQ=4.2.1 czmq=4.0.2 protobuf=3.0.0 atomics=gcc intrinsics    libwebsockets=2.0.3
    Aug 24 17:57:07 mksocfpga-nano-soc msgd:0: configured: sha=3008e4a
    Aug 24 17:57:07 mksocfpga-nano-soc msgd:0: built:      Jun 15 2019 06:25:18 sha=3008e4a
    Aug 24 17:57:07 mksocfpga-nano-soc msgd:0: register_stuff: actual hostname as announced by avahi='mksocfpga-nano-soc.local'
    Aug 24 17:57:07 mksocfpga-nano-soc msgd:0: zeroconf: registering: 'Log service on mksocfpga-nano-soc.local pid 2651'
    Aug 24 17:57:08 mksocfpga-nano-soc msgd:0: zeroconf: registered 'Log service on mksocfpga-nano-soc.local pid 2651' _machinekit._tcp 49152 TXT "uuid=76731441-5f29-4b88-85a4-6fba2ec67cd7" "instance=d30a921a-c687-11e9-a585-823af0754b92" "service=log" "dsn=tcp://mksocfpga-nano-soc.local:49152"
    Aug 24 17:57:08 mksocfpga-nano-soc msgd:0: hal_lib:2656:rt hm2: loading Mesa HostMot2 driver version 0.15
    Aug 24 17:57:15 mksocfpga-nano-soc msgd:0: hal_lib:2656:rt hm2_soc_ol: loading Mesa AnyIO HostMot2 socfpga overlay driver version 0.9
    Aug 24 17:57:15 mksocfpga-nano-soc msgd:0: hal_lib:2656:rt hm2/hm2_de0n.0: IDRom:
    Aug 24 17:57:15 mksocfpga-nano-soc msgd:0: hal_lib:2656:rt hm2/hm2_de0n.0:     IDRom Type: 0x00000003
    Aug 24 17:57:15 mksocfpga-nano-soc msgd:0: hal_lib:2656:rt hm2/hm2_de0n.0:     Offset to Modules: 0x00000040
    Aug 24 17:57:15 mksocfpga-nano-soc msgd:0: hal_lib:2656:rt hm2/hm2_de0n.0:     Offset to Pin Description: 0x000001C0
    Aug 24 17:57:15 mksocfpga-nano-soc msgd:0: hal_lib:2656:rt hm2/hm2_de0n.0:     Board Name: TERADE0N
    Aug 24 17:57:15 mksocfpga-nano-soc msgd:0: hal_lib:2656:rt hm2/hm2_de0n.0:     FPGA Size: 9
    Aug 24 17:57:15 mksocfpga-nano-soc msgd:0: hal_lib:2656:rt hm2/hm2_de0n.0:     FPGA Pins: 144
    Aug 24 17:57:15 mksocfpga-nano-soc msgd:0: hal_lib:2656:rt hm2/hm2_de0n.0:     Port Width: 17
    Aug 24 17:57:15 mksocfpga-nano-soc msgd:0: hal_lib:2656:rt hm2/hm2_de0n.0:     IO Ports: 4
    Aug 24 17:57:15 mksocfpga-nano-soc msgd:0: hal_lib:2656:rt hm2/hm2_de0n.0:     IO Width: 68
    Aug 24 17:57:15 mksocfpga-nano-soc msgd:0: hal_lib:2656:rt hm2/hm2_de0n.0:     Clock Low: 50000000 Hz (50000 KHz, 50 MHz)
    Aug 24 17:57:15 mksocfpga-nano-soc msgd:0: hal_lib:2656:rt hm2/hm2_de0n.0:     Clock High: 200000000 Hz (200000 KHz, 200 MHz)
    Aug 24 17:57:15 mksocfpga-nano-soc msgd:0: hal_lib:2656:rt hm2/hm2_de0n.0:     Instance Stride 0: 0x00000004
    Aug 24 17:57:15 mksocfpga-nano-soc msgd:0: hal_lib:2656:rt hm2/hm2_de0n.0:     Instance Stride 1: 0x00000040
    Aug 24 17:57:15 mksocfpga-nano-soc msgd:0: hal_lib:2656:rt hm2/hm2_de0n.0:     Register Stride 0: 0x00000100
    Aug 24 17:57:15 mksocfpga-nano-soc msgd:0: hal_lib:2656:rt hm2/hm2_de0n.0:     Register Stride 1: 0x00000100
    Aug 24 17:57:15 mksocfpga-nano-soc msgd:0: hal_lib:2656:rt hm2/hm2_de0n.0: Smart Serial Firmware Version 43
    Aug 24 17:57:15 mksocfpga-nano-soc msgd:0: hal_lib:2656:rt hm2/hm2_de0n.0: 68 I/O Pins used:
    Aug 24 17:57:15 mksocfpga-nano-soc msgd:0: hal_lib:2656:rt hm2/hm2_de0n.0:     IO Pin 000 (GPIO0.P2-01): StepGen #0, pin Direction (Output)
    Aug 24 17:57:15 mksocfpga-nano-soc msgd:0: hal_lib:2656:rt hm2/hm2_de0n.0:     IO Pin 001 (GPIO0.P2-02): StepGen #0, pin Step (Output)
    Aug 24 17:57:15 mksocfpga-nano-soc msgd:0: hal_lib:2656:rt hm2/hm2_de0n.0:     IO Pin 002 (GPIO0.P2-03): StepGen #1, pin Direction (Output)
    Aug 24 17:57:15 mksocfpga-nano-soc msgd:0: hal_lib:2656:rt hm2/hm2_de0n.0:     IO Pin 003 (GPIO0.P2-04): StepGen #1, pin Step (Output)
    Aug 24 17:57:15 mksocfpga-nano-soc msgd:0: hal_lib:2656:rt hm2/hm2_de0n.0:     IO Pin 004 (GPIO0.P2-05): StepGen #2, pin Direction (Output)
    Aug 24 17:57:15 mksocfpga-nano-soc msgd:0: hal_lib:2656:rt hm2/hm2_de0n.0:     IO Pin 005 (GPIO0.P2-06): StepGen #2, pin Step (Output)
    Aug 24 17:57:15 mksocfpga-nano-soc msgd:0: hal_lib:2656:rt hm2/hm2_de0n.0:     IO Pin 006 (GPIO0.P2-07): IOPort
    Aug 24 17:57:15 mksocfpga-nano-soc msgd:0: hal_lib:2656:rt hm2/hm2_de0n.0:     IO Pin 007 (GPIO0.P2-08): IOPort
    Aug 24 17:57:15 mksocfpga-nano-soc msgd:0: hal_lib:2656:rt hm2/hm2_de0n.0:     IO Pin 008 (GPIO0.P2-09): IOPort
    Aug 24 17:57:15 mksocfpga-nano-soc msgd:0: hal_lib:2656:rt hm2/hm2_de0n.0:     IO Pin 009 (GPIO0.P2-10): IOPort
    Aug 24 17:57:15 mksocfpga-nano-soc msgd:0: hal_lib:2656:rt hm2/hm2_de0n.0:     IO Pin 010 (GPIO0.P2-11): IOPort
    Aug 24 17:57:15 mksocfpga-nano-soc msgd:0: hal_lib:2656:rt hm2/hm2_de0n.0:     IO Pin 011 (GPIO0.P2-12): IOPort
    Aug 24 17:57:15 mksocfpga-nano-soc msgd:0: hal_lib:2656:rt hm2/hm2_de0n.0:     IO Pin 012 (GPIO0.P2-13): IOPort
    Aug 24 17:57:15 mksocfpga-nano-soc msgd:0: hal_lib:2656:rt hm2/hm2_de0n.0:     IO Pin 013 (GPIO0.P2-14): IOPort
    Aug 24 17:57:15 mksocfpga-nano-soc msgd:0: hal_lib:2656:rt hm2/hm2_de0n.0:     IO Pin 014 (GPIO0.P2-15): Encoder #0, pin Index (Input)
    Aug 24 17:57:15 mksocfpga-nano-soc msgd:0: hal_lib:2656:rt hm2/hm2_de0n.0:     IO Pin 015 (GPIO0.P2-16): Encoder #0, pin B (Input)
    Aug 24 17:57:15 mksocfpga-nano-soc msgd:0: hal_lib:2656:rt hm2/hm2_de0n.0:     IO Pin 016 (GPIO0.P2-17): Encoder #0, pin A (Input)
[... cut ...]
    Aug 24 17:57:42 mksocfpga-nano-soc msgd:0: rtapi_app exit detected - scheduled shutdown
    Aug 24 17:57:44 mksocfpga-nano-soc msgd:0: msgd shutting down
    Aug 24 17:57:44 mksocfpga-nano-soc msgd:0: zeroconf: unregistering 'Log service on mksocfpga-nano-soc.local pid 2651'
    Aug 24 17:57:44 mksocfpga-nano-soc msgd:0: log buffer hwm: 3% (199 msgs, 17162 bytes out of 524288)
    Aug 24 17:57:44 mksocfpga-nano-soc msgd:0: normal shutdown - global segment detached
    machinekit@mksocfpga-nano-soc:~$

the-snowwhite, 2019-08-24:
only SSerial info found is this:

    Aug 24 17:57:15 mksocfpga-nano-soc msgd:0: hal_lib:2656:rt hm2/hm2_de0n.0: SSerial: 0
    Aug 24 17:57:15 mksocfpga-nano-soc msgd:0: hal_lib:2656:rt hm2/hm2_de0n.0:   version 43

No SSerial pins

the-snowwhite, 2019-08-24:
BTW found this explanation of the sserial_port_0=00xx  parameter, this seems to be for 2 SSerial cores in config ?
https://forum.linuxcnc.org/32-documents/26138-hostmot2-sserial-port-n-xxxxxxxx

blazini36, 2019-08-24:
> BTW found this explanation of the sserial_port_0=00xx parameter, this seems to be for 2 SSerial cores in config ?
> https://forum.linuxcnc.org/32-documents/26138-hostmot2-sserial-port-n-xxxxxxxx

It's just a typical usage, setting a SS core to mode 0 that does not exist should not cause an issue, it's a carryover from MESA boards that use the first channel for onboard I/O and include an expansion port. If "sserial_port_0=0xxx" made a difference I'd be shocked.

I'm still wrapping my brain around your findings trying to compare it to a typical LinuxCNC setup since I don't have much MK experience otherwise.

blazini36, 2019-08-24:
Carryover from The MK GG..........

I'm not sure if this helps you but SS pins always look like normal I/O if they don't detect a remote on the channel after coming up. Since I have it out I connected the 8i20 to my project machine's 7i96 that I don't typically use SS with. This is LinuxCNC 2.8pre1

No 8i20 connected, note IO pins 30 and 31:
`shade@Viewer:~$ halrun
halcmd: source /home/shade/linuxcnc/SS_Test.halrun 
Note: Using POSIX realtime
hm2: loading Mesa HostMot2 driver version 0.15
hm2_eth: loading Mesa AnyIO HostMot2 ethernet driver version 0.2
hm2_eth: 10.10.10.10: Hardware address: 00:60:1b:16:00:21
hm2_eth: discovered 7I96
hm2/hm2_7i96.0: Smart Serial Firmware Version 43
hm2/hm2_7i96.0: 51 I/O Pins used:
hm2/hm2_7i96.0:     IO Pin 000 (TB3-01): IOPort
hm2/hm2_7i96.0:     IO Pin 001 (TB3-02): IOPort
hm2/hm2_7i96.0:     IO Pin 002 (TB3-03): IOPort
hm2/hm2_7i96.0:     IO Pin 003 (TB3-04): IOPort
hm2/hm2_7i96.0:     IO Pin 004 (TB3-05): IOPort
hm2/hm2_7i96.0:     IO Pin 005 (TB3-06): IOPort
hm2/hm2_7i96.0:     IO Pin 006 (TB3-07): IOPort
hm2/hm2_7i96.0:     IO Pin 007 (TB3-08): IOPort
hm2/hm2_7i96.0:     IO Pin 008 (TB3-09): IOPort
hm2/hm2_7i96.0:     IO Pin 009 (TB3-10): IOPort
hm2/hm2_7i96.0:     IO Pin 010 (TB3-11): IOPort
hm2/hm2_7i96.0:     IO Pin 011 (TB3-13/TB3-14): SSR #0, pin Out-00 (Output)
hm2/hm2_7i96.0:     IO Pin 012 (TB3-15/TB3-16): SSR #0, pin Out-01 (Output)
hm2/hm2_7i96.0:     IO Pin 013 (TB3-17/TB3-18): SSR #0, pin Out-02 (Output)
hm2/hm2_7i96.0:     IO Pin 014 (TB3-19/TB3-20): SSR #0, pin Out-03 (Output)
hm2/hm2_7i96.0:     IO Pin 015 (TB3-21/TB3-22): SSR #0, pin Out-04 (Output)
hm2/hm2_7i96.0:     IO Pin 016 (TB3-23/TB3-24): SSR #0, pin Out-05 (Output)
hm2/hm2_7i96.0:     IO Pin 017 (TB1-02/TB1-03): StepGen #0, pin Step (Output)
hm2/hm2_7i96.0:     IO Pin 018 (TB1-04/TB1-05): StepGen #0, pin Direction (Output)
hm2/hm2_7i96.0:     IO Pin 019 (TB1-08/TB1-09): StepGen #1, pin Step (Output)
hm2/hm2_7i96.0:     IO Pin 020 (TB1-10/TB1-11): StepGen #1, pin Direction (Output)
hm2/hm2_7i96.0:     IO Pin 021 (TB1-14/TB1-15): StepGen #2, pin Step (Output)
hm2/hm2_7i96.0:     IO Pin 022 (TB1-16/TB1-17): StepGen #2, pin Direction (Output)
hm2/hm2_7i96.0:     IO Pin 023 (TB1-20/TB1-21): StepGen #3, pin Step (Output)
hm2/hm2_7i96.0:     IO Pin 024 (TB1-22-TB1-23): StepGen #3, pin Direction (Output)
hm2/hm2_7i96.0:     IO Pin 025 (TB2-01/TB2-03): StepGen #4, pin Step (Output)
hm2/hm2_7i96.0:     IO Pin 026 (TB2-04/TB2-05): StepGen #4, pin Direction (Output)
hm2/hm2_7i96.0:     IO Pin 027 (TB2-07/TB2-08): Encoder #0, pin A (Input)
hm2/hm2_7i96.0:     IO Pin 028 (TB2-10/TB2-11): Encoder #0, pin B (Input)
hm2/hm2_7i96.0:     IO Pin 029 (TB2-13/TB2-14): Encoder #0, pin Index (Input)
hm2/hm2_7i96.0:     IO Pin 030 (TB2-16/TB2-17): IOPort
hm2/hm2_7i96.0:     IO Pin 031 (TB2-18/TB2-19): IOPort
hm2/hm2_7i96.0:     IO Pin 032 (internal): IOPort
hm2/hm2_7i96.0:     IO Pin 033 (internal): SSR #0, pin AC Ref (internal) (Output)
hm2/hm2_7i96.0:     IO Pin 034 (P1-01): IOPort
hm2/hm2_7i96.0:     IO Pin 035 (P1-02): IOPort
hm2/hm2_7i96.0:     IO Pin 036 (P1-03): IOPort
hm2/hm2_7i96.0:     IO Pin 037 (P1-04): IOPort
hm2/hm2_7i96.0:     IO Pin 038 (P1-05): IOPort
hm2/hm2_7i96.0:     IO Pin 039 (P1-06): IOPort
hm2/hm2_7i96.0:     IO Pin 040 (P1-07): IOPort
hm2/hm2_7i96.0:     IO Pin 041 (P1-08): IOPort
hm2/hm2_7i96.0:     IO Pin 042 (P1-09): IOPort
hm2/hm2_7i96.0:     IO Pin 043 (P1-11): IOPort
hm2/hm2_7i96.0:     IO Pin 044 (P1-13): IOPort
hm2/hm2_7i96.0:     IO Pin 045 (P1-15): IOPort
hm2/hm2_7i96.0:     IO Pin 046 (P1-17): IOPort
hm2/hm2_7i96.0:     IO Pin 047 (P1-19): IOPort
hm2/hm2_7i96.0:     IO Pin 048 (P1-21): IOPort
hm2/hm2_7i96.0:     IO Pin 049 (P1-23): IOPort
hm2/hm2_7i96.0:     IO Pin 050 (P1-25): IOPort
hm2/hm2_7i96.0: registered`

With the 8i20 connected and the same source file:
`shade@Viewer:~$ halrun
halcmd: source /home/shade/linuxcnc/SS_Test.halrun 
[... 99 lines cut ...]
    14  bit   OUT          TRUE  hm2_7i96.0.8i20.0.0.status.current-lim-not
    14  bit   OUT         FALSE  hm2_7i96.0.8i20.0.0.status.ext-reset
    14  bit   OUT          TRUE  hm2_7i96.0.8i20.0.0.status.ext-reset-not
    14  bit   OUT          TRUE  hm2_7i96.0.8i20.0.0.status.no-enable
    14  bit   OUT         FALSE  hm2_7i96.0.8i20.0.0.status.no-enable-not
    14  bit   OUT         FALSE  hm2_7i96.0.8i20.0.0.status.pid-on
    14  bit   OUT          TRUE  hm2_7i96.0.8i20.0.0.status.pid-on-not
    14  bit   OUT         FALSE  hm2_7i96.0.8i20.0.0.status.sw-reset
    14  bit   OUT          TRUE  hm2_7i96.0.8i20.0.0.status.sw-reset-not
    14  bit   OUT         FALSE  hm2_7i96.0.8i20.0.0.status.wd-reset
    14  bit   OUT          TRUE  hm2_7i96.0.8i20.0.0.status.wd-reset-not
`

Source file is just this:
`loadrt trivkins 
loadrt motmod servo_period_nsec=1000000 num_joints=3
loadrt hostmot2
loadrt hm2_eth board_ip="10.10.10.10" config="firmware=hm2/7i96/7i96.BIT num_encoders=1 num_pwmgens=0 num_stepgens=5 sserial_port_0=00xxxx" 
setp   hm2_7i96.0.watchdog.timeout_ns 5000000

addf hm2_7i96.0.read          servo-thread
addf hm2_7i96.0.write         servo-thread

start
`

the-snowwhite, 2019-08-24:
OK SSerial.c states that pind don't get created if nothing is probed and that the Tx pins are only momentairly enabled for initial probing so I did a new set of scope probing and got some results:..

[image]

[image]

the-snowwhite, 2019-08-24:
With some measurements:
DE10_Nano_FB_Cramps.st_fpga_soc_dc1f_ss.rbf_ssreial-pin35_1us-falling_edge-mess:
[image]

the-snowwhite, 2019-08-24:
DE10_Nano_FB_Cramps.st_fpga_soc_dc1f_ss.rbf_ssreial-pin35_1us-falling_edge-mess_aligned:
[image]

blazini36, 2019-08-24:
It's definately *a message* but I'm not sure if it's the right message or the right timing, I was getting something similar when I checked before you made changes to the firmware source. According to Peter Wallace, you should see a 0xDF and the timing is 2.5mbaud.

According to the manual for your scope, assuming this is the one, it can do serial decoding. If you can set it up to decode rs232 @ 2.5mbaud you might see something.

the-snowwhite, 2019-08-24:
SSerial-Serial_Decoding:
[image]

SSerial-Serial_Decoded
[image]

Well I do see DF as first byte ... ?

the-snowwhite, 2019-08-24:
i use this exact hal command sequence:

    halrun
    loadrt trivkins
    loadrt tp
    loadrt motmod servo_period_nsec=1000000 num_joints=3 num_aio=51 num_dio=21 tp=tp kins=trivkins
    loadrt hostmot2 debug_idrom=1 debug_modules=1
    newinst hm2_soc_ol hm2-socfpga0 already_programmed=1 -- config="num_stepgens=3  sserial_port_0=00xx" debug=1

blazini36, 2019-08-24:
I'd say that transmission looks good, doubt it's a coincidence. I'm still not getting any where, trying to figure out if a simple rs232 shot to the 8i20 from the Nano will work to eliminate any hardware problems.

the-snowwhite, 2019-08-25:
I have been looking at the source files machinekit-hal/src/hal/drivers/mesa-hostmot2/sserial.c and sserial.h there is only 1 update (to sserial.c) in machinekit(-hal) since 2015 as to the ones in the lcnc repository which have noticable differences.(especially compairing the 2  hm2_sserial_parse_md function's )

Compare patch here:

    --- /home/mib/Developer/the-snowwhite_git/test/machinekit/hm2_sserial_parse_md.txt
    +++ /home/mib/Developer/the-snowwhite_git/test/linuxcnc/hm2_sserial_parse_md.txt
    @@ -3,7 +3,7 @@
        int i, c;
        int pin = -1;
        int port_pin, port;
    -    u32 ddr_reg, src_reg, buff;
    +    rtapi_u32 ddr_reg, src_reg, buff;
        int r = -EINVAL;
        int count = 0;
        int chan_counts[] = {0,0,0,0,0,0,0,0};
    @@ -94,9 +94,9 @@
            }
    
            hm2->llio->write(hm2->llio, hm2->ioport.ddr_addr + 4 * port,
    -                         &ddr_reg, sizeof(u32));
    +                         &ddr_reg, sizeof(rtapi_u32));
            hm2->llio->write(hm2->llio, hm2->ioport.alt_source_addr + 4 * port,
    -                         &src_reg, sizeof(u32));
    +                         &src_reg, sizeof(rtapi_u32));
    
        }
    
    @@ -124,7 +124,7 @@
            hm2->sserial.version = buff;
    
            r = check_set_baudrate(hm2, inst) < 0;
    -        if (r < 0) {goto fail0;}
    +        if (r < 0) goto fail0;
    
            //start up in setup mode
            r = hm2_sserial_stopstart(hm2, md, inst, 0xF00) < 0;
    @@ -133,21 +133,21 @@
            inst->num_remotes = 0;
    
            for (c = 0 ; c < inst->num_channels ; c++) {
    -            u32 addr0, addr1, addr2;
    -            u32 user0, user1, user2;
    +            rtapi_u32 addr0, addr1, addr2;
    +            rtapi_u32 user0, user1, user2;
    
                addr0 = md->base_address + 3 * md->register_stride
    -                                    + i * md->instance_stride + c * sizeof(u32);
    +                                    + i * md->instance_stride + c * sizeof(rtapi_u32);
                HM2READ(addr0, user0);
                HM2_DBG("Inst %i Chan %i User0 = %x\n", i, c, user0);
    
                addr1 = md->base_address + 4 * md->register_stride
    -                                    + i * md->instance_stride + c * sizeof(u32);
    +                                    + i * md->instance_stride + c * sizeof(rtapi_u32);
                HM2READ(addr1, user1);
                HM2_DBG("Inst %i Chan %i User1 = %x\n", i, c, user1);
    
                addr2 = md->base_address + 5 * md->register_stride
    -            + i * md->instance_stride + c * sizeof(u32);
    +            + i * md->instance_stride + c * sizeof(rtapi_u32);
                HM2READ(addr2, user2);
                HM2_DBG("Inst %i Chan %i User2 = %x\n", i, c, user2);
    
    @@ -217,7 +217,7 @@
                }
            }
            if (inst->num_remotes > 0){
    -            if ((r = hm2_sserial_setup_channel(hm2, inst, count)) < 0 ) {
    +            if ((r = hm2_sserial_setup_channel(hm2, inst, i)) < 0 ) {
                    HM2_ERR("Smart Serial setup failure on instance %i\n",
                            inst->device_id);
                    goto fail0;}
    @@ -225,14 +225,24 @@
                    HM2_ERR("Remote setup failure on instance %i\n",
                            inst->device_id);
                    goto fail0;}
    +            // Nothing happens without a "Do It" command
    +            if ((r = hm2_register_tram_write_region(hm2,inst->command_reg_addr,
    +                                       sizeof(rtapi_u32),
    +                                       &inst->command_reg_write)) < 0){
    +                HM2_ERR("error registering tram DoIt write to sserial "
    +                "command register (%d)\n", i);
    +                goto fail0;}
    +
                if ((r = hm2_sserial_stopstart(hm2, md, inst, 0x900)) < 0 ){
                    HM2_ERR("Failed to restart device %i on instance\n",
                            inst->device_id);
                    goto fail0;}
    -            if ((r = hm2_sserial_check_errors(hm2, inst)) < 0) {
    +            if ((r = hm2_sserial_check_local_errors(hm2, inst)) < 0) {
                    //goto fail0; // Ignore it for the moment.
                }
                //only increment the instance index if this one is populated
    +            //otherwise the "slot" is re-used to keep active ports
    +            //contiguous in the array
                count++ ;
            }
        }
    @@ -243,16 +253,19 @@
        buff=0x800; //Stop All
        for (i = 0 ; i < hm2->sserial.num_instances ; i++) {
            hm2_sserial_instance_t *inst = &hm2->sserial.instance[i];
    -        hm2->llio->write(hm2->llio, inst->command_reg_addr, &buff, sizeof(u32));
    +        hm2->llio->write(hm2->llio, inst->command_reg_addr, &buff, sizeof(rtapi_u32));
    +        if (hm2_sserial_waitfor(hm2, inst->command_reg_addr, 0xFFFFFFFF,51) < 0){
    +            return -EINVAL;
    +        }
        }
        // Return the physical ports to default
        ddr_reg = 0;
        src_reg = 0;
        for (port  = 0; port < hm2->ioport.num_instances; port ++) {
            hm2->llio->write(hm2->llio, hm2->ioport.ddr_addr + 4 * port,
    -                         &ddr_reg, sizeof(u32));
    +                         &ddr_reg, sizeof(rtapi_u32));
            hm2->llio->write(hm2->llio, hm2->ioport.alt_source_addr + 4 * port,
    -                         &src_reg, sizeof(u32));
    +                         &src_reg, sizeof(rtapi_u32));
        }
        return hm2->sserial.num_instances;
    
Due to a lack of confimation of a working Mksocfpga SSerial setup...:
If it were me I would test out Machinekit'fying a copy of the sserial.c and sserial.h files from linuxcnc repo, and see if that does the trick.

blazini36, 2019-08-25:
If you fix and compile them, I'll try em out!

the-snowwhite, 2019-08-25:
OK give me some time

the-snowwhite, 2019-08-25:
mk-hal deb for test with sserial parse only mod.
Here:
https://groups.google.com/forum/#!topic/machinekit/eVhvTnuhblE%5B126-150%5D

Due to the number of changes I choose to creative a conservative change only modding the:
hm2_sserial_parse_md() function, this is what does the intial probing and pin setting..
The file you should use I think is:
 machinekit-hal-rt-preempt_0.2.1566764039.gitf3af8e09e-1~stretch_armhf.deb

I added the other one just in case...

blazini36, 2019-08-26:
Definitely on the right track there! I made a quick mod to my prototype board to get a receiver output voltage closer to the 3.3v GPIO pin tolerance and installed the .deb you posted, I now have discovery.
'''
machinekit@mksocfpga-nano-soc:~$ halrun
msgd:0 running
rtapi:0 running
halrun: Realtime already running.  Use 'halrun -U' to stop existing realtime session.
machinekit@mksocfpga-nano-soc:~$ halrun -U
machinekit@mksocfpga-nano-soc:~$ halrun
msgd:0 stopped
rtapi:0 stopped
rtapi_msgd command:  /usr/libexec/linuxcnc/rtapi_msgd --instance=0 --rtmsglevel=1 --usrmsglevel=1 --halsize=524288
rtapi_app command:  /usr/libexec/linuxcnc/rtapi_app_rt-preempt --instance=0
halcmd: source ~/Desktop/SS_Test_MK.halrun
halcmd: show pin *8i20*
Component Pins:
  Comp   Inst Type  Dir         Value  Name                                            Epsilon Flags  linked to:
   517    519 float IN              0  hm2_5i25.0.8i20.0.0.angle                 0.000010        --l-
   517    519 float OUT             0  hm2_5i25.0.8i20.0.0.bus-voltage           0.000010        --l-
   517    519 float OUT             0  hm2_5i25.0.8i20.0.0.card-temp             0.000010        --l-
   517    519 float IN              0  hm2_5i25.0.8i20.0.0.current               0.000010        --l-
   517    519 bit   OUT         FALSE  hm2_5i25.0.8i20.0.0.fault.U-current               --l-
   517    519 bit   OUT         FALSE  hm2_5i25.0.8i20.0.0.fault.U-current-not           --l-
   517    519 bit   OUT         FALSE  hm2_5i25.0.8i20.0.0.fault.V-current               --l-
   517    519 bit   OUT         FALSE  hm2_5i25.0.8i20.0.0.fault.V-current-not           --l-
   517    519 bit   OUT         FALSE  hm2_5i25.0.8i20.0.0.fault.W-current               --l-
   517    519 bit   OUT         FALSE  hm2_5i25.0.8i20.0.0.fault.W-current-not           --l-
   517    519 bit   OUT         FALSE  hm2_5i25.0.8i20.0.0.fault.bus-high                --l-
   517    519 bit   OUT         FALSE  hm2_5i25.0.8i20.0.0.fault.bus-high-not            --l-
   517    519 bit   OUT         FALSE  hm2_5i25.0.8i20.0.0.fault.bus-overv               --l-
   517    519 bit   OUT         FALSE  hm2_5i25.0.8i20.0.0.fault.bus-overv-not           --l-
   517    519 bit   OUT         FALSE  hm2_5i25.0.8i20.0.0.fault.bus-underv              --l-
   517    519 bit   OUT         FALSE  hm2_5i25.0.8i20.0.0.fault.bus-underv-not          --l-
   517    519 bit   OUT         FALSE  hm2_5i25.0.8i20.0.0.fault.framingr                --l-
   517    519 bit   OUT         FALSE  hm2_5i25.0.8i20.0.0.fault.framingr-not            --l-
   517    519 bit   OUT         FALSE  hm2_5i25.0.8i20.0.0.fault.module                  --l-
   517    519 bit   OUT         FALSE  hm2_5i25.0.8i20.0.0.fault.module-not              --l-
   517    519 bit   OUT         FALSE  hm2_5i25.0.8i20.0.0.fault.no-enable               --l-
   517    519 bit   OUT         FALSE  hm2_5i25.0.8i20.0.0.fault.no-enable-not           --l-
   517    519 bit   OUT         FALSE  hm2_5i25.0.8i20.0.0.fault.overcurrent             --l-
   517    519 bit   OUT         FALSE  hm2_5i25.0.8i20.0.0.fault.overcurrent-no          --l-
   517    519 bit   OUT         FALSE  hm2_5i25.0.8i20.0.0.fault.overrun                 --l-
   517    519 bit   OUT         FALSE  hm2_5i25.0.8i20.0.0.fault.overrun-not             --l-
   517    519 bit   OUT         FALSE  hm2_5i25.0.8i20.0.0.fault.overtemp                --l-
   517    519 bit   OUT         FALSE  hm2_5i25.0.8i20.0.0.fault.overtemp-not            --l-
   517    519 bit   OUT         FALSE  hm2_5i25.0.8i20.0.0.fault.watchdog                --l-
   517    519 bit   OUT         FALSE  hm2_5i25.0.8i20.0.0.fault.watchdog-not            --l-
   517    519 bit   OUT         FALSE  hm2_5i25.0.8i20.0.0.status.brake-old              --l-
   517    519 bit   OUT         FALSE  hm2_5i25.0.8i20.0.0.status.brake-old-not          --l-
   517    519 bit   OUT         FALSE  hm2_5i25.0.8i20.0.0.status.brake-on               --l-
   517    519 bit   OUT         FALSE  hm2_5i25.0.8i20.0.0.status.brake-on-not           --l-
   517    519 bit   OUT         FALSE  hm2_5i25.0.8i20.0.0.status.bus-underv             --l-
   517    519 bit   OUT         FALSE  hm2_5i25.0.8i20.0.0.status.bus-underv-no          --l-
   517    519 bit   OUT         FALSE  hm2_5i25.0.8i20.0.0.status.current-lim            --l-
   517    519 bit   OUT         FALSE  hm2_5i25.0.8i20.0.0.status.current-lim-n          --l-
   517    519 bit   OUT         FALSE  hm2_5i25.0.8i20.0.0.status.ext-reset              --l-
   517    519 bit   OUT         FALSE  hm2_5i25.0.8i20.0.0.status.ext-reset-not          --l-
   517    519 bit   OUT         FALSE  hm2_5i25.0.8i20.0.0.status.no-enable              --l-
   517    519 bit   OUT         FALSE  hm2_5i25.0.8i20.0.0.status.no-enable-not          --l-
   517    519 bit   OUT         FALSE  hm2_5i25.0.8i20.0.0.status.pid-on                 --l-
   517    519 bit   OUT         FALSE  hm2_5i25.0.8i20.0.0.status.pid-on-not             --l-
   517    519 bit   OUT         FALSE  hm2_5i25.0.8i20.0.0.status.sw-reset               --l-
   517    519 bit   OUT         FALSE  hm2_5i25.0.8i20.0.0.status.sw-reset-not           --l-
   517    519 bit   OUT         FALSE  hm2_5i25.0.8i20.0.0.status.wd-reset               --l-
   517    519 bit   OUT         FALSE  hm2_5i25.0.8i20.0.0.status.wd-reset-not           --l-
'''

You say there's still work to be done on the driver right?

the-snowwhite, 2019-08-26:
BTW
sudo dpkg -i machinekit-hal-rt-preempt_0.2.1566764039.gitf3af8e09e-1~stretch_armhf.deb
?

blazini36, 2019-08-26:
??

the-snowwhite, 2019-08-26:
Ups I forgot to refresh and overlooked your former message

the-snowwhite, 2019-08-26:
> You say there's still work to be done on the driver right?
I do not know test if the rest of it works as supposed to.
I dont have the Machinekit knowledge to be able to see in the source code if the other lcnc mods are mk compatible therefore I left them out.

blazini36, 2019-08-27:
> Your removal and re-install commands look okay.
>
> It did get a bit confusing since I was changing things around quite a bit
> trying to get it working until the quartus firmware issues were resolved. I
> very well could have screwed up my hardware in the meantime.
>
> Yeah most probably what happened and what is now giving the "DoIt not
> cleared from previous servo thread", error.... Perhaps you should clear
> your SSerial interface experience logs and work back towards your original
> HW design from when the SSerial hdl core was missing and start over...

I didn't make any software changes other than installing the deb you packed
up. The hardware as is connects to a LinuxCNC install without issue.
the  "DoIt not cleared from previous servo thread" is where we're at as far
as MK goes.

the-snowwhite, 2019-08-27:
Your removal and re-install commands look okay.

> It did get a bit confusing since I was changing things around quite a bit trying to get it working until the quartus firmware issues were resolved. I very well could have screwed up my hardware in the meantime.

Yeah most probably what happened and what is now giving the "DoIt not cleared from previous servo thread", error.... Perhaps you should clear your SSerial interface experience logs and work back towards your original HW design from when the SSerial hdl core was missing and start over...

blazini36, 2019-08-27:
> OK
> I'll try building a full Mk modded sserial.c/h and see where that goes...

I did kind of have the thought that maybe SS did did discovery at a lower baud rate and then switched to the 2.5mbaud. Kind of thinking that maybe on the hardware side there were signal issues at higher speeds.

PCW confirmed that SS runs at 2.5mbaud the whole time from discovery to actual operation , but that’s all he really said. So I have to think that if my board can handle discovery which is 2 way communication @ 2.5mbaud, then there’s no real reason it should have comm errors afterwards.

I wish I knew more about it but for now I’d have to assume the hardware is good and your new firmware is good.

the-snowwhite, 2019-08-27:
OK
I'll try building a full Mk modded sserial.c/h and see where that goes...

the-snowwhite, 2019-08-27:
OK
Lcnc SSerial Mk modded (without rtapi_stdint.h):
posted in the [GGroup](https://groups.google.com/forum/#!topic/machinekit/eVhvTnuhblE%5B126-150%5D)

the-snowwhite, 2019-08-27:
Let me know when you get around to testing the lcnc ssreial SW mod... :-)

the-snowwhite, 2023-11-11:
@blazini36  Closing due to lack of response

closed 2023-11-11 by the-snowwhite