machinekit/mksocfpga #116: Mksocfpga Failing builds

the-snowwhite, 2020-10-22:
@cerna 
https://github.com/machinekit/mksocfpga/pull/115/checks?check_run_id=1294764215

AFAIK someone fulled the plug on the old build system a while ago and
someone is yet to place in the replacement ?

cerna, 2020-10-22:
@the-snowwhite,
yes. This is the same as what is root behind machinekit/machinekit-hal#268 - the Jenkins server was turned off silently with everything dependent on it shut down too.

I knew this was going to cause a problem with mksocfpga, but A) the repository was long-time dormant and B) I am FPGA virgin - I have read multiple articles and tutorials, looked into the AMBA AXI interface and affiliated Linux kernel modules, have few devices which have FPGA integrated and was generally interested in the technology, but never actually programmed any nor own development board. So I was very bashful about building a CI/CD workflow for it.

The failure in #115 looks like was on Jessie system and a first build after the repository was changed from TRAVIS-CI ORG to COM. Jessie is EOLed and I actually turned off building and testing Machinekit-HAL for it. While the error does not look as connected to issues which generally happen after Debian distribution is laid to rest, it is still valid point if this test is viable.

Have you tried what the Travis does locally on your machine?

the-snowwhite, 2020-10-22:
Unh I found the online travis report [seems like it failed early in the](https://travis-ci.com/github/machinekit/mksocfpga/builds/191760419#L198)
 (to my pr unrelated) 
quartus build fase.
Since @cdsteinkuehler mastered this docker image I hope he can chime in on what seems to be happening there.
AKAIK this failure is unrelated to any changes to code in the mksocfpga repo.

cerna, 2020-10-22:
First try on #115 using the same commands as TRACI-CI locally was too unsuccessful, but it failed on different error:

```
docker run -v $(pwd):/work cdsteinkuehler/jessie-quartus-15.1.2 /work/Scripts/travis_build.sh
qsys-generate soc_system.qsys --synthesis=VERILOG 
2020.10.22.22:37:20 Info: Saving generation log to /work/HW/QuartusProjects/DE0_Nano_SoC_Cramps/soc_system/soc_system_generation.rpt
2020.10.22.22:37:20 Info: Starting: <b>Create HDL design files for synthesis</b>
2020.10.22.22:37:20 Info: qsys-generate /work/HW/QuartusProjects/DE0_Nano_SoC_Cramps/soc_system.qsys --synthesis=VERILOG --output-directory=/work/HW/QuartusProjects/DE0_Nano_SoC_Cramps/soc_system/synthesis --family="Cyclone V" --part=5CSEMA4U23C6
2020.10.22.22:37:20 Info: Loading DE0_Nano_SoC_Cramps/soc_system.qsys
2020.10.22.22:37:20 Info: Reading input file
2020.10.22.22:37:20 Info: Adding button_pio [altera_avalon_pio 15.1]
2020.10.22.22:37:20 Info: Parameterizing module button_pio
2020.10.22.22:37:20 Info: Adding clk_0 [clock_source 15.1]
2020.10.22.22:37:20 Info: Parameterizing module clk_0
2020.10.22.22:37:20 Info: Adding dipsw_pio [altera_avalon_pio 15.1]
2020.10.22.22:37:20 Info: Parameterizing module dipsw_pio
2020.10.22.22:37:20 Info: Adding fpga_only_master [altera_jtag_avalon_master 15.1]
2020.10.22.22:37:20 Info: Parameterizing module fpga_only_master
2020.10.22.22:37:20 Info: Adding hm2reg_io_0 [hm2reg_io 1.0]
2020.10.22.22:37:20 Info: Parameterizing module hm2reg_io_0
2020.10.22.22:37:20 Info: Adding hps_0 [altera_hps 15.1]
2020.10.22.22:37:20 Info: Parameterizing module hps_0
2020.10.22.22:37:20 Info: Adding hps_only_master [altera_jtag_avalon_master 15.1]
2020.10.22.22:37:20 Info: Parameterizing module hps_only_master
2020.10.22.22:37:20 Info: Adding intr_capturer_0 [intr_capturer 100.99.98.97]
2020.10.22.22:37:20 Info: Parameterizing module intr_capturer_0
2020.10.22.22:37:20 Info: Adding jtag_uart [altera_avalon_jtag_uart 15.1]
2020.10.22.22:37:20 Info: Parameterizing module jtag_uart
2020.10.22.22:37:20 Info: Adding led_pio [altera_avalon_pio 15.1]
2020.10.22.22:37:20 Info: Parameterizing module led_pio
2020.10.22.22:37:20 Info: Adding mm_bridge_0 [altera_avalon_mm_bridge 15.1]
2020.10.22.22:37:20 Info: Parameterizing module mm_bridge_0
2020.10.22.22:37:20 Info: Adding onchip_memory2_0 [altera_avalon_onchip_memory2 15.1]
2020.10.22.22:37:20 Info: Parameterizing module onchip_memory2_0
2020.10.22.22:37:20 Info: Adding pll_0 [altera_pll 15.1]
2020.10.22.22:37:20 Info: Parameterizing module pll_0
2020.10.22.22:37:20 Info: Adding sysid_qsys [altera_avalon_sysid_qsys 15.1]
2020.10.22.22:37:20 Info: Parameterizing module sysid_qsys
2020.10.22.22:37:20 Info: Building connections
2020.10.22.22:37:20 Info: Parameterizing connections
2020.10.22.22:37:20 Info: Validating
2020.10.22.22:37:29 Info: Done reading input file
2020.10.22.22:37:32 Info: soc_system.hps_0: HPS Main PLL counter settings: n = 0  m = 73
2020.10.22.22:37:32 Info: soc_system.hps_0: HPS peripherial PLL counter settings: n = 0  m = 39
2020.10.22.22:37:32 Info: soc_system.pll_0: The legal reference clock frequency is 5.0 MHz..800.0 MHz
2020.10.22.22:37:32 Info: soc_system.pll_0: Able to implement PLL with user settings
2020.10.22.22:37:32 Info: soc_system.sysid_qsys: System ID is not assigned automatically. Edit the System ID parameter to provide a unique ID
2020.10.22.22:37:32 Info: soc_system.sysid_qsys: Time stamp will be automatically updated when this component is generated.
2020.10.22.22:38:06 Info: soc_system: Generating <b>soc_system</b> "<b>soc_system</b>" for QUARTUS_SYNTH
2020.10.22.22:38:13 Info: Interconnect is inserted between master hps_0.h2f_lw_axi_master and slave mm_bridge_0.s0 because the master is of type axi and the slave is of type avalon.
2020.10.22.22:38:18 Info: Interconnect is inserted between master hps_only_master.master and slave hps_0.f2h_axi_slave because the master is of type avalon and the slave is of type axi.
2020.10.22.22:38:19 Warning: hps_0.f2h_irq0: Cannot connect clock for <b>irq_mapper.sender</b>
2020.10.22.22:38:19 Warning: hps_0.f2h_irq0: Cannot connect reset for <b>irq_mapper.sender</b>
2020.10.22.22:38:19 Warning: hps_0.f2h_irq1: Cannot connect clock for <b>irq_mapper_001.sender</b>
2020.10.22.22:38:19 Warning: hps_0.f2h_irq1: Cannot connect reset for <b>irq_mapper_001.sender</b>
2020.10.22.22:38:25 Info: button_pio: Starting RTL generation for module 'soc_system_button_pio'
2020.10.22.22:38:25 Info: button_pio:   Generation command is [exec /home/builder/altera_lite/15.1/quartus/linux64/perl/bin/perl -I /home/builder/altera_lite/15.1/quartus/linux64/perl/lib -I /home/builder/altera_lite/15.1/quartus/sopc_builder/bin/europa -I /home/builder/altera_lite/15.1/quartus/sopc_builder/bin/perl_lib -I /home/builder/altera_lite/15.1/quartus/sopc_builder/bin -I /home/builder/altera_lite/15.1/quartus/../ip/altera/sopc_builder_ip/common -I /home/builder/altera_lite/15.1/quartus/../ip/altera/sopc_builder_ip/altera_avalon_pio -- /home/builder/altera_lite/15.1/quartus/../ip/altera/sopc_builder_ip/altera_avalon_pio/generate_rtl.pl --name=soc_system_button_pio --dir=/tmp/alt8557_2162494273812915309.dir/0001_button_pio_gen/ --quartus_dir=/home/builder/altera_lite/15.1/quartus --verilog --config=/tmp/alt8557_2162494273812915309.dir/0001_button_pio_gen//soc_system_button_pio_component_configuration.pl  --do_build_sim=0  ]
2020.10.22.22:38:25 Info: button_pio: Done RTL generation for module 'soc_system_button_pio'
[... cut ...]
    Error: Peak virtual memory: 1249 megabytes
    Error: Processing ended: Thu Oct 22 22:45:54 2020
    Error: Elapsed time: 00:06:35
    Error: Total CPU time (on all processors): 00:07:08
make: *** [stamp/quartus_pin_assignments.stamp] Error 3
Makefile:210: recipe for target 'stamp/quartus_pin_assignments.stamp' failed
```

BTW, the situation with Jenkins is why I am proponent of public infrastructure. That way it will be quickly reproducible and not dependent on any Machinekit organization member.

the-snowwhite, 2020-10-22:
That seem's like a correct run:
This Error:

    Error (10481): VHDL Use Clause error at MakeIOPorts.vhd(12): design library "pin" does not contain primary unit "Pintypes". 
    Verify that the primary unit exists in the library and has been successfully compiled. File: 
    /work/HW/hm2/wrappers/MakeIOPorts.vhd Line: 12

Stems from this warning:

    Warning (125092): Tcl Script File hm3_DE0_Nano_SoC_Cramps.qip not found

The problem is that:
travis_build.sh is broken:
it has to do more than just initiate the quartus build itself:

   cd /work/HW/QuartusProjects/DE0_Nano_SoC_Cramps/
   make -j$(nproc) rbf dts dtb

this file(hm3_DE0_Nano_SoC_Cramps.qip) is only generated when you initialize the build correctly by running some pre steps:

The way is pointed out in the [readme ](https://github.com/machinekit/mksocfpga)(also if you do not do so next thing missing will be the firmware id memory file(.mif)):

## Building a single CycloneV bit file:

build docker images from https://github.com/machinekit/mksocfpga.git

### Launch a Docker build machine with the latest mksocfpga source code
````
docker pull cdsteinkuehler/jessie-quartus-15.1.2

git clone https://github.com/machinekit/mksocfpga.git/
cd mksocfpga
docker run -itv $(pwd):/work cdsteinkuehler/jessie-quartus-15.1.2 /bin/bash

# You should now have a command prompt for the Docker Quartus build image
````

### Build the firmware-id protobuf bindings
````
# From the Docker Quartus build image command line, run:
cd /work/HW/firmware-tag
make TOPDIR=/work py-proto
````

### See bitfile configs available in a project
````
# From the Docker Quartus build image command line, run:
cd /work/HW/QuartusProjects/<project>
./build.sh ?
````

### Build the desired FPGA bit file
````
# From the Docker Quartus build image command line, run:
cd /work/HW/QuartusProjects/<project>
./build.sh <config>

the-snowwhite, 2023-11-11:
@cerna 
I think if #117 solved this issue it can be closed ?