machinekit/mksocfpga #121: Quartus Hostmot2 FPGA compiling error

Godtec, 2021-01-30:
Hello all:

Getting error on Hand compiling the mkscofpga in Quartus De0-nano-soc-DB25 code. (image attached)
 
My experience with VHDL is almost nothing. (limited Verilog HDL guy myself)

Error is saying that library unit "work" cannot find the primary unit "HostMot2_cfg"?
The Hostmot2.vhd file is in the directory HW/hm2... ???
I have complied the Qsys Soc-system separately without issue.

I am a stuck here.
Any help would be great!

[image]

Also, compiling the docker image following the mksocfpga github page seems to work for me. (takes a while tho.)
I would like to get the Quartus build environment working for me, this is the best experience for me.
I do not have a lot experience with Docker.

This is related to the groups.io posting on DRO feedback implementation. (posted Jan 29th)

Thanks to all who can give guidance!
Mike Kennedy

the-snowwhite, 2021-01-31:
@Godtec 
Since it's years since I have dabbled with the mksocfpga quartus stuff I'm struggling to recall however.
I remember the best way to get started from scratch (with a local Quartus build environment) 
is to first do a docker run. (this generates the necessary config files) in your local mksocfpga repo.
Then you should be able to load the same project as you just have docker compiled into quartus 15.1 AFAIKR
And get it to compile locally.

Godtec, 2021-01-31:
Hi Michael! (The-snowwhite!)

I really appreciate you answering and guiding me on this!

So, I did exactly what you have suggested, and I was able to create a docker version of the mksocfpga folder.
Then I opened it with Quartus 15.1 and it complied fine!
Off to the next steps!

Thanks!
Mike Kennedy

closed 2021-01-31 by Godtec