machinekit/mksocfpga #123: Lychee Hex support ?

youprint, 2022-05-06:
Two years that i have this tiny FPGA in my desk and nothing to do with.
[image]
it looks like a raspberry but with a Zync 7020.
I tryed to open one of those vivado project with Vivado 2022 and no XPR file and no success.
How can i add this board to be supported ?

claudiolorini, 2022-05-06:
It seems to be a nice board, but with classic chinesium stile documentation (i haven't managed to find any useful documentation on the interwebs).
What are you planning to do with this? step-dir signals out of the 2.54 connector? without proper documentation or schematic of the board i think it will be quite a hard task to achieve...

youprint, 2022-05-06:
4 step/dir and 4 encoder with a pwm output would be enough for me. that's not a 2.54 but a 2mm connector.
i already succefully ported hostmost2 to a XC7S15 using Vivado 2022.1 so i don't think it's going to be so hard. i just need a good start with a viable Vivado project.

claudiolorini, 2022-05-06:
[image]

..just 1 pin to spare, close call. 
good luck!

cdsteinkuehler, 2022-05-06:
Refer to the README file in the HW/VivadoProjects directory.  You have to first create the Vivado project using the tcl scripts.  Xilinx Vivado is VERY particular about exact versions when running these tcl scripts, so I recommend you use the exact version of Vivado expected by the script.  Once you have a project created, you can migrate it to a newer version of Vivado and make changes required for your platform (eg: change target device and top-level I/O definitions).

the-snowwhite, 2023-11-11:
@youprint 
Do you still have porting to this board in mind ?